rm-hull/luma.led_matrix #4: installing error

thib3113, 2015-01-04:
i launch 
'''shell
sudo python setup.py install
'''
and i have this error :
'''Shell
building 'max7219.spi' extension
creating build/temp.linux-armv6l-2.7
creating build/temp.linux-armv6l-2.7/src
gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c src/spi.c -o build/temp.linux-armv6l-2.7/src/spi.o
src/spi.c:20:20: fatal error: Python.h: Aucun fichier ou dossier de ce type
compilation terminated.
error: command 'gcc' failed with exit status 1
'''

thib3113, 2015-01-04:
it's ok, just i have python by default but not python-dev

rm-hull, 2015-01-04:
Glad you found the problem -- I'll add a section to the readme file to indicate that `python-dev` is a pre-requisite

commit 14a0a6e9 (2015-01-04): Add pre-requisite for python-dev, fixes #4
commit 9eb918dc (2015-02-13): Merge branch 'master' of github.com:rm-hull/max7219 into cascaded

* 'master' of github.com:rm-hull/max7219:
  Added MIT license file
  Ignores
  Added schematic for 3x seven-segement devices
  Add link to spi group so dont have to run as root
  Crop images
  Added images & link to datasheet
  Add pre-requisite for python-dev, fixes #4
  Fix missing import
  initial
  Tweak build instructions
  arch
  arch
  Add a Bitdeli badge to README
closed 2015-01-04 by rm-hull