rm-hull/luma.led_matrix #10: No way to rotate fonts?

tarasius, 2015-02-18:
Good day.
Is there any way to rotate font without re-specifying constants of characters? I.e. some rotation parameter. If no, it would be nice to have.
P.S.: I have Cyrillic font if somebody needs it.

rm-hull, 2015-02-18:
Hello, it should be quite easy to extend the code do some bit-fiddling to rotate 90, 180 or 270 degrees. I'll have a look at implementing it in the next few days.

In the meantime, if you're Cyrillic font works, then by all means fork the project, add to the `font.py` file and submit a pull request

rm-hull, 2015-02-19:
From Graphics Gems II

Yap, Sue-Ken, **A Fast 90-Degree Bitmap Rotator**, p. 84-85, code: p. 514-515, http://tog.acm.org/resources/GraphicsGems/gemsii/rotate8x8.c.

Just need to port it across to Python

rm-hull, 2015-02-20:
Ah! I see now why you want to be able to rotate (I just merged your PR) ... so I will write a little python script to rotate the UKR_FONT definitions,

commit 509a8208 (2015-02-20): Added matrix.orientation() and matrix.invert() functions - fixes #10
closed 2015-02-20 by rm-hull