rm-hull/luma.led_matrix #17: Python

Depado, 2015-03-13:


rm-hull, 2015-03-13:
Sure - never really considered Python 3 support, but I will add something like:

```python
try:
    xrange
except NameError:
    xrange = range
```
and check for other 2-to-3 compatibility issues as well

Depado, 2015-03-13:
In led.py line 360, there is a division. Division changes in python 3, it will return a float unless '//' is used. Anyway I tried your program but nothing happens on the matrix. When I plug the chip the other way around, it displays 3 exclamation mark, do you know anything about that kind of behaviour ? :/

rm-hull, 2015-03-13:
So, can I assume that you have a matrix rather than seven segment? and when you say "plug the chip the other way" do you mean the max7219 IC or the LED matrix?

It is easy to get the LED matrix plugged in the wrong way.

What happens when you run `python examples/matrix_test.py` ? 
Can you upload a picture of the circuit board when it is showing the exclamation marks?

Depado, 2015-03-13:
Yes I have a matrix. The led matrix is plugged the right way (notched side is on the bottom). Here are some pics of how I wired it. [Pictures](http://imgur.com/auFQdu8,qTBtLln,f47XgW4,5KwuYHW#0)
Nothing happens when I run the program.

Depado, 2015-03-13:
(I have put some print in the program to see what's executed, it runs smooth, but nothing happens on the matrix, I suspect a problem with the wiring or the assembly of the matrix... Though I have no idea what I could have done wrong)

rm-hull, 2015-03-13:
If you look at this pic, the semi-circle notch at the end of the chip should be aligned with the white outline below. The writing on the chip should be oriented the same way as the white 'MAX7219' lettering on the printed circuit board... Looking at your pics, the chip is plugged in the wrong way round.

[image]

The wiring from your earlier pictures looks correct. 

Try changing the chip round (and hope that hasnt been fried). 

If it still doesnt work, try taking out the LED matrix, and rotate it 180 degrees and re-insert it

Depado, 2015-03-13:
Already tried all that. I guess I fried the chip or something. I'll buy a new one and try again. Thanks for the help ! And thanks for the lib !

commit e414b834 (2015-03-15): Add python3 compatibility - fixes #17
closed 2015-03-15 by rm-hull