rm-hull/luma.led_matrix #19: How to scroll text in from top/bottom?

mB-PiBox, 2015-08-27:
If possible, how would I go about scrolling text down from the top, or up from the bottom.

I can get the letters already displayed to scroll down after being drawn on the display, but how would you scroll letters up from the bottom and stop in position on the display. (like scrolling left-to-right, except stopping once they reach the top of the display.)

rm-hull, 2015-08-28:
We would need an off-screen buffer which you would populate with the relevant bit patterns that represent the characters you want to display. Then, on calling scroll up or down, bit-OR the existing buffer with the MSB or LSB of the off-screen buffer and then shift that up or down respectively. Should be pretty straightforward - I'll have a go in the next day or two ... unless you fancy having a go at implementing it: I can write up some notes on this ticket to get you started

mB-PiBox, 2015-08-28:
I'm still learning the ins and outs of python and the Max7219
any note would be helpful and greatly appreciated.
most importantly would be "how do you store something in a 'buffer'?"

rm-hull, 2015-08-28:
Have a look at the `set_byte` and `letter` methods in https://github.com/rm-hull/max7219/blob/master/max7219/led.py as a starter, to see how letters are translated into the correct bit patterns to show. 

I'll make some changes to the code and write up some notes if that helps, but this has made me think a more generic approach might be much better: another python library I wrote to output to OLED devices (https://github.com/rm-hull/ssd1306) just relies on Pillow (python imaging library) - it exposes a canvas onto which you can draw text, images, lines, circles, etc. The canvas is then rendered onto the device.

We could adapt a similar approach for the max7219 devices, but set up a viewport on the canvas so that only an 8x8 square or 40x16 rectangle (if you have 5x2 matrices) is rendered. Scrolling can be implemented by moving the viewport around the canvas, but you would get all the drawing and text handling primitives of Pillow for free!

mB-PiBox, 2015-08-28:
Right now I'm using the following code to output the time to a 4 8x8(64LED) Matrices.

clock = time.strftime("%H%M")
for n, c in enumerate(clock): 
		device.letter(n, ord(c))

trying to get the number to scroll in from the top, when the min/hr changes.
Would the use of Pillow allow for this effect?

rm-hull, 2015-08-28:
With the pillow version, you would write something like:

```python
clock = time.strftime("%H%M")

# Pillow ops to draw the time onto a draw context
img = Image.new(...)
fnt = ImageFont.truetype(...)
dc = ImageDraw.Draw(img)
dc.text((0,0), clock, font=fnt, fill=(255,255,255,255))

# Scroll the time up
for y in range(7, -1, -1): 
  offset = (0, y)  
  device.render(img, offset)    # Anticipated new method in max7219 code
  time.sleep(0.1)

# Scroll the time to the right
for x in range(8 * 4): 
  offset = (x, 0)  
  device.render(img, offset)
  time.sleep(0.1)
```

Where there is `...` this would need to be fleshed out with a suitable image type/size/font name. 

The new `render` method on the device class would be responsible for reading the pixels from the image at the given offset and for the size of the cascaded matrices and write them out to the devices.

briceparent, 2016-03-05:
If I understood well the problem here, I've created a pull request #22 to allow this by adding a vertical argument to led.matrix. Ex : 
`device = led.matrix(cascaded=2, vertical=True)`
At least, it works for me. I tested with 2 matrix only, it's all I have, but it should work with more.

rm-hull, 2016-03-05:
@mB-PiBox I've merged the change provided by @briceparent - if this doesn't solve your problem feel free to reopen this issue

mB-PiBox, 2016-03-05:
@rm-hull Unfortunately that is not what I was referring to; what @briceparent is for if the matrices are stacked on top of each other (vertical), instead of next to one another (horizontal.)

I made a small gif of what I was trying to do, but have not been able to yet.
[image]

briceparent, 2016-03-05:
Ok! That's not at all what I understood, I need to practice my english...
What I made was to allow the matrix to be placed with the headers on the left and right, instead of on the top and bottom (as some matrix are sold grouped like this). But I like this idea, I'd take a look at it when I've got time !

mB-PiBox, 2016-03-05:
@briceparent thank you for trying. I'm still learning Python, basic stuff I got, but this stuff is still a little over my head. I play around with it, every couple days, trying to get it to work, but have not had any success.

rm-hull, 2017-01-19:
v0.3.0 now features a viewport component which allows exactly this:

[image]

I will add a demo that does exactly that

closed 2017-01-19 by rm-hull