rm-hull/luma.led_matrix #20: max7219.led module not found

Nick-Hofmann, 2015-11-29:
After running:
sudo python max7219/examples/matrix_test/py                     I get the error:
Traceback (most recent call last):
        File "max7219/examples/matrix_test.oy", line 3, in <module>
        import max7219.led as led
ImportError: No module named max7219.led

There were no errors when cloning from the repo. I have installed wiringPi, and I have also changed the SPI settings, just like in the README.txt

Any help would be greatly appreciated.

rm-hull, 2015-11-29:
Hi, are you using raspian?

If so, did you do the following at the command line?

    $ sudo apt-get install python-dev python-pip
    $ sudo pip install spidev
    $ sudo python setup.py install

Try that, the output from the last command should be something like:

```
running install
running build
running build_py
running install_lib
running install_egg_info
Removing /usr/local/lib/python2.7/dist-packages/max7219-0.1.4.egg-info
Writing /usr/local/lib/python2.7/dist-packages/max7219-0.1.4.egg-info
```

Then try the test program again, but maybe change directory first :

    $ cd max7219
    $ sudo python examples/matrix_test.py

Let me know how you get on

Richard

OranJuiceJohnes, 2015-12-25:
running raspian jessie ... same problem ... and yes, the install seems to fail ...

root@orange42 ~ # python max7219/setup.py install
running install
running build
running build_py
package init file 'max7219/__init__.py' not found (or not a regular file)
copying max7219/setup.py -> build/lib.linux-armv7l-2.7/max7219
package init file 'max7219/__init__.py' not found (or not a regular file)
running install_lib
copying build/lib.linux-armv7l-2.7/max7219/setup.py -> /usr/local/lib/python2.7/dist-packages/max7219
byte-compiling /usr/local/lib/python2.7/dist-packages/max7219/setup.py to setup.pyc
running install_egg_info
Removing /usr/local/lib/python2.7/dist-packages/max7219-0.1.6.egg-info
Writing /usr/local/lib/python2.7/dist-packages/max7219-0.1.6.egg-info

rm-hull, 2015-12-25:
Yes, you have to be in the directory for the install to work... maybe i should update the readme to make it clearer

closed 2016-03-05 by rm-hull

aocana, 2017-04-07:
hi, 
i have the exactly same issue:

`user@raspberrypi-WEATHER:~/scripts/panel-led/max7219$ sudo python setup.py install
running install
running bdist_egg
running egg_info
creating luma.led_matrix.egg-info
writing requirements to luma.led_matrix.egg-info/requires.txt
writing luma.led_matrix.egg-info/PKG-INFO
writing namespace_packages to luma.led_matrix.egg-info/namespace_packages.txt
writing top-level names to luma.led_matrix.egg-info/top_level.txt
writing dependency_links to luma.led_matrix.egg-info/dependency_links.txt
writing manifest file 'luma.led_matrix.egg-info/SOURCES.txt'
reading manifest file 'luma.led_matrix.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
no previously-included directories found matching 'doc/_build'
warning: no previously-included files matching '__pycache__' found under directory '*'
warning: no previously-included files matching '*.py[co]' found under directory '*'
warning: no previously-included files matching '*~' found under directory '*'
warning: no previously-included files matching '.coverage' found under directory '*'
writing manifest file 'luma.led_matrix.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv6l/egg
running install_lib
running build_py
creating build
creating build/lib.linux-armv6l-2.7
creating build/lib.linux-armv6l-2.7/luma
creating build/lib.linux-armv6l-2.7/luma/led_matrix
copying luma/led_matrix/virtual.py -> build/lib.linux-armv6l-2.7/luma/led_matrix
copying luma/led_matrix/device.py -> build/lib.linux-armv6l-2.7/luma/led_matrix
copying luma/led_matrix/segment_mapper.py -> build/lib.linux-armv6l-2.7/luma/led_matrix
copying luma/led_matrix/helpers.py -> build/lib.linux-armv6l-2.7/luma/led_matrix
copying luma/led_matrix/legacy.py -> build/lib.linux-armv6l-2.7/luma/led_matrix
copying luma/led_matrix/const.py -> build/lib.linux-armv6l-2.7/luma/led_matrix
copying luma/led_matrix/__init__.py -> build/lib.linux-armv6l-2.7/luma/led_matrix
creating build/bdist.linux-armv6l
creating build/bdist.linux-armv6l/egg
creating build/bdist.linux-armv6l/egg/luma
creating build/bdist.linux-armv6l/egg/luma/led_matrix
copying build/lib.linux-armv6l-2.7/luma/led_matrix/virtual.py -> build/bdist.linux-armv6l/egg/luma/led_matrix
copying build/lib.linux-armv6l-2.7/luma/led_matrix/device.py -> build/bdist.linux-armv6l/egg/luma/led_matrix
copying build/lib.linux-armv6l-2.7/luma/led_matrix/segment_mapper.py -> build/bdist.linux-armv6l/egg/luma/led_matrix
copying build/lib.linux-armv6l-2.7/luma/led_matrix/helpers.py -> build/bdist.linux-armv6l/egg/luma/led_matrix
copying build/lib.linux-armv6l-2.7/luma/led_matrix/legacy.py -> build/bdist.linux-armv6l/egg/luma/led_matrix
copying build/lib.linux-armv6l-2.7/luma/led_matrix/const.py -> build/bdist.linux-armv6l/egg/luma/led_matrix
copying build/lib.linux-armv6l-2.7/luma/led_matrix/__init__.py -> build/bdist.linux-armv6l/egg/luma/led_matrix
byte-compiling build/bdist.linux-armv6l/egg/luma/led_matrix/virtual.py to virtual.pyc
byte-compiling build/bdist.linux-armv6l/egg/luma/led_matrix/device.py to device.pyc
byte-compiling build/bdist.linux-armv6l/egg/luma/led_matrix/segment_mapper.py to segment_mapper.pyc
byte-compiling build/bdist.linux-armv6l/egg/luma/led_matrix/helpers.py to helpers.pyc
byte-compiling build/bdist.linux-armv6l/egg/luma/led_matrix/legacy.py to legacy.pyc
byte-compiling build/bdist.linux-armv6l/egg/luma/led_matrix/const.py to const.pyc
byte-compiling build/bdist.linux-armv6l/egg/luma/led_matrix/__init__.py to __init__.pyc
creating build/bdist.linux-armv6l/egg/EGG-INFO
copying luma.led_matrix.egg-info/PKG-INFO -> build/bdist.linux-armv6l/egg/EGG-INFO
copying luma.led_matrix.egg-info/SOURCES.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
copying luma.led_matrix.egg-info/dependency_links.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
copying luma.led_matrix.egg-info/namespace_packages.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
copying luma.led_matrix.egg-info/not-zip-safe -> build/bdist.linux-armv6l/egg/EGG-INFO
copying luma.led_matrix.egg-info/requires.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
copying luma.led_matrix.egg-info/top_level.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
creating dist
creating 'dist/luma.led_matrix-0.9.0-py2.7.egg' and adding 'build/bdist.linux-armv6l/egg' to it
removing 'build/bdist.linux-armv6l/egg' (and everything under it)
Processing luma.led_matrix-0.9.0-py2.7.egg
removing '/usr/local/lib/python2.7/dist-packages/luma.led_matrix-0.9.0-py2.7.egg' (and everything under it)
creating /usr/local/lib/python2.7/dist-packages/luma.led_matrix-0.9.0-py2.7.egg
Extracting luma.led_matrix-0.9.0-py2.7.egg to /usr/local/lib/python2.7/dist-packages
luma.led-matrix 0.9.0 is already the active version in easy-install.pth
[... 55 lines cut ...]

this is the code into the python file:

`#!/usr/bin/env python

import os
import max7219.led as led

def getCPUtemperature():
	res = os.popen("vcgencmd measure_temp").readline()
	return(res.replace("temp=","").replace("'C\n",""))

device = led.matrix(2)
device.orientation(270)

try:
	while True:
		device.show_message("CPU temperature: " + getCPUtemperature() + "C");
except KeyboardInterrupt:
	pass
finally:
	device.clear()
`

any help..?

JohannesRegner, 2017-04-14:
@aocana i have the same problem... See ticket #104

rm-hull, 2017-04-14:
@JohannesRegner - The library has been completely rewritten and it has a different namespace now. The classes are quite different but there is a nod to some legacy methods.

Have a look at the [documentation](http://luma-led-matrix.readthedocs.io/en/latest/python-usage.html) and [examples](https://github.com/rm-hull/luma.led_matrix/blob/master/examples/matrix_demo.py).