rm-hull/luma.led_matrix #24: its not an issue, but a doubt

m-devan, 2016-05-13:
how to use specify set of digits for different purpose?
for example 
first 3 digits for high score
next 2 digits for timer
last 3 digits for current score

thanks in advance for your help
dev

rm-hull, 2016-05-13:
You should be able to call the `letter` function to write a digit at a given position, see https://github.com/rm-hull/max7219/blob/master/max7219/led.py#L247-L255

m-devan, 2016-05-13:
thanks for the quick response. 
I want to display a number at a given position

rm-hull, 2016-05-13:
If you look at the date example: https://github.com/rm-hull/max7219/blob/master/max7219/led.py#L247-L255, you can see how it uses the `letter` function how it and splits out writing the individual digits from the day, month and year - you should be able to do the same sort of thing.

m-devan, 2016-05-13:
thanks a lot

closed 2016-05-13 by m-devan

rm-hull, 2016-05-13:
Sorry - I posted the wrong link earlier - https://github.com/rm-hull/max7219/blob/master/examples/sevensegment_test.py#L15-L25 - this is what I meant

```python
    # Set day
    device.letter(deviceId, 8, int(day / 10))     # Tens
    device.letter(deviceId, 7, day % 10)          # Ones
    device.letter(deviceId, 6, '-')               # dash
    # Set day
    device.letter(deviceId, 5, int(month / 10))     # Tens
    device.letter(deviceId, 4, month % 10)     # Ones
    device.letter(deviceId, 3, '-')               # dash
    # Set day
    device.letter(deviceId, 2, int(year / 10))     # Tens
    device.letter(deviceId, 1, year % 10)     # Ones
```