rm-hull/luma.led_matrix #25: device = led.matrix(cascaded=1)   File "/usr/local/lib/python2.7/dist-packages/max7219/led.py", line 51, in __init__     self._spi.open(spi_bus, spi_device) IOError: [Errno 2] No such file or directory

tebe4rt, 2016-06-09:
pi@raspberrypi:~/max7219 $ sudo python examples/matrix_test.py
Traceback (most recent call last):
  File "examples/matrix_test.py", line 8, in <module>
    device = led.matrix(cascaded=1)
  File "/usr/local/lib/python2.7/dist-packages/max7219/led.py", line 51, in __init__
    self._spi.open(spi_bus, spi_device)
IOError: [Errno 2] No such file or directory

rm-hull, 2016-06-09:
If you look at the [pre-requisites](https://github.com/rm-hull/max7219#pre-requisites) in the README, and copy the commands, what output do you get?

Are you on a gen3 RPI ?

arnrou, 2016-07-08:
Hi all,

I've the same error, I'm on a gen3 RPI. How can I resolve this ?
I have no output when i write $ dmesg | grep spi
and no files find with ls -l /dev/spi*

Thank you

rm-hull, 2016-07-08:
If you have no /dev/spi files and nothing is showing using `dmesg` then this implies the kernel SPI drivers are not loaded. The link in the prerequisites is out of date (I will look to update that shortly), but if you look at https://learn.sparkfun.com/tutorials/raspberry-pi-spi-and-i2c-tutorial#spi-on-pi:

1. Run `sudo raspi-config .`
2. Use the down arrow to select _9 Advanced Options_
3. Arrow down to _A6 SPI._
4. Select **yes** when it asks you to enable SPI,
5. Also select **yes** when it asks about automatically loading the kernel module.
6. Use the right arrow to select the **&lt;Finish&gt;** button.
7. Select **yes** when it asks to reboot.

[image]

thijstriemstra, 2016-10-04:
I have a similar problem using the latest kernel, `dmesg | grep spi` doesn't show any output, but `ls -l /dev/spi*` shows the devices are available.

rm-hull, 2016-10-04:
@thijstriemstra :
* what is the exact output of `ls -l /dev/spi*` ?
* what does `lsmod` output ?
* which type of raspberry pi are you using ?

thijstriemstra, 2016-10-04:
@rm-hull Enabled spi in `/boot/config.txt`:

```
dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=spi=on
```

`lsmod` shows:

```
$ lsmod
Module                  Size  Used by
bnep                   10340  2 
hci_uart               17943  1 
btbcm                   5929  1 hci_uart
bluetooth             326105  22 bnep,btbcm,hci_uart
cpufreq_stats           3463  0 
brcmfmac              186339  0 
brcmutil                5661  1 brcmfmac
cfg80211              427855  1 brcmfmac
joydev                  9024  0 
snd_soc_bcm2835_i2s     6354  0 
rfkill                 16037  3 cfg80211,bluetooth
snd_soc_core          125885  1 snd_soc_bcm2835_i2s
snd_bcm2835            20447  0 
snd_pcm_dmaengine       3391  1 snd_soc_core
snd_pcm                75762  3 snd_bcm2835,snd_soc_core,snd_pcm_dmaengine
snd_timer              19288  1 snd_pcm
snd                    51908  4 snd_bcm2835,snd_soc_core,snd_timer,snd_pcm
i2c_bcm2708             4834  0 
bcm2835_gpiomem         3040  0 
spi_bcm2835             6678  0 
bcm2835_wdt             3225  0 
evdev                  11396  0 
uio_pdrv_genirq         3164  0 
uio                     8000  1 uio_pdrv_genirq
ipv6                  347530  34
```

This is on a RaspberryPi 3 but it's using https://github.com/FooDeas/raspberrypi-ua-netinst so the issue might be there (also see thread at https://github.com/FooDeas/raspberrypi-ua-netinst/issues/21).

I also tested on an older RaspberryPi B+ v1.2 (report [here](https://github.com/FooDeas/raspberrypi-ua-netinst/issues/21#issuecomment-251520605)) and that worked fine (thank god). I think something's funky in the latest raspberry kernel or whatever.

rm-hull, 2016-10-04:
Hmmm... i've tried the code on every Pi variant except RPi 3.
 
The Pi Zero I've got plugged in right now (on a recent-_ish_ jessie lite / kernel 4.4.21+) also shows nothing on `dmesg | grep -i spi` but there are /dev/spi* entries, but it does work.

And I just did an `rpi-update` to kernel 4.4.23+ and that seems to work as well

thijstriemstra, 2016-10-04:
> The Pi Zero I've got plugged in right now (on a recent-ish jessie lite / kernel 4.4.21+) also shows nothing on dmesg | grep -i spi

Perhaps that should go into the readme, I found it confusing there was no output at all.

closed 2017-01-19 by rm-hull

viktor6, 2018-01-25:
What's wrong with the script?
`pi@raspberrypi:~ $ python timedate.py
^CTraceback (most recent call last):
  File "timedate.py", line 21, in <module>
    device.letter(n, ord(c))
  File "/usr/local/lib/python2.7/dist-packages/max7219/led.py", line 413, in letter
    self.flush()
  File "/usr/local/lib/python2.7/dist-packages/max7219/led.py", line 121, in flush
    buf = self._preprocess_buffer(list(self._buffer))
  File "/usr/local/lib/python2.7/dist-packages/max7219/led.py", line 505, in _preprocess_buffer
    buf = self._rotate(buf)
  File "/usr/local/lib/python2.7/dist-packages/max7219/led.py", line 490, in _rotate
    tile = rotate(tile)
  File "/usr/local/lib/python2.7/dist-packages/max7219/rotate8x8.py", line 39, in rotate
    low |= ltab[i][value & 0x0f]
KeyboardInterrupt
pi@raspberrypi:~ $
`

rm-hull, 2018-01-25:
@viktor6 You are using the old version of the library (when it was called max7219), which is no longer supported.

viktor6, 2018-01-25:
You mean that I use the old library in the script?