rm-hull/luma.led_matrix #26: AssertionError: Invalid deviceId: 1

sav25, 2016-06-09:
Hi

I'm trying to make my own version of the 7-segment modules. The wiring all checks out, however I'm only using 1 4-character 7-segment display - not 2 like the modules have.

I'm trying the **sevensegment_test.py** file (which works fine on the generic module) and setting the device number to 1 in '**device = led.sevensegment(cascaded=1)**', however I'm getting the following error:

**AssertionError: Invalid deviceId: 1**

I have tried using 0 as a guess, but it then gives me:

**AssertionError: Must have at least one device!**

I then tried entering 2, which runs the script, but the display shows nothing.

I wanted to check that i'm not doing something wrong with the code, as the wiring looks perfect.

Thanks

rm-hull, 2016-06-09:
Can you paste the stack trace?

initialising with cascaded=1 should be OK, but the example as is will be trying to write out data to the 2nd and 3rd devices, so you probably need to knock out the code that does that.

Have you got a link/info for the 4-character 7-segment display you are using? Just curious really...

sav25, 2016-06-10:
Sorry, I'm not the best at this stuff - I've Googled what a stack trace is - but I have no idea how to get you one from what I'm doing here.

I'm currently using one of these (it's common cathode but the sheet is labelled anode): http://www.lumex.com/content/files/ProductAttachment/LDQ-M516RI.pdf

I'll have a look at removing the code, thanks. Any other pointers would be great.

sav25, 2016-06-10:
Alternatively, do you know what the wiring of these modules is so that I can check it matches what I'm trying to do?

I'm 95% sure my wiring is correct, with the segment pins going to the right segments of the display etc.  followed this circuit diagram to kick off: http://embedded-lab.com/blog/wp-content/uploads/2012/12/SPI7SEGDISP8_56Circuit.jpg

sav25, 2016-06-10:
Oddly enough I have the lower segment in the second character lit as soon as I fire up the Pi as well, which doesn't happen on the pre-made modules. Something's definitely not right.

rm-hull, 2016-06-10:
> Sorry, I'm not the best at this stuff - I've Googled what a stack trace is - but I have no idea how to get you one from what I'm doing here.

Can you paste the full output when the assertion error is printed?

> do you know what the wiring of these modules is so that I can check it matches what I'm trying to do?

Not really, the modules are on a PCB and I've glued them in place, so I cant really check. There may be a wiring digram online somewhere. However, looking at the 4-character module you have, it is different in as much as it has inputs for the middle 'colon' on pins D5 and D6 whereas the 8 character units do not

> Oddly enough I have the lower segment in the second character lit as soon as I fire up the Pi

I've observed something similar, and have just put it down to a noisy power spike when powering up

sav25, 2016-06-10:
There you go:

pi@raspberrypi:~/rmhull $ sudo python sevensegment_test.py
Traceback (most recent call last):
  File "sevensegment_test.py", line 48, in <module>
    date(device, 1)
  File "sevensegment_test.py", line 16, in date
    device.letter(deviceId, 8, int(day / 10))     # Tens
  File "/home/pi/rmhull/max7219/led.py", line 255, in letter
    self.set_byte(deviceId, position, value, redraw)
  File "/home/pi/rmhull/max7219/led.py", line 152, in set_byte
    assert 0 <= deviceId < self._cascaded, "Invalid deviceId: {0}".format(deviceId)
AssertionError: Invalid deviceId: 1

rm-hull, 2016-06-10:
So if you comment out https://github.com/rm-hull/max7219/blob/master/examples/sevensegment_test.py#L48 when **cascaded=1** then you should at least get the time displaying

sav25, 2016-06-10:
Interesting change of events - I used a multimeter to track the traces that these modules use, and I now have it working exactly as the module does.

However, the example program now gets to the section where the display shows '00000407' after the fast counter section) and then kicks out the program with the following:

pi@raspberrypi:~/rmhull $ sudo python sevensegment_test.py
Traceback (most recent call last):
  File "sevensegment_test.py", line 93, in <module>
    device.write_number(deviceId=2, value=c, decimalPlaces=2)
  File "/home/pi/rmhull/max7219/led.py", line 264, in write_number
    assert 0 <= deviceId < self._cascaded, "Invalid deviceId: {0}".format(deviceId)
AssertionError: Invalid deviceId: 2

Can you help?

Thank you!

rm-hull, 2016-06-13:
Thats good to hear!

So if you read the traceback (a.k.a stack trace), the error is at line 93 of _sevensegment_test.py_ when it tries to call:

```python
device.write_number(deviceId=2, value=c, decimalPlaces=2)
```

the deviceId is invalid - you must've set the cascaded= flag to something less than 2; if you only have one device, then comment out the code where you see any calls the module with an argument of `deviceId=1` or `device=2`

sav25, 2016-06-13:
Just tried it, spot on!

Thanks again for your help, I think this is all sorted now. Cheers!

closed 2016-06-13 by rm-hull