rm-hull/luma.led_matrix #28: One Module of four matrices (8x8) all showing the same

ozett, 2016-08-19:
hi,
i have an already cascaded module, where 4 pieces of 8x8 matrix-modules are bound together with the pcb on the back.

if i run the test with cascade=1, they 4 modules (or should i say the 1 module with 4 matrices) show all the same on there four 8x8 segments.

if i run the test with cascade=2, the 4 modules show all the same, with one colum later on the next module.

how do i set orientation (or something else?), that the scrolling goes over the one module (which consists out of four single 8x8) modules from left to right/right to left  ---
and not showing all the same?

thanks for helping out. i found your python very helpful, 
your explantions get me starting in a minute. great!  :smile: 
:+1:

rm-hull, 2016-08-19:
Did you try `cascade=4` ?

ozett, 2016-08-20:
yes, thanks for reply.
i first tried this. and after your question i checked this again in the code, 
and it is still `cascade=4.`

i made a animated gif from video to show what it looks like :eyes: 

i think the order sequence of each 8x8 module must be different,
or some kind of overall matrix-rotation can change this.

very appreciate your help, as you are the "Father of the Code" :family:  :smile: 

[image]

[image]

ozett, 2016-08-20:
uhhh, i think i found "my" mistake! in understanding how to set orientation...
i will follow to explain it with an image ..

ozett, 2016-08-20:
i changed the orientation at the beginning to globally 90 degree rotated,
which works for the first part of the demo all right, but after a while it was scrammbled again.

i overlook in the demo-code, where further down it was hardcoded to set the orientation fix to 0 degree. 
this puzzled me while i was observing the module-animations...

[image]

ozett, 2016-08-20:
the hardcoded orientation-degree in the demo-code happend after the orientation demo.
there it was set back to 0 degree, which was not that what my module needed (90°) and what i changed only at the beginning of the demo code. 

it looked like this:

[image]

now its running fine. thanks a lot for helping into the right direction :arrow_forward:

ozett, 2016-08-20:
:smile: 

[image]

rm-hull, 2016-08-20:
Fantastic - I'd only ever tried it with two matrixes (I have 3x of the seven-segment displays), so it is good to see it working with 4 matrixes.

Btw - what software did you use to produce the animated gifs?

ozett, 2016-08-21:
:+1: 
and for the gif i tried ezgif.com online, and was faszinated
:+1:

closed 2016-08-29 by rm-hull

ozett, 2016-10-03:
i have it now running on esp8266, and the max71 lib together with the adafruit gfx-lib allowed to set the orientation indiviually for each 8x8 segment within my strip with hardsoldered four of them.

just to give an impression, it is a running text, but segement 2 and 4 are rotated 90 degree..
cool, eh?  🆒 

[image]

OURLEDS, 2019-01-31:
The problem here is that is the following things to have in consideration:
Python or arduino

For Arduino users and MAX72xx library or MD parole here are the solutions taht works with the newest hadware (leds modules):

Just open the example, and if it is not working or mirrored ... just change the headers for one of this:
#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
#define HARDWARE_TYPE MD_MAX72XX::GENERIC_HW
#define HARDWARE_TYPE MD_MAX72XX::ICSTATION_HW

You can just change it and compile with one of those lines each time ... or just take unplug one 8x8 module and check the name on the board below the chip .... if it says FC16 is FC16 ... if it is 1088AS like mine it should work with the ICSTATION﻿

If youa re using python no f.ck idea.
But for arduino don't get confused with downlaoding files and changingin the src files.h
That is a VERY VERY VERY OUTDATED solution ...

THIS SOLUTION WORKS FOR DEVICES BOUGHTS in 2019 and using arduino 1.8.8 with the MAX72xx library of Oct 2018 version 3.0.2

If arduino is not 1.8.8 you bought the device long time ago or it is not a max7219 or if the max72xx library is not version 3.0.2 ... it probably won't work this simple change of headers ...

All my day lost getting this work ... the job here is done ... Now I can fly away...

jarleven, 2021-12-02:
For mirrored of flipped modules you can try the patch below.
Inspiration from : https://note.nkmk.me/en/python-pillow-flip-mirror/

```bash
sudo -H pip install --upgrade --ignore-installed Pillow
```
```
from PIL import Image, ImageOps
image = ImageOps.flip(image)
```
```diff
--- /usr/local/lib/python3.7/dist-packages/luma/led_matrix/device.py	2021-12-02 18:51:39.150976657 +0000
+++ ./device.py	2021-12-02 18:51:08.871460758 +0000
@@ -40,6 +40,7 @@
 from luma.core.util import observable
 from luma.core.virtual import sevensegment
 from luma.led_matrix.segment_mapper import dot_muncher, regular
+from PIL import Image, ImageOps
 
 
 __all__ = ["max7219", "ws2812", "neopixel", "neosegment", "apa102", "unicornhathd"]
@@ -121,6 +122,7 @@
         assert(image.mode == self.mode)
         assert(image.size == self.size)
 
+        image = ImageOps.flip(image)
         image = self.preprocess(image)
 
         i = 0
```