rm-hull/luma.led_matrix #29: Showing text without scrolling

bauerj, 2016-09-02:
Is there currently a way to show some string centered (without scrolling) if it fits on the displays? It might be useful for showing things like the current time or temperature where you don't want the text to scroll.

If not, I could send you a Pull Request if you like.

rm-hull, 2016-09-03:
Which type of display are you talking about? The 8x8 matrixes or the 7-segment displays?

If you mean the 7-segment displays, then you could use the `letter` function to position specific letters at different digit places (and it would be quite easy to wrap that up into a `write_text` function that manages left/right/centre justification). From memory, I think the position is counted from the right most digit.

The matrix version would be a tiny bit more complicated as you would need to take account of the font widths.

I'm away for a few days, so by all means go for a PR (try and stick to the PEP8 style guide and please add a docstring to any public methods).

bauerj, 2016-09-03:
> Which type of display are you talking about? The 8x8 matrixes or the 7-segment displays?

Yes, I am talking about the 8x8 matrices. I'll send you a PR implementing it :+1:

PR #30 (2016-09-05): By default don't scroll text if it fits on the displays
This changes the behaviour of `matrix.show_message` to do the following:

1. Check if the rendered text is longer than all the displays combined
2. If it **is** longer:
    1.  Proceed as usual
3. If it's not longer:
    1. Calculate the margin needed to center the text
    2. Display the text with the margin as offset and flush immediately.

It looks like this: [image]

Closes #29
closed 2016-09-05 by rm-hull