rm-hull/luma.led_matrix #39: Issue with "scrolling" code, but "scrolling alphabet" works fine

someevil, 2016-10-18:
Hey, 

I am running a Raspberry Pi 3, and using this seven segment device with this code nicely. 

I just tried to implement the scrolling text code as described, but nothing happens at all. The code does not provide an error when executed, the display shows nothing, and command line simply returns to waiting for its next command.
```
# Scrolling
print('Scrolling...')
for x in range(2):
    for _ in range(8):
        device.scroll_right()
        time.sleep(0.1)
    time.sleep(1)
device.clear()
```

I then tried the scrolling alphabet code as below, it works perfectly:
```
# Scrolling Alphabet Text
print('Scrolling alphabet text...')
device.show_message("HELLO EVERYONE!")
device.show_message("0123456789 abcdefghijklmnopqrstuvwxyz")
```

I'm just wondering if anyone else is having troubles with the scrolling... section?

#### Linux Kernel version
Linux pi3 4.4.21-v7+ #911 SMP Thu Sep 15 14:22:38 BST 2016 armv7l GNU/Linux

rm-hull, 2016-10-18:
Do you have a single 7-segment unit? If so, make sure that you initialise with `cascaded=1` - any other value will place the text in a buffer which will not be displayed unless you scroll enough.

Also in the 1st snippet, presumably you write out some text? In fact, that snippet looks very much like https://github.com/AverageManVsPi/ZeroSeg/blob/master/examples/scrolling_example.py which is a little bit wrong if there's only one device.

How about:

```python
device = led.sevensegment(cascaded=1)

while True:
    device.write_text(deviceId=0, text="HELLO")
    for _ in range(8):
        device.scroll_right()
        time.sleep(0.1)
    
    time.sleep(1)
    device.clear() 
```

someevil, 2016-10-18:
Hey, 
Yes it is the seven segment, sorry for not indicating this in my initial comment.

I have actually copied the code from your file https://github.com/rm-hull/max7219/blob/master/examples/sevensegment_test.py and have been modifying each of the different display types just to learn a bit more about the display.

I will try the code you suggested and report back asap.

On a side note, can you please explain what the 'cascaded' item is referring to? I have mine set = 3 but unsure if/when/why I would change it?

I have the same question with regards to the DeviceID, mine is set to 2 and it runs, I note yours is = 0

Thanks heaps, it's really appreciated.

rm-hull, 2016-10-18:
'cascaded' is the number of devices that are daisy chained together, so if you only have one, the it should be = 1 and the deviceId should alway be 0 <= deviceId < cascaded. This is a zero-based index for each daisy chained device, and if you only have one, it should be = 0.

I have 3, hence the reason they are set in the examples; maybe I should update the code to assume that most folks have one.

If you only have one device, and set cascaded=3, and deviceId=2 then this is the 'last' device so you will see the output for that device. That is true for any cascaded=n, deviceId=n-1, where n>0 !

Its like this because of the way the data is serially shifted off the MOSI pin on the Raspberry Pi and clocked into the DIN (data-in) pin on the max7219 chip.

I'll make a note to tidy up the examples so they aren't ambiguous, and will expand on the relationship between cascaded and the deviceId in the README & docstrings.

thijstriemstra, 2016-10-19:
@rm-hull that info would be a welcome addition to the docs..

thijstriemstra, 2016-11-22:
@someevil what version of Python are you using?

closed 2017-01-19 by rm-hull