rm-hull/luma.led_matrix #41: Usage of time.sleep in show_message blocks app

thijstriemstra, 2016-10-27:
`show_message` has a coupe of calls to `time.sleep`, causing my PyQt5 app to freeze. Can an additional write text method be introduced that can be used to write text, that doesn't invoke `time.sleep`?

Using python 3.5.2.

thijstriemstra, 2016-10-27:
Basically something like http://stackoverflow.com/questions/32808893/how-to-use-async-await-in-python-3-5, available in Python 3.5 and newer.

rm-hull, 2016-10-27:
I used asyncio in [zaup](https://github.com/rm-hull/zaup/blob/master/zaup.py) which uses the max7219 library. I am hesitant of introducing asyncio into the library, only because I would still like it to be supported/usable in Python 2.7 still. 

Having said that, I guess we could introduce a basic asyncio fallback shim that relies on `time.sleep` if used on python 2.7

thijstriemstra, 2016-10-27:
@rm-hull yea sounds good. I'm not too familiar yet with the new async/await constructs but I think we need a separate module for these definitions and import them on py3.5 and newer to prevent syntax errors on older py versions.

rm-hull, 2016-10-27:
I created a PR (#42) with an implementation which works "as is" on both Python 2 and 3 - and is async friendly on Python 3 if you supply an asyncio event loop - see the async examples.

See what you think of the code - there are a couple of examples in the PR - if you let me know if this suits, and I will roll it into a new version.

rm-hull, 2016-11-13:
@thijstriemstra - did you get a chance to try out the async code changes ?

thijstriemstra, 2016-11-13:
I'll give it a try tonight and get back to you @rm-hull

thijstriemstra, 2016-11-22:
Here's a Pyqt test app that blocks with the `time.sleep` example: https://gist.github.com/thijstriemstra/fe24d55e5432a9a40ab12662f3a1a226

Will try with new example now.

thijstriemstra, 2016-11-22:
Check https://gist.github.com/thijstriemstra/c75830a9f30c1000e31bbf4a0940caaa

This uses an asyncio-compatible event loop for Qt and it doesn't block (the slider still works while counter is running). It doesn't use your PR though so I'm not sure if your code changes are even needed if I can workaround it like this. Thoughts?

rm-hull, 2016-11-24:
The only _blocking_ method in the max7219 lib is the `show_message` which will do a `time.sleep(..)` to scroll the message across the display if there are not enough digits to display the entire message. Your 2nd example uses `write_text(..)` (https://gist.github.com/thijstriemstra/c75830a9f30c1000e31bbf4a0940caaa#file-async-pyqt-sevensegment-py-L51) which doesnt block. 

The 1st example should probably be rewritten so instead of https://gist.github.com/thijstriemstra/fe24d55e5432a9a40ab12662f3a1a226#file-pyqt_sevensegment_test-py-L49-L53, it should employ the event loop to either `await asyncio.sleep(..)` or `loop.call_later(..)`

_[The PR allows you to supply the event loop such that `show_message` wont block]_

rm-hull, 2017-01-19:
Closing issue, as scrolling should be handled by manipulating viewport.set_position(...) to simulate scroll behaviour

commit 56481308 (2016-10-27): Async scrolling (for Python 3) - fixes #41
closed 2017-01-19 by rm-hull