rm-hull/luma.led_matrix #50: Not an issue: max7219 Python Webserver!

fullphat, 2017-01-13:
I'm sure I'm not the first, but I've written a Python-based web server that communicates with a max7219 8x8 LED Matrix connected to a Raspberry Pi.  My test system is a Pi 3 running Kali Linux, but I'm sure it'll work on most other installation.  It requires Python 2.7.

You can download it from SourceForge here: 

https://sourceforge.net/projects/snarlwin/files/Extras/RedPowder/

and there's more information about it here:

http://snarl.fullphat.net/max7219-led-array-on-raspberry-pi/

It's also usable from Snarl (our enterprise notification platform).

rm-hull, 2017-01-13:
Good stuff - its great that you managed to get it working on Kali Linux (I will check that distro out).

It would be fantastic if you would be prepared to help by adding install instructions for Kali in the documentation in github - ultimately, it gets automatically published here: https://max7219.readthedocs.io/en/latest/

I'm planning some significant new features in the near future, and there will be a project rename as part of that (to **luma.led_matrix**), as well as a major documentation refresh/rewrite.

Any issues or problems, please feel free to raise them, no matter how trivial - as they help others who may be facing the same issues.

fullphat, 2017-01-13:
Sure - happy to help out!

I've only tried the basics at the moment, but I am hoping to do more with it.  I wrote up a how-to guide in my blog so I can pretty much transfer that over to GitHub.

Good job on the API - it works a treat! 

Actually, there is one thing I noticed - I made the web server create a new thread when it talks to the device so the source doesn't have to wait for the scrolling message to finish before receiving the HTML reply back - this generates a side effect of what are effectively multiple messages overlaid on each other if I send two requests simultaneously.  Is it possible to test if the device is busy scrolling a message?  And can the device be "reset" part way through a scrolling message?

Are there any other fonts available?  A chunky digital number one would be very handy!

Thanks   

Chris
full phat products
__________________________
http://getsnarl.info
Twitter: @snarlhappens

rm-hull, 2017-01-14:
Maybe rather than creating a new thread when it talks to the device, you should create a threadpool of size 1, and add tasks to that to do the scrolling - this would ensure any scrolling messages are played out sequentially rather than interleaving. Here is a pretty robust implementation based on queues: http://code.activestate.com/recipes/577187-python-thread-pool/

I'm planning to make some significant changes to the library, bringing it under the hood of the [luma.core](https://github.com/rm-hull/luma.core), so that you abstract away from the device itself, and instead draw on a [pillow](http://pillow.readthedocs.io/en/latest/reference/ImageDraw.html) drawing canvas instead.

This has the advantage that you can easily draw graphics primitives and text in any color using regular pillow ImageDraw methods, and the resulting canvas will be properly rendered onto the underlying device (whether that is a monochrome LCD, color OLED, or a bunch of  LED matrices driven by the max7219 chip). Pillow supports TTF font rendering, so if you can find a 'chunky digital number' TTF font, then it will be supported by the upcoming version of the max7219 library.

The other great thing [luma.core](https://github.com/rm-hull/luma.core) supports is the concept of a viewport - this  allows quite a bit of flexibilty in terms of scrolling (in any direction), as the example below:

[image]

See the source code: https://github.com/rm-hull/luma.examples/blob/master/examples/crawl.py - once the max7219 driver is written (shouldn't be too far off), then this code would work unchanged on a bunch of cascaded 8x8 matrices.

fullphat, 2017-01-15:
Sounds good!  I've also got a Unicorn pHat (which needs soldering) and a full Unicorn Hat on the way, so a generic driver for these would be great.

I'm not too hot on Python just yet, but I'll take a look at the thread pool stuff and see how it could work. 

Chris
full phat products
__________________________
http://getsnarl.info
Twitter: @snarlhappens

rm-hull, 2017-01-22:
So the library has now changed as I outlined in an earlier comment - it would be good if you could check it out and let me know any bugs or suggest improvements (particularly around better quality docs)

> I've also got a Unicorn pHat (which needs soldering) and a full Unicorn Hat on the way, so a generic driver for these would be great.

I just snagged a cheap pHat from ebay, so yes, expect a driver for that soon :)

closed 2017-01-22 by rm-hull

fullphat, 2017-01-22:
Will do!

I was browsing your other luma repos and off the back of that have ordered a PCD8544 LCD display, so I'm going to see if I can rework my web server code to handle different device types. :)

Chris
full phat products
__________________________
http://getsnarl.info
Twitter: @snarlhappens

fullphat, 2017-01-22:
Quick update...

Installation didn't go so well on my Kali Pi.  I ran the install on the raspbian device for comparison and there was a lot less debug output and it seemed to work.  I'm running it again on the Kali one, but this time with output to a file in case you want the full detail. :)

On a side note, I have my web server checking if the thread that's scrolling the message is still alive and, if it is, it simply drops further requests.  I'll expand on this by letting it add the request to a queue which the thread can pull from when it completes.

One really useful option would be a way to interrupt show_message() so high priority messages can be displayed immediately.

Chris
full phat products
__________________________
http://getsnarl.info
Twitter: @tweetsnarl

fullphat, 2017-01-22:
Pillow (or lack of it) seemed to be the cause of the issues.  I installed libjpeg-dev, zlib-dev and libtiff-dev and the build has now completed ok.  However, when I run matrix_demo.py I get:

root@blackpowder:~/luma.led_matrix/examples# python matrix_demo.py
Traceback (most recent call last):
  File "matrix_demo.py", line 8, in <module>
    from luma.led_matrix import legacy
ImportError: No module named luma.led_matrix

rm-hull, 2017-01-22:
Ah, ok, I'll update the docs for the pillow install.

To install the library, rather than cloning from github, can you try:

    $ sudo pip install --upgrade luma.led_matrix

fullphat, 2017-01-23:
Still the same issue I'm afraid.  No module named luma.led_matrix.  The pip install command above seemed to complete fine (but didn't seem to alter anything).

rm-hull, 2017-01-23:
What is the output of the following commands:

    $ pip --version

and

    $ pip list | grep luma

thijstriemstra, 2017-01-23:
First uninstall, then install!

fullphat, 2017-01-24:
Just as a by-the-by, I've been making good progress in various areas.  Using the Adafruit_Nokia_LCD library, I've been able to create a simplistic character-based text printer then even scrolls vertically when it reaches the end of the display (it wraps lines as well).  Next up will be to support proper pixel-level graphics, but it's enough for what I need right now.

On to RedPowder (the web server bridge), I've upgraded the API to /v2/ already and you can use "device=nnn" to access particular types of device.  The device handlers themselves are separate python modules that are dynamically loaded by the server itself.  Currently I support:

```
null (just prints to stdout for testing)
blink1
max7219
```

I'm going to try to add some abstraction as well, so:

`http://127.0.0.1/v2?device=max7219&text=Hello, world!`

Would send a scrolling "Hello, world!" message across the matrix display, while:

`http://127.0.0.1/v2?device=pcd8544&text=Hello, world!`

Would print it to the LCD display. :)