rm-hull/luma.led_matrix #51: Support for 4-digit display

thijstriemstra, 2017-01-15:
Recently got one of these [4 Bits TM1637 Digital Tube LED Clock](http://www.ebay.com/itm/310788450125) displays. Would be cool to support this in luma.led_matrix.

thijstriemstra, 2017-01-15:
I've ordered one for you @rm-hull if you want to play around with it.

twdkeule, 2017-01-19:
I already encoutered this while looking to buy one: [raspberrytips.nl](https://raspberrytips.nl/files/tm1637.py)

Also found this on [github](https://github.com/johnlr/raspberrypi-tm1637/blob/master/tm1637.py)

thijstriemstra, 2017-01-19:
Ok. It uses wiringPi2. Thanks for info though.

rm-hull, 2017-02-14:
@thijstriemstra - Package arrived today

thijstriemstra, 2017-02-15:
Yay nice :)

rm-hull, 2017-03-04:
I connected this up, but not getting anything displayed (connected to an RPi Zero) using the scripts from https://raspberrytips.nl/tm1637-4-digit-led-display-raspberry-pi/

Not sure if it's a dud or something specific to the pi zero.

thijstriemstra, 2017-03-14:
Here's an arduino reference library with a wiring table: https://github.com/bremme/arduino-tm1637

# Hardware setup

| TM1637 PIN | Arduino PIN      | Description         |
|------------|------------------|---------------------|
| CLK        |  Any digital pin | Clock               |
| DIO        |  Any digital pin | Digital output      |
| VCC        |  5V              | Supply voltage      |
| GND        |  GND             | Ground              |

And using https://github.com/johnlr/raspberrypi-tm1637 I couldn't get it to work with CLK on gpio 21 and DIO on 20.

thijstriemstra, 2017-03-14:
Maybe this is easy to port.. https://github.com/blueSolder/tm1637-Chip/blob/master/tm1637_Chip.py

thijstriemstra, 2017-03-14:
Also couldn't get https://raspberrytips.nl/tm1637-4-digit-led-display-raspberry-pi/ to work.

thijstriemstra, 2020-10-18:
any chance we can resurrect this @rm-hull? :) one of my most used projects uses this display and it feels weird to use a 'third-party' library for this, I'd love to use luma for this.