rm-hull/luma.led_matrix #55: API Documentation missing

rm-hull, 2017-01-19:


thijstriemstra, 2017-01-23:
This looks fixed to me: https://luma-led-matrix.readthedocs.io/en/latest/api-documentation.html

fullphat, 2017-01-23:
Many functions are still undocumented, and some are missing detail:

contrast(value)[source]
Sets the LED intensity

What is "value"?  I'm guessing it's a float between 0.1 and 1.0, but it could be an int between 0 and 1, 0 and 255...

I'm a bit confused by this too:

luma.led_matrix.legacy.show_message

Will there be a forthcoming luma.led_matrix.show_message()

That offers more features?

Chris
full phat products
__________________________
http://getsnarl.info
Twitter: @tweetsnarl

rm-hull, 2017-01-23:
Yeah, the docstring content needs to be filled out in more detail. This will happen shortly.

The legacy namespace is really there as a stopgap - to provide some illusion of backwards compatibility.

There are building blocks in luma.Core now that allow you to move a viewport over a larger virtual device. This can be programmed to give the appearance of scrolling. There have been questions to make scroll message scroll in any direction, work with asyncio, and yours about interrupting it if there is a higher priority message.

By providing the mechanism to scroll, you can implement it to suit your individual needs, rather than the library providing a kitchen-sink version that is complicated and does everything.

So think of the legacy version as a simple reference version, and no, there most likely won't be any more scroll implementations in the main codebase. It is likely that there maybe demos and examples added instead to show you what is possible though.

thijstriemstra, 2017-02-26:
@rm-hull any news?

rm-hull, 2017-02-26:
I think this ticket can be closed now

closed 2017-02-26 by rm-hull