rm-hull/luma.led_matrix #56: Issue with example

mB-PiBox, 2017-01-19:
I ran the setup.py from the recent update/changes from the output it looks like all went through correctly, however if I run "sudo python matrix_test.py" I get the following:

```
Traceback (most recent call last):
  File "matrix_test.py", line 8, in <module>
    import luma.led_matrix.legacy as legacy
ImportError: No module named luma.led_matrix.legacy
```

rm-hull, 2017-01-19:
What is the output of:

    $ ls -l /usr/local/lib/python2.7/dist-packages/luma/led_matrix/

Also:

    $ ls -l /usr/local/lib/python2.7/site-packages/luma/led_matrix/

One should work, one shouldn't - I cant remember which dir they are deployed to by default.

It should list something like:

    $ ls -l /usr/local/lib/python2.7/dist-packages/luma/led_matrix/
    total 156
    -rw-r--r-- 1 root staff   276 Jan 19 11:05 const.py
    -rw-r--r-- 1 root staff   547 Jan 19 11:05 const.pyc
    -rw-r--r-- 1 root staff  5475 Jan 19 11:05 device.py
    -rw-r--r-- 1 root staff  7078 Jan 19 11:05 device.pyc
    -rw-r--r-- 1 root staff   162 Jan 19 11:05 __init__.py
    -rw-r--r-- 1 root staff   223 Jan 19 11:05 __init__.pyc
    -rw-r--r-- 1 root staff 76013 Jan 19 11:05 legacy.py
    -rw-r--r-- 1 root staff 37913 Jan 19 11:05 legacy.pyc
    -rw-r--r-- 1 root staff  1918 Jan 19 11:05 segment_mapper.py
    -rw-r--r-- 1 root staff  2107 Jan 19 11:05 segment_mapper.pyc

Also which version of pip are you using?

    $ pip --version
    pip 9.0.1 from /usr/local/lib/python2.7/dist-packages (python 2.7)

rm-hull, 2017-01-19:
@mB-PiBox - can you try running the example from the project root directory:

    $ sudo python examples/matrix_test.py

Also, you shouldn't strictly need to use sudo if you add your user to the relevant spi and gpio groups:

    $ sudo usermod -a -G i2c,spi,gpio pi

_(you'll need to log out and back in again for this to take effect)_

mB-PiBox, 2017-01-19:
```
$ pip --version
pip 1.5.6 from /usr/lib/python2.7/dist-packages (python 2.7)
```

There is no luma fold uner dist-packages or site-packages.
there is a luma.core-0.1.3_-py2.7.egg  &  luma.led_matrix-0.3.0-py2.7.egg within dist-package folder however.

running
`sudo python examples/matrix_test.py`
returns the same result of
```
Traceback (most recent call last):
  File "examples/matrix_test.py", line 8, in <module>
    import luma.led_matrix.legacy as legacy
ImportError: No module named luma.led_matrix.legacy
```

rm-hull, 2017-01-19:
Could you try upgrading your pip version, and then try re-installing?

    $ sudo pip install --upgrade pip
    $ sudo pip install --upgrade luma.core luma.led_matrix

thijstriemstra, 2017-01-19:
Why not from luma.led_matrix import legacy. Why import legacy as legacy?

mB-PiBox, 2017-01-19:
@rm-hull I'll have to work on updating pip later, with Raspbian python come packaged within which uses python-pip via apt-get (which is not updated often, unfortunately.) So I'll have to install an updated PIP version manually from 'get-pip.py' which supposedly could 'cause issues if python is managed by the OS. I'm going to have to figure/find out how to get python installed removed from OS control, and install python manually via download from python.org. Work is super busy at the moment, so I don't have time to play around with it, and certainly no time if something breaks to fix it... so I will put this on hold for a short time until I can get back too it. Hopefully this will be soon, as I've been looking for Vertical scrolling on the matrix for some time to complete several projects that have been on hold.

I'll post back if all work or not when I get that time to work on it.

Thank you for taking the time to review the issue, and big thanks for the update with vertical scrolling ;)

thijstriemstra, 2017-01-19:
@mB-PiBox take a look at virtualenv, no need to install a python from source.

rm-hull, 2017-01-19:
> Why not from luma.led_matrix import legacy. Why import legacy as legacy?

Good point... it was late, i have no other excuse ;-)

thijstriemstra, 2017-01-19:
Alright cool, I thought it was some other python voodoo. I'll open PRs where I encounter it.

thijstriemstra, 2017-01-21:
@rm-hull the other example is still using the incorrect max7219 import

rm-hull, 2017-01-21:
@thijstriemstra 

>  the other example is still using the incorrect max7219 import

Yes, I hadn't got round to updating it yet... There's an outstanding ticket to address this: #54

thijstriemstra, 2017-01-24:
@rm-hull is it correct the matrix example doesn't fully work yet? It's only displaying parts of the whi\ole demo, e.g. the alternative font stuff is not displaying anything.

rm-hull, 2017-01-24:
If you look here, I tried it with 2 daisy chained devices: https://github.com/rm-hull/luma.led_matrix/blob/master/examples/matrix_demo.py#L17

If you've only got one device, change it to be `cascaded=1` and see if that works for you. That probably ought to be committed too

rm-hull, 2017-01-26:
@mB-PiBox - as long as you've got a version of pip installed, it should self-update with:

    $ sudo -H pip install --upgrade pip
    $ pip --version
    pip 9.0.1 from /usr/local/lib/python2.7/dist-packages (python 2.7)

Next, something clearly something got corrupted along the way on your RPi, so I would suggest running:

    $ pip list | grep luma

Then for each luma.* package listed, uninstall it:

    $ sudo -H pip uninstall luma.core 
    $ sudo -H pip uninstall luma.led_matrix

And so on ...

Next re-install... In theory _luma.core_ should be installed as a dependency, but for the sake of caution, do it explicitly:

    $ sudo -H pip install luma.core luma.led_matrix

Pull the latest from the git repo (I just made some fixes to the demo), and inside the project root dir:

    $ git pull

And retry the demo:

    $ python examples/matrix_demo.py

rm-hull, 2017-01-26:
@thijstriemstra 

> ... is it correct the matrix example doesn't fully work yet? It's only displaying parts of the whole demo, e.g. the alternative font stuff is not displaying anything. The 7-segment example worked fine.

I changed the `cascaded=1` flag, but noticed that the `fill=` param was also missing in various places, if you could check & let me know if it works for you

mB-PiBox, 2017-01-27:
@rm-hull I tried all things I could think of to update pip, including your example command. Unfortunately they all return the same 
```
$ sudo -H pip install --upgrade pip
...
Found existing installation: pip 1.5.6
   Not uninstalling pip at /usr/lib/python2.7/dist-packages, owned by OS
$ pip --version
pip 1.5.6 from /usr/local/lib/python2.7/dist-packages (python 2.7)

```

thijstriemstra, 2017-01-27:
Yea, it's not possible to upgrade the system pip for some reason. with sudo doesnt help either. that's why a virtual environment is probably a better idea, where you can freely update your packages.

fullphat, 2017-01-27:
I was running a very old version of pip (same as yours) and I encountered this message as well, although pip did actually update.  Have you rebooted since?

Here's what I used to update pip:

sudo pip install --upgrade pip

rm-hull, 2017-01-27:
@mB-PiBox Also, try with -i if that doesn't work:

    sudo -i pip install --upgrade pip

Ultimately, perhaps the only way out of this is to go down the `getpip.py` route.

I don't remember having any of this bother upgrading pip, but appreciate when things do wrong they generally go spectacularly wrong.

I think part of the reason why we're seeing these issues is that we've used some setup features that the stock pip on RPi doesn't support, and this leaves packages installed in a broken state.

thijstriemstra, 2017-01-27:
This [comment](https://github.com/pypa/pip/issues/2964#issuecomment-206066524) suggests:

install pip (even if it is already installed):
```
wget https://bootstrap.pypa.io/get-pip.py
```
Then run the following:
```
sudo python get-pip.py
```

Looking at the other comments in that thread I suggest we use these instructions as well. Fuck system pip :-1:

thijstriemstra, 2017-01-27:
> I changed the cascaded=1 flag, but noticed that the fill= param was also missing in various places, if you could pull from master & check + let me know if it works for you

@rm-hull works great now, thanks. For this ticket I suggest updating the pip installation instructions and letting people know they have to upgrade or this package will be broken :(

pete-r61, 2017-02-06:
Hello, I used a pi-top at first finding out no pip update was done as it was owned bi the OS.
Next tried on basic RP2 and update was successful.

SPI is turned ON

Did following as well:
$ sudo usermod -a -G spi,gpio pi
$ git clone https://github.com/rm-hull/luma.led_matrix.git
$ cd luma.led_matrix
$ sudo apt-get install python-dev python-pip libfreetype6-dev libjpeg8-dev
$ sudo python setup.py install

...just to make sure: also used $ sudo -H pip install luma.core luma.led_matrix    

Still loads of errors of modules not found when running matrix_demo.py
**from luma.led_matrix import legacy
ImportError: No module named luma.led_matrix**

When doing pip list grep luma   .... I do get:
**luma .core (0.1.14)
luma.led-matrix (0.4.4)**

Now this last line says luma.le**d-m**atrix (0.4.4)  and not luma.le**d_m**atrix (0.4.4)

But anyway I do not know what to do to get it to work.  

Some advise anyone?

rm-hull, 2017-02-06:
Hi @pete-r61,

The default version of pip installed on Raspian is really old and I think this is the cause of issues people are having. Can you confirm which version of pip you have? try:

    $ pip --version
    pip 9.0.1 from /usr/local/lib/python2.7/site-packages (python 2.7)

If yours reports a different version from 9.0.1, can you try:

    $ sudo pip install --upgrade pip
    $ sudo pip uninstall luma.core luma.led_matrix

Make sure there are no stray `luma*` files or directories in `/usr/local/lib/python2.7/site-package` (I assume you are on python 2.7 ?), if there are remove them.

And then try and re-install:

    $ sudo pip install luma.core luma.led_matrix
    
This is what I see:

    $ pip list | grep luma
    luma.core (0.1.14)
    luma.lcd (0.2.1)
    luma.led-matrix (0.4.4)
    luma.oled (2.2.2)

So I don't think the underscore / dash in the name is a problem per se.

pete-r61, 2017-02-07:
problem seems pip... but I can not install 9.0.1...  While using the install cmd this version is downloaded... but even on this RP2  (so no pi-top) it is not installed ...it says:

pi@raspberrypi:~ $ sudo pip install --upgrade pip
Downloading/unpacking pip from https://pypi.python.org/packages/b6/ac/7015eb97dc749283ffdec1c3a88ddb8ae03b8fad0f0e611408f196358da3/pip-9.0.1-py2.py3-none-any.whl#md5=297dbd16ef53bcef0447d245815f5144
  Downloading pip-9.0.1-py2.py3-none-any.whl (1.3MB): 1.3MB downloaded
Installing collected packages: pip
  Found existing installation: pip 1.5.6
    Not uninstalling pip at /usr/lib/python2.7/dist-packages, owned by OS
Successfully installed pip
Cleaning up...

pi@raspberrypi:~ $ sudo pip --version
pip 1.5.6 from /usr/lib/python2.7/dist-packages (python 2.7)

So if problem is the version of pip.... then how can I get it installed. Seems OS blocks it.

thijstriemstra, 2017-02-07:
@pete-r61 can you try these instructions: https://github.com/rm-hull/luma.led_matrix/issues/56#issuecomment-275752815

rm-hull, 2017-02-07:
@pete-r61 it looked like pip 9.0.1 installed ok, but 1.5.6 still takes preference. Try:

    $ sudo apt-get purge python-pip
    $ pip --version

Hopefully it should report 9.0.1 now.

I updated the install instructions yesterday to reflect this. If you then resume as per here: 
http://luma-led-matrix.readthedocs.io/en/latest/install.html#installing-from-pypi

frekel, 2017-02-24:
I can confirm that pip was the issue. Follow https://github.com/rm-hull/luma.led_matrix/issues/56#issuecomment-277845830 to get this running!

thijstriemstra, 2017-02-26:
@rm-hull let's include these instructions and close this ticket? what's the current status?

rm-hull, 2017-03-04:
The [install instructions](http://luma-led-matrix.readthedocs.io/en/latest/install.html#installing-from-pypi) reflect the fact that pip 9.0.1 is required, and what steps are needed to get there.

closed 2017-03-04 by rm-hull