rm-hull/luma.led_matrix #57: Error initialising sevensegment

twdkeule, 2017-01-19:
When using the sevensegment class a TypeError occurs:
> Traceback (most recent call last):
>   File "8_digit.py", line 21, in <module>
>     seg = sevensegment(device)
>   File "/usr/local/lib/python3.4/dist-packages/luma/led_matrix/device.py", line 123, in __init__
>     self.text = ""
>   File "/usr/local/lib/python3.4/dist-packages/luma/led_matrix/device.py", line 131, in text
>     self._text_buffer = observable(bytearray(value), observer=self.flush)
> TypeError: string argument without an encoding

According to the Python3 documentation, the encoding needs to be added
> If it is a string, you must also give the encoding (and optionally, errors) parameters; bytearray() then converts the string to bytes using str.encode().

twdkeule, 2017-01-19:
I tried to fix it locally by editing device.py and adding the encoding 'utf-8' on line 131, but I ran into another error:

> Traceback (most recent call last):
>   File "8_digit.py", line 21, in <module>
>     seg = sevensegment(device)
>   File "/home/pi/scripts/device.py", line 123, in __init__
>     self.text = ""
>   File "/home/pi/scripts/device.py", line 132, in text
>     self.flush(self._text_buffer)
>   File "/home/pi/scripts/device.py", line 135, in flush
>     data = bytearray(self.segment_mapper(buf, notfound=self.undefined)).ljust(int(self._bufsize), '\0')
> TypeError: an integer is required

Fyi, the code I am trying to run is 
```
from sys import argv
from time import sleep
from luma.core.serial import spi
from luma.core.render import canvas
from device import max7219, sevensegment

if __name__ == "__main__":
	cascade = 1
	serial = spi(port=0, device=0)
	device = max7219(serial, cascaded=cascade)
	seg = sevensegment(device)
	# Do stuff with seg ...
```

rm-hull, 2017-01-19:
So as you've discovered, this is "hot off the forge", and clearly needs some tests writing to smoke these issues out. 

I tried it on python2, but there some UTF-8 encoding differences with python3, and it has highlighted a bug in the code in the segment mapper: I just pushed a code fix into the repo, but wont do a release just yet - if you sync across the changes and let me know if that lets you get further.

p.s. If you've only got one device, you can skip the `cascaded=` flag and it should default

twdkeule, 2017-01-19:
Thanks for the fast response.
 I don't mean to be ungrateful, but you made a typo and added too many .decode()'s, so I made a pull request #59

rm-hull, 2017-01-19:
No - many thanks for the PR - merged it

rm-hull, 2017-01-19:
Although ... the tests around that now seem to fail - have a look here: https://travis-ci.org/rm-hull/luma.led_matrix/jobs/193409843

The test suite can be run if you install `tox` - and then run

    $ tox 

or 

    $ tox -e py27

I tend to run the tests on a linux desktop machine, as its a lot faster than on a RPi

rm-hull, 2017-01-19:
I've pushed a fix - master is now green - without the decode on the bytearray, it returns the ordinals rather than the ascii characters

closed 2017-01-19 by rm-hull