rm-hull/luma.led_matrix #58: Python3 installation with setup.py does not install

twdkeule, 2017-01-19:
While trying to install from source with `sudo python3 setup.py install` the luma.led_matrix module is not installed to /usr/local/lib/python3.4/dist-packages/, only an egg is placed there.

> pi@raspberrypi:\~/luma.led_matrix $ sudo python3 setup.py install
> running install
> running bdist_egg
> running egg_info
> writing luma.led_matrix.egg-info/PKG-INFO
> writing requirements to luma.led_matrix.egg-info/requires.txt
> writing namespace_packages to luma.led_matrix.egg-info/namespace_packages.txt
> writing dependency_links to luma.led_matrix.egg-info/dependency_links.txt
> writing top-level names to luma.led_matrix.egg-info/top_level.txt
> reading manifest file 'luma.led_matrix.egg-info/SOURCES.txt'
> reading manifest template 'MANIFEST.in'
> no previously-included directories found matching 'doc/_build'
> warning: no previously-included files matching '__pycache__' found under directory '*'
> warning: no previously-included files matching '*.py[co]' found under directory '*'
> warning: no previously-included files matching '*~' found under directory '*'
> warning: no previously-included files matching '.coverage' found under directory '*'
> writing manifest file 'luma.led_matrix.egg-info/SOURCES.txt'
> installing library code to build/bdist.linux-armv7l/egg
> running install_lib
> running build_py
> creating build/bdist.linux-armv7l/egg
> creating build/bdist.linux-armv7l/egg/luma
> creating build/bdist.linux-armv7l/egg/luma/led_matrix
> copying build/lib/luma/led_matrix/segment_mapper.py -> build/bdist.linux-armv7l/egg/luma/led_matrix
> copying build/lib/luma/led_matrix/legacy.py -> build/bdist.linux-armv7l/egg/luma/led_matrix
> copying build/lib/luma/led_matrix/const.py -> build/bdist.linux-armv7l/egg/luma/led_matrix
> copying build/lib/luma/led_matrix/__init__.py -> build/bdist.linux-armv7l/egg/luma/led_matrix
> copying build/lib/luma/led_matrix/device.py -> build/bdist.linux-armv7l/egg/luma/led_matrix
> byte-compiling build/bdist.linux-armv7l/egg/luma/led_matrix/segment_mapper.py to segment_mapper.cpython-34.pyc
> byte-compiling build/bdist.linux-armv7l/egg/luma/led_matrix/legacy.py to legacy.cpython-34.pyc
> byte-compiling build/bdist.linux-armv7l/egg/luma/led_matrix/const.py to const.cpython-34.pyc
> byte-compiling build/bdist.linux-armv7l/egg/luma/led_matrix/__init__.py to __init__.cpython-34.pyc
> byte-compiling build/bdist.linux-armv7l/egg/luma/led_matrix/device.py to device.cpython-34.pyc
> creating build/bdist.linux-armv7l/egg/EGG-INFO
> copying luma.led_matrix.egg-info/PKG-INFO -> build/bdist.linux-armv7l/egg/EGG-INFO
> copying luma.led_matrix.egg-info/SOURCES.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
> copying luma.led_matrix.egg-info/dependency_links.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
> copying luma.led_matrix.egg-info/namespace_packages.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
> copying luma.led_matrix.egg-info/requires.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
> copying luma.led_matrix.egg-info/top_level.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
> zip_safe flag not set; analyzing archive contents...
> creating 'dist/luma.led_matrix-0.3.0-py3.4.egg' and adding 'build/bdist.linux-armv7l/egg' to it
> removing 'build/bdist.linux-armv7l/egg' (and everything under it)
> Processing luma.led_matrix-0.3.0-py3.4.egg
> Copying luma.led_matrix-0.3.0-py3.4.egg to /usr/local/lib/python3.4/dist-packages
> Adding luma.led-matrix 0.3.0 to easy-install.pth file
> 
> Installed /usr/local/lib/python3.4/dist-packages/luma.led_matrix-0.3.0-py3.4.egg
> Processing dependencies for luma.led-matrix==0.3.0
> Searching for luma.core==0.1.3
> Best match: luma.core 0.1.3
> Adding luma.core 0.1.3 to easy-install.pth file
> 
> Using /usr/local/lib/python3.4/dist-packages
> Searching for smbus2==0.1.4
> Best match: smbus2 0.1.4
> Adding smbus2 0.1.4 to easy-install.pth file
> 
> Using /usr/local/lib/python3.4/dist-packages
> Searching for spidev==3.2
> Best match: spidev 3.2
> Adding spidev 3.2 to easy-install.pth file
> 
> Using /usr/local/lib/python3.4/dist-packages
> Searching for RPi.GPIO==0.6.3
> Best match: RPi.GPIO 0.6.3
> Adding RPi.GPIO 0.6.3 to easy-install.pth file
> 
> Using /usr/lib/python3/dist-packages
> Searching for Pillow==4.0.0
> Best match: Pillow 4.0.0
> Adding Pillow 4.0.0 to easy-install.pth file
> 
> Using /usr/local/lib/python3.4/dist-packages
> Searching for olefile==0.44
> Best match: olefile 0.44
> Adding olefile 0.44 to easy-install.pth file
> 
> Using /usr/local/lib/python3.4/dist-packages
> Finished processing dependencies for luma.led-matrix==0.3.0

After installation:

> pi@raspberrypi:~/luma.led_matrix $ ls -lh /usr/local/lib/python3.4/dist-packages/
> total 148K
> -rw-r--r-- 1 root staff 1.1K Jan 19 11:47 OleFileIO_PL.py
> drwxr-sr-x 3 root staff 4.0K Jan 19 11:48 PIL
> drwxr-sr-x 2 root staff 4.0K Jan 19 11:48 Pillow-4.0.0.egg-info
> drwxr-sr-x 2 root staff 4.0K Jan 19 12:06 __pycache__
> -rw-r--r-- 1 root staff   34 Jan 19 15:10 easy-install.pth
> -rw-r--r-- 1 root staff  126 Jan 19 12:06 easy_install.py
> drwxr-sr-x 3 root staff 4.0K Jan 19 12:12 luma
> -rw-r--r-- 1 root staff  545 Jan 19 11:35 luma.core-0.1.3-py2.7-nspkg.pth
> drwxr-sr-x 2 root staff 4.0K Jan 19 11:35 luma.core-0.1.3.dist-info
> -rw-r--r-- 1 root staff  29K Jan 19 15:10 luma.led_matrix-0.3.0-py3.4.egg
> drwxr-sr-x 3 root staff 4.0K Jan 19 11:48 olefile
> drwxr-sr-x 2 root staff 4.0K Jan 19 11:48 olefile-0.44.egg-info
> drwxr-sr-x 5 root staff 4.0K Jan 19 12:06 pkg_resources
> drwxr-sr-x 5 root staff 4.0K Jan 19 12:06 setuptools
> drwxr-sr-x 2 root staff 4.0K Jan 19 12:06 setuptools-33.1.1.dist-info
> drwxr-sr-x 3 root staff 4.0K Jan 19 11:35 smbus2
> drwxr-sr-x 2 root staff 4.0K Jan 19 11:35 smbus2-0.1.4.egg-info
> drwxr-sr-x 2 root staff 4.0K Jan 19 11:35 spidev-3.2.egg-info
> -rwxr-xr-x 1 root staff  47K Jan 19 11:35 spidev.cpython-34m.so

> pi@raspberrypi:~/luma.led_matrix $ ls -lh /usr/local/lib/python3.4/dist-packages/luma
> total 4.0K
> drwxr-sr-x 3 root staff 4.0K Jan 19 11:35 core

twdkeule, 2017-01-20:
Also tried it on an Ubuntu 16.10 laptop. Still same error:
python3 test.py                 
Traceback (most recent call last):
  File "test.py", line 5, in <module>
    from luma.led_matrix.device import max7219, sevensegment
ImportError: No module named 'luma.led_matrix

However, when using python2 everything works as expected.

rm-hull, 2017-01-20:
Yeah, there's a bit of work to do to get it working properly on Python 3:

* I'd used bytearray, without really exploring how painful it was to make it work across 2 and 3, so I think I will look at something different (this relates to the other issue you opened)
* Namespaced packaged are what I think is probably the root cause of this

If you can bear with us while it is a bit choppy, it'll soon settle down - in fact, your input has been really useful, so please don't hold back on raising issues.

twdkeule, 2017-01-21:
I think I found it. After installation the luma.core package was located in a regular directory, while luma.led_matrix was part of an .egg-file. 
So I added `zip_safe=False,` to setup.py to install led_matrix in a folder and now it works.

thijstriemstra, 2017-01-23:
This should be fixed in the next release.

closed 2017-01-24 by twdkeule