rm-hull/luma.led_matrix #66: Any news on support for the ws2812 / Unicorn Hat?

fullphat, 2017-01-26:
Thanks :)

rm-hull, 2017-01-26:
If you grab the latest (0.4.0) with:

    $ sudo -H pip install --upgrade luma.led_matrix

Then head over to http://luma-led-matrix.readthedocs.io/en/latest/python-usage.html#ws2812-neopixels

fullphat, 2017-01-27:
Done the above and pasted the same code from the readthedocs.io link you posted into 'test.py' and ran it - received the following error: 

```
pi@ninja:~/luma.phat $ sudo python test.py
Traceback (most recent call last):
  File "test.py", line 5, in <module>
    device = neopixel(width=8, height=8)
  File "/usr/local/lib/python2.7/dist-packages/luma/led_matrix/device.py", line 211, in __init__
    self._ws2812 = dma_interface or self.__ws2812__()
AttributeError: 'neopixel' object has no attribute '__ws2812__'
pi@ninja:~/luma.phat $ 
```

Suggestions? :)

thijstriemstra, 2017-01-27:
@fullphat pip install ws2812

rm-hull, 2017-01-27:
> pip install ws2812

No, shouldn't need to do that - i did some refucktoring which caused this - just putting out a patched version now to fix this

thijstriemstra, 2017-01-27:
> refucktoring
:+1:

rm-hull, 2017-01-27:
Ok, 0.4.1 is on pypi - @fullphat : if you can try:

   $ sudo -H pip install --upgrade luma.led_matrix

and let me know how it goes ... and _pics or it didn't happen_ ;-)

fullphat, 2017-01-27:
pi@ninja:~/luma.phat $ sudo python test.py
Traceback (most recent call last):
  File "test.py", line 5, in <module>
    device = neopixel(width=8, height=8)
  File "/usr/local/lib/python2.7/dist-packages/luma/led_matrix/device.py", line 212, in __init__
    self._ws2812 = dma_interface or self.__ws2812__()
TypeError: __ws2812__() takes exactly 2 arguments (1 given)
pi@ninja:~/luma.phat $ 

close but no 🚬 😆

rm-hull, 2017-01-27:
Can you tell I'm doing this without an actual RPi to test it on?  😵 

Pull down 0.4.2 with `$ sudo -H pip install --upgrade luma.led_matrix`

fullphat, 2017-01-27:
It's certainly got an Agile feel to it :rofl: 

That's worked, although the first time I ran, it did nothing.  I stuck a wait loop in and I got a matrix full of colour.  Ran it again and nothing.  It seems to not work, then work, then not work.  Getting there though!  Will post a pic up in a bit. 😋 

What's supposed to appear using this?

```
with canvas(device) as draw:
    draw.line((0, 0, 0, device.height), fill="red")
    draw.line((1, 0, 1, device.height), fill="orange")
    draw.line((2, 0, 2, device.height), fill="yellow")
    draw.line((3, 0, 3, device.height), fill="green")
    draw.line((4, 0, 4, device.height), fill="blue")
    draw.line((5, 0, 5, device.height), fill="indigo")
    draw.line((6, 0, 6, device.height), fill="violet")
    draw.line((7, 0, 7, device.height), fill="white")
```

rm-hull, 2017-01-27:
It's quite hard to take a photo as the neopixels are so bright.
Basically a vertical red line, orange line, yellow, etc

[image]

You probably want a `time.sleep(30)` or something at the end (not inside the `with` context though) - otherwise when the python script exits, the LEDs are automatically switched off

```python
with canvas(device) as draw:
    draw.line((0, 0, 0, device.height), fill="red")
    draw.line((1, 0, 1, device.height), fill="orange")
    draw.line((2, 0, 2, device.height), fill="yellow")
    draw.line((3, 0, 3, device.height), fill="green")
    draw.line((4, 0, 4, device.height), fill="blue")
    draw.line((5, 0, 5, device.height), fill="indigo")
    draw.line((6, 0, 6, device.height), fill="violet")
    draw.line((7, 0, 7, device.height), fill="white")

time.sleep(30)
```

fullphat, 2017-01-27:
Mine is more a random mix of colour, although I have the 8x8 hat (I changed the height const).  The thing is bright - I've gone for a Pibow Ninja case with a diffuser top - much better on the eyes!

With 8x4 setting:

thijstriemstra, 2017-01-27:
@fullphat can you post a picture?

rm-hull, 2017-01-27:
@fullphat I'm not seeing any images on your last post...
Hmm, it could be that the 8x8 hat is wired differently. 

If you run this script, changing `cascaded=32` where appropriate for the 8x8 hat what do you see?

```python
import time

from luma.led_matrix.device import neopixel
from luma.core.render import canvas

device = neopixel(cascaded=32)

for i in range(device.cascaded):
    with canvas(device) as draw:
        draw.point((i, 0), fill="green")
    time.sleep(0.5)
```

On the 8x4 pHat, a green dot scans across the first row from left to right, then the second row left to right, etc

If it's not the same for the 8x8, it means that we need to decode how they are laid out physically across the PCB and then have an indirection mapping of some sort

fullphat, 2017-01-27:
[image]
[image]

Sorry - was gambling teaching them to an email would work.

I'll do some testing using setpixel() to see if I can figure out what's going on.  The Pimoroni docs do say the 8x8 is just double the height of the 8x4...

thijstriemstra, 2017-01-27:
that looks like an UFO sighting..

rm-hull, 2017-01-27:
Ha!, look at this: https://github.com/pimoroni/unicorn-hat/blob/master/library/UnicornHat/unicornhat.py#L33-L42

For want of a better way to describe it the 8x8 'snakes', starting at the asterisk:
```
<<<<<<<< * 
>>>>>>>>
<<<<<<<<
>>>>>>>>
<<<<<<<<
>>>>>>>>
<<<<<<<<
>>>>>>>>
```

While the 8x4 'scans' https://github.com/pimoroni/unicorn-hat/blob/master/library/UnicornHat/unicornhat.py#L44-L49:

```
* >>>>>>>>
  >>>>>>>>
  >>>>>>>>
  >>>>>>>>
```

So it looks like we do need a mapping a bit like the first pimoroni link

rm-hull, 2017-01-27:
If you flipped every other column in the pic, it would render as expected. I shan't do it tonight now, but I will add a pluggable translation mapping for the 8x8 hat

fullphat, 2017-01-27:
yep, confirm i get the same, although just to add:

first time i ran it - nothing appeared at all
second time i ran it, four rows (#4-#7) of blinding white leds :) and a green dot 'snaking' from top to bottom along rows 0-3

fullphat, 2017-01-28:
I also changed the code to read cascaded=64 and, while it does work, once it hits row 4, the green dot starts to 'jump' occasionally, and sometimes two dots appear.

rm-hull, 2017-01-29:
@fullphat 0.4.3 released, if you upgrade with `$ sudo -H pip install --upgrade luma.led_matrix`, and then look here: http://luma-led-matrix.readthedocs.io/en/latest/python-usage.html#pimoroni-unicorn-hat

re: the "jumping' dot and two dots issue - the timing for WS2812 is quite critical, and perhaps the underlying library used to do the DMA bit-banging is not as tolerant with longer chains. We can investigate that as a separate issue if you can confirm that 0.4.3 fixes the layout issue on 8x8.

fullphat, 2017-01-29:
Done and tested.  Green dot moves correctly from left to right down each line!  Still jumps on line 4 however.

I got carried away and tested the rainbow stripes sample as well - that still doesn't work however, and there does still seem to be an initialisation (or tidy up) issue as well as often it only works once, and sometimes it leaves random LEDs lit when it ends.

rm-hull, 2017-01-29:
@fullphat Ok, great - lets close this issue (the thread is getting long enough) - can you open another issue please for the jumping dot/two dot problems and in it, add the following details:

* Description from above
* Any observations & patterns
* Your device type (PiZero / 2B / 3B, etc)
* OS & kernel version (paste output of `uname -a`)
* Output of `dmesg` after running the demos
* What amp rating does the power supply provide?
* What other peripherals are connected? screen, keyboard, mouse, WIFI, etc?
* Do you get the same erratic behaviour if you use the official Pimoroni drivers?

On the 8x4 Unicorn pHAT, I had to solder the header on - is this the same for the 8x8 HAT? - if so check for any bad solder joints

closed 2017-01-29 by rm-hull

fullphat, 2017-01-29:
Agreed!