rm-hull/luma.led_matrix #69: Orientation wrong on 8x32 module

smalldave, 2017-01-29:
Hi,

I've got the same problem as is being described over on [this arduino forum](https://forum.arduino.cc/index.php?topic=402656.0)

This diagram from there neatly illustrates the problem.
[image]

The module looks like this.

[image]

I guess I could break it apart and re-orientate it but that seems a little drastic.

Is there a simple workaround for this?

If not I'm happy to try and put together a pull request if you could give me some pointers.

rm-hull, 2017-01-29:
By coincidence, I just bought 3 of these 8x8x4 modules two days ago, so should have something to play with in a few weeks.

I think the difference is that some of the LED matrices are common-anode, others are common-cathode, and this affects how they are wired up, and consequently the scroll direction. I highly doubt you would be able to break it apart and re-orientate it.

The good news is that we can do that orientation fix in the code... the library has undergone a significant rewrite recently, and there was a block-level `rotate8x8.py` module that did this previously, which perhaps should be re-introduced (see the code [here](https://github.com/rm-hull/luma.led_matrix/blob/0.2.3/max7219/rotate8x8.py), and a related [issue](https://github.com/rm-hull/luma.led_matrix/issues/28)).

There are probably two different ways to tackle this:
* Looking at https://github.com/rm-hull/luma.led_matrix/blob/master/luma/led_matrix/device.py#L83-L94, you see how it varies _y_ at line 87, maybe swap x and y, so that _x_ varies over `self._w` and _y_ is fixed.
* Use the rotate8x8.py module (or something similar in Pillow) to pre-process the image to re-orient every 8x8 pixel block - there is already a preprocess(image) method which could/should be overridden. 

_[I think I prefer the 2nd option personally]_

Either way, the class would be constructed with something like:

```python
device = max7219(cascaded=4, common_cathode=True)
```

If `common_cathode` is the wrong name then `common_anode`, but it would be optional and default to False.

Happy for you to either submit a PR, or add you as a collaborator and we can then work on a shared branch.

rm-hull, 2017-01-29:
I've added some code in commit d1b65d7de7d78fcf26be49da697ae1577feb2431 - if you could test it on the **block-orientation** branch and let me know how it goes

rm-hull, 2017-01-30:
On reflection, not sure calling the parameter `common_row_cathode` is the best thing to do, on the basis that (a) it might be wrong, (b) someone with 'the other type' of device might not immediately make the connection with the argument name and the behavior they are seeing.

Open to suggestions for a better name, throwing these out there:
* vertical_align
* block_realign
* fix_alignment

Any better names?

cc: @thijstriemstra

smalldave, 2017-01-30:
I think I prefer block_orientation_horizontal=True/False. I know whether it is horizontal or vertical is a bit arbitrary but it's more clear to me what the transformation is.

I was just settling in to have a look at making the change and updating the tests and I see you've done it all :).

Still learned a bit about virtualenv and pip (I don't have a huge amount of experience in Python) so that's a plus.

Anything you want me to do shout. 
I will be quite slow though :)

rm-hull, 2017-02-01:
Lets go with `block_orientation='vertical'` to fix the behaviour you describe in the opening issue description, and this defaults to `block_orientation='horizontal'` for current behaviour.

@pmounsey [wrote](https://github.com/rm-hull/luma.led_matrix/commit/a43d1549f54d4e697fe36b760dbe8fadd1c2465e#commitcomment-20691041):

> Nice. Thanks for your work on this. Can MAX7219 FC-16 1x4 or 8 x32 be referenced horizontal versus vertical?

I'll add some opt-args to the max7219 example for this - expect the [documentation](http://luma-led-matrix.readthedocs.io/en/latest/python-usage.html#examples) to be updated to reflect the flags required when this gets merged into master

closed 2017-02-02 by rm-hull

gs-1998, 2018-05-11:
can i get the solution to this problem?

serestre, 2018-05-22:
can i get the solution to this problem?

rm-hull, 2018-05-22:
Look at the documentation

PR #74 (2017-02-02): Block orientation
Fixes #69
commit 643b8c58 (2017-02-02): Block orientation (#74)

Fixes #69 

* Support common-row anode and common-row cathode LED matrices
* Block realignment by 90deg clockwise
* Add documentation
* Add args to demo
commit 58434e60 (2017-02-15): Block orientation (#74)

Fixes #69 

* Support common-row anode and common-row cathode LED matrices
* Block realignment by 90deg clockwise
* Add documentation
* Add args to demo