rm-hull/luma.led_matrix #70: Mock assert_called() fails on py34 only

rm-hull, 2017-01-29:
See FIXME's in `tests/test_neopixel.py`

thijstriemstra, 2017-01-30:
Is this visible on travis?

rm-hull, 2017-01-30:
Can't find the failed builds on the led_matrix repo, but heres a similar one on luma.oled: https://travis-ci.org/rm-hull/luma.oled/builds/196357642

thijstriemstra, 2017-01-30:
`assert_called` isn't documented on https://docs.python.org/3.5/library/unittest.mock.html. `assert_called_with` is and I think this should be used instead. Making a PR now.

thijstriemstra, 2017-01-30:
There's actually a `called` boolean that looks better: https://docs.python.org/3.5/library/unittest.mock.html#unittest.mock.Mock.called

thijstriemstra, 2017-01-30:
See https://github.com/rm-hull/luma.oled/pull/132 for luma.oled, working on luma.led_matrix PR now.

PR #73 (2017-01-30): use called in test
fixes #70
commit 7759a4cc (2017-01-30): use called in test (#73)

Fixes #70
commit 9fbe6fa7 (2017-02-15): use called in test (#73)

Fixes #70
closed 2017-01-30 by rm-hull