rm-hull/luma.led_matrix #72: Odd behaviour when using numa.ledmatrix on Unicorn 8x8 HAT

fullphat, 2017-01-29:
> Description

The following code works, but get sporadic behaviour - most commonly missed pixels:

```
import time

from luma.led_matrix.device import neopixel, UNICORN_HAT
from luma.core.render import canvas

device = neopixel(width=8, height=8, mapping=UNICORN_HAT)

for y in range(device.height):
    for x in range(device.width):
        with canvas(device) as draw:
            draw.point((x, y), fill="green")
        time.sleep(0.5)
```

> Any observations and patterns

On launch, the entire HAT will light up briefly.  Sometimes it's white/bright colours, sometimes it's a random selection of colours.  Green dot begins moving across the display, but after a short while - usually by line #3 or #4 it will start 'jumping' (missing pixels).

> Your device type (PiZero / 2B / 3B, etc)

Pi 3B (less than a week old), no mods other than the HAT itself.

> OS and kernel version (paste output of uname -a)

`Linux ninja 4.4.38-v7+ #938 SMP Thu Dec 15 15:22:21 GMT 2016 armv7l GNU/Linux`

`$ cat /etc/*-release`

```
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

No mods other than enabled SPI, disabled boot to Desktop Manager and enabled SSH

> Output of dmesg after running the demos

Do you want entire output?  Ok to paste in comments?

> What amp rating does the power supply provide?

2.5A (https://www.amazon.co.uk/gp/product/B01M03T0UB/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1)

> What other peripherals are connected? screen, keyboard, mouse, WIFI, etc?

None - just power and the HAT.  Wifi is built-in on Pi 3.  Pi is accessed via SSH.

> Do you get the same erratic behaviour if you use the official Pimoroni drivers?

No, these work consistently fine.

> On the 8x4 Unicorn pHAT, I had to solder the header on - is this the same for the 8x8 HAT? - if so check for any bad solder joints

No, arrived complete and good to go.

rm-hull, 2017-01-29:
@fullphat: could you post the contents of `/boot/config.txt` please?

This issue seems relevant/related: https://github.com/jgarff/rpi_ws281x/issues/103

Also to note, I arbitrarily picked the [ws2812](https://github.com/pimoroni/unicorn-hat/tree/master/library/ws2812) library to do the low level bit-banging. I will create a branch and use the 'other' library ([rpi_ws281x](https://github.com/jgarff/rpi_ws281x)) to see if we get consistent results.

thijstriemstra, 2017-01-31:
@fullphat can you post a video?

rm-hull, 2017-02-01:
@fullphat - there is a branch **rpi_ws281x** with a different low-level neopixel interface implementation, which works on my 8x4 pHAT, but fails miserably on Travis CI (it wonit build/install rpi-ws281x, see https://travis-ci.org/rm-hull/luma.led_matrix/jobs/197477506)

Curious to see if this fixes the issues you describe..

fullphat, 2017-02-08:
Gents - apologies - been maxed out with work.  Give me a few days and I'll do some videos and try out the branched library.

rm-hull, 2017-03-26:
@fullphat - any chance you could try the **rpi_ws281x** branch on your 8x8 unicorn hat?

thijstriemstra, 2017-05-08:
@fullphat any news?

fullphat, 2017-05-10:
sorrysorrysorrysorry - will try to get to this ASAP!

closed 2017-08-16 by rm-hull

fullphat, 2017-08-18:
Apologies all - I've since purchased the HD Hat and have been using the Pimoroni/PIL libraries to run it.  Will try to get back to testing this but I'm rammed right now.