rm-hull/luma.led_matrix #86: software PWM for individual LEDs?

davidchatting, 2017-03-02:
Hi - in theory would it be possible simulate PWM for individual LEDs by repeatably drawing to the display at a high rate and determining if this pixel should be on or off for this cycle?

I have been experimenting with code like this - but it runs much too slowly (like 100 times too slow):

c=canvas(device)    #luma.core.render.canvas
    for threshold in range(255,0,-1):
        with c as display:  #ImageDraw
            for x in range(0, 16):
                for y in range(0, 16):
                    if i[x][y]>=threshold:
                        drawPixel(x,y,display)

Any thoughts? Thanks!

David

rm-hull, 2017-03-02:
It ought be possible, but I would be inclined to pre-render some 8x8px images with the bit patterns you want, and then blast through them in a loop... It would be a lot quicker than drawing onto the nested loops

davidchatting, 2017-03-02:
That's a good idea - let me try that!

thijstriemstra, 2017-03-14:
any news @davidchatting?

closed 2017-04-09 by rm-hull