rm-hull/luma.led_matrix #87: MAX7219 - any way to slow scrolling text down?

fullphat, 2017-03-03:
Raspbian Jessie on a Pi 3 running headless with a MAX7219 8x8 Matrix and single Blink(1) connected.

When doing:

```legacy.show_message(device, text, fill="white", font=font)```

The text rockets across the LED like greased lightning!  Is there any way to slow it down? :)

rm-hull, 2017-03-03:
Yes, although the API docstring hasn't been filled out for that method yet, if you look at the [method signature](https://luma-core.readthedocs.io/en/latest/api-documentation.html#luma.core.legacy.show_message), there is a `scroll_delay` which defaults to 0.03, so you should be able to do:

```python
legacy.show_message(device, text, fill="white", font=font, scroll_delay=0.1)
```

Just make sure you're on the latest version:

   $ sudo -H pip install -U luma.led_matrix

fullphat, 2017-03-03:
Woop! 👍 luma.led_matrix needed the update but it's worked a treat.  A delay of 0.05 seems to be the best fit for me.