rm-hull/luma.led_matrix #93: Manufactured shield (width 16, height 8) is flipped

juri1999, 2017-03-18:
I have bought a ready-to-use-shield with a connector for pi, which has the displays in the wrong order, when I use the option --orientation vertical. The orientation is alright. This means it shows BA instead of AB. Is it possible to switch the order in the library, because i don't want to solder.

rm-hull, 2017-03-18:
Have you got a link for the product?
Not entirely sure what you mean, can you paste a photo into this issue?

The block orientation is there to fix a phase problem, but there is also a `--rotate` option which might work for you. Can you try this with and without the `--block-orientation` to see if that meets your needs.

If it turns out you do need a vertical flip or mirror we can have a look at that afterwards.

juri1999, 2017-03-18:
You can find my board at:
https://www.raspduino.de/shop-right/module/output/optisch/led-dot-matrix-raspberry-pi-detail.html
On the photo you can see a green or blue pcb. In my delivery the two modules are mounted on a red pcb. You can find similar 2-dotmatrix-shields on ebay having pcb with different colours, I don't know whether all the red pcb have the same wiring. 

My Problem is that i need a flip and not a mirror. I would say that it is more a swap.
Example:
/luma.led_matrix/examples# python matrix_demo.py --cascaded 2 --block-orientation vertical
The text is coming first into the left Display, but it should come first into the right Display.
As you can see in my attached video.

The rotation is not helpful, i can turn my raspberry on the table myself as i want.

Could you tell me how to fix this, or if you could do this for me in the library, this would be nice.
As an alternative I would appreciate a link where to buy a compatible 2 matrix shield for Raspberry.

[VID_20170318_131547069.zip](https://github.com/rm-hull/luma.led_matrix/files/852398/VID_20170318_131547069.zip)

rm-hull, 2017-03-18:
Yes, I see the problem. It looks like there was a PCB layout error, but it can be solved relatively easily in software. If you look at this [line of code](https://github.com/rm-hull/luma.led_matrix/blob/master/luma/led_matrix/device.py#L91), it rotates 90° clockwise, whereas for you device it looks like it needs to be corrected 90° anti-clockwise. I will add another flag to cater for this (something like `--block-orientation=flip-vertical`) and build a new release. 

But in the meantime, if you could verify that this proposed fix would work, by altering the file yourself (it will probably be _/usr/local/lib/python2.7/luma/led_matrix/device.py_ or _/usr/local/lib/python3.4/dist-packages/luma/led_matrix/device.py_ depending on which version of python you're on) and try changing the `-90` to just `90` on line 91 and try re-running again with the example:

```
/luma.led_matrix/examples# python matrix_demo.py --cascaded 2 --block-orientation vertical
```
Let me know if it works, and if so I'll proceed with a new build.

juri1999, 2017-03-18:
Yes, this solved the problem. And it is so easy. Thank you very much, i would never have found that out alone (90 instead of -90).
All working now. See attached Video with "90" in device.py:
/luma.led_matrix/examples# python matrix_demo.py --cascaded 2 --block-orientation vertical
[VID_20170318_193842970.zip](https://github.com/rm-hull/luma.led_matrix/files/852661/VID_20170318_193842970.zip)

I don't need your option in a new version right now. Do it whenever you want or leave it.  I can go along with this version.

P.S.:  Maybe in the future some more of my boards will come up. I have found my board somewhere else, it is looking identical to the red one there:
http://www.chipskeytech.com/product/60580852799-801840723/Raspberry_Pi_MAX7219_LED_Dot_Matrix_8_16.html?spm=a2700.8304367.0.0.Nwo81e
On the site is a link with C-Code, a very simple and poor demo and not as perfect as your library, but it is working with my board. So the board has no fault but only a strange design. 
Would be nice to know where you can get a 2-dotmatrix-Shield for Raspberry which is working directly with luma without --block-orientation, but it is not very necessary.

rm-hull, 2017-03-18:
Yes, it is a strange design (this is what I meant when I said it was a PCB layout error) - presumably there were constraints that forced it to be that way. Also, you may find issue https://github.com/rm-hull/luma.led_matrix/issues/69 has some background related to this.

These [MAX7219](http://www.ebay.co.uk/itm/121762381493) modules are laid out without needing --block-orientation.

PR #94 (2017-03-19): Changed block_orientation valid parameters
Used to take “horizontal” or “vertical”, now additionally allows 0, 90, -90.
The strings are marked as deprecated and are no longer documented, but
are still supported up until the next major version.

Fixes #93
closed 2017-03-18 by rm-hull