rm-hull/luma.led_matrix #95: Support for CJMCU 8x8 64 Bit WS2812 5050 RGB LED Driver Development Board for Arduino TE579

javieralapps4up, 2017-03-20:
Recently got 4 of these CJMCU 8x8 64 Bit WS2812 5050 RGB LED Driver Development Board for Arduino TE579. Would be cool to support this in luma.led_matrix.

#### Type of Raspberry Pi 
RPi3 and Zero

#### Linux Kernel version
Linux raspberrypi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux

#### Actual behaviour
No error messages when execute examples but nothing happens

thijstriemstra, 2017-03-20:
Do you have a link to the ebay/site where you purchased it? A PDF datasheet would be essential as well.

rm-hull, 2017-03-20:
The `neopixel` class already supports ws2812 rgb addressable LEDs, so all you need to do is wire it up. The data pin should (i think) go to bcm 18 (the pwm line)

javieralapps4up, 2017-03-20:
thijstriemstra: http://www.ebay.es/itm/272573871294?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

rm-hull: I done it. No results. I cant with unicorn HAT too, may be my fault...

rm-hull, 2017-03-20:
So I should've mentioned these neopixels are very fussy on two levels: 

(a) they have to be driven at 800KHz and precise timing is crucial. I have one of the 8x4 Unicorn pHATs and that _seems_ to be fairly reliable. @fullphat has tried it with an 8x8 Unicorn HAT and things seem to go a bit harywire IIRC - see the discourse in issue #72. Search the internet, but the basic problem is that the RPi linux kernel can't emit timing signals in a guaranteed manner to support large chains of these devices.

(b) The voltage from the PWM pin needs level shifting from 3V3 to 5V. The Unicorn pHAT has some circuitry built in to do this, but I'm guessing the board you bought doesn't. See https://learn.adafruit.com/neopixels-on-raspberry-pi/wiring for details how to do this.

There is a new emerging "superled" called [APA102C](https://cpldcpu.com/2014/08/27/apa102/) which is driven by SPI, so interfacing would be a whole lot simpler.

rm-hull, 2017-03-20:
Also heed the warning from Adafruit:

[image]

fullphat, 2017-03-20:
Sorry, I've been a bit quiet around here due to work and Snarl commitments.  Just to add though that the library Pimoroni point you to has worked flawlessly, so I'm not totally sure if it's a hardware issue.  I do only have a single 8x8 HAT, so I can't say if trying to drive more than one causes problems.

closed 2017-04-09 by rm-hull