rm-hull/luma.led_matrix #100: Support for Scroll pHAT HD

thijstriemstra, 2017-03-28:
* 17x7 matrix
* Uses the IS31FL3731 LED matrix driver chip
* pinout: https://pinout.xyz/pinout/scroll_phat_hd

See https://shop.pimoroni.com/collections/raspberry-pi/products/scroll-phat-hd

[image]

thijstriemstra, 2017-03-28:
I'll be able to get my hands on one later this week and test it.

rm-hull, 2017-03-29:
Looks like it is an i2c device, so should be easy to drive

thijstriemstra, 2017-06-12:
Finally got one on my desk, will test this now.

thijstriemstra, 2017-07-03:
ping @rm-hull

rm-hull, 2017-07-03:
First things first, connect it up, and do `i2cdetect -y 1` and see what address(es) it responds with.

Looking at the datasheet and adafruit arduino code, I'd be inclined to hack a little script first to understand it a bit more before diving into writing the display driver. The hack script should:
* sets up the i2c using smbus2 (use https://github.com/rm-hull/luma.core/blob/master/contrib/tca9548a_scan.py as inspiration)
* initializes the device into picture mode - see https://github.com/adafruit/Adafruit_IS31FL3731/blob/master/Adafruit_IS31FL3731.cpp#L25-L33 and https://github.com/adafruit/Adafruit_IS31FL3731/blob/master/Adafruit_IS31FL3731.cpp#L140
* then add some code that does something like:

```python
for x in range(17):
    for y in range(7):
        bus.write_byte(y, x, 0xFF)
        time.sleep(1)
```

and see what happens - if that worked (in some sense), you should see the LEDs progressively lighting up one by one - and that should give us a clue how to map the display (it will either be a raster scan or snake scan - similar to the WS2812 neopixel mapping on the unicorn HAT - https://github.com/rm-hull/luma.led_matrix/blob/master/luma/led_matrix/device.py#L281-L292)

Once we understand how to address it, writing a luma.led_matrix driver will be pretty simple, and we should be able to convert RGB colored images into greyscale so that the pixels have different brightness levels.

Quite happy to try organising a remote coding session (hangouts / slack / remote desktop 
/ screenhero) ?

dglaude, 2018-01-04:
Is there some status update on this?
I have that hardware and there is a library from @pimoroni .
Do you want a (a) pure independent implementation, (b) it is OK to call their library, (c) it is OK to adapt their code and reuse it (they use MIT license) while removing all the unnecessary part (font, ...)?

Gadgetoid, 2018-08-24:
I most emphatically promote borrowing and re-purposing our code!

If you need any help getting Scroll pHAT HD up and running, I'd be happy to test/review your code and answer any questions you might have.

I'd also love to see support for the Unicorn HAT HD, which uses SPI.

The Scroll pHAT HD pixels aren't addressed in any way that would make sense to standardise, but you can convert X/Y coordinates into an X index into the pixel memory using the `_pixel_addr` method on this class:

```python
class ScrollPhatHD(Matrix):
    width = 17
    height = 7

    def _pixel_addr(self, x, y):
        if x > 8:
            x = x - 8
            y = 6 - (y + 8)
        else:
            x = 8 - x

        return x * 16 + y
```

https://github.com/pimoroni/scroll-phat-hd/blob/1637fce37f9d8c353ef8d658792164e1e1ea02a2/library/scrollphathd/is31fl3731.py#L647-L658

rm-hull, 2019-05-25:
> I'd also love to see support for the Unicorn HAT HD, which uses SPI.

@Gadgetoid - see #188 - wip, but is working

Trotter73, 2026-08-17:
Hi,

Its been a while since I last looked at this, the above patch was basically a wrapper for the Pimoroni driver, I've scrubbed and it now  now implements IS31FL3731 stuff from scratch and is independent of the Pimoroni driver...

I've not tested extensively but it does work, I'll raise a PR to see what you guys think.

Full disclosure I have used Claude to help package this, I've never contributed before and used it go get something that is hopefully close to the format you would want without pestering you too much....