rm-hull/luma.led_matrix #102: ImportError: No module named max7219.led

aocana, 2017-04-07:
#### Type of Raspberry Pi 

raspberry pi model b+

#### Linux Kernel version

linux 4.4.34+ #930 Wed Nov 23 15:12:30 GMT 2016 armv6l GNU/Linux

#### Expected behaviour

hi,
i have this issue executing any python file : No module named max7219.led 

`user@raspberrypi-WEATHER:/scripts/panel-led/max7219$ sudo python setup.py install
running install
running bdist_egg
running egg_info
creating luma.led_matrix.egg-info
writing requirements to luma.led_matrix.egg-info/requires.txt
writing luma.led_matrix.egg-info/PKG-INFO
writing namespace_packages to luma.led_matrix.egg-info/namespace_packages.txt
writing top-level names to luma.led_matrix.egg-info/top_level.txt
writing dependency_links to luma.led_matrix.egg-info/dependency_links.txt
writing manifest file 'luma.led_matrix.egg-info/SOURCES.txt'
reading manifest file 'luma.led_matrix.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
no previously-included directories found matching 'doc/_build'
warning: no previously-included files matching 'pycache' found under directory ''
warning: no previously-included files matching '.py[co]' found under directory ''
warning: no previously-included files matching '' found under directory ''
warning: no previously-included files matching '.coverage' found under directory ''
writing manifest file 'luma.led_matrix.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv6l/egg
running install_lib
running build_py
creating build
creating build/lib.linux-armv6l-2.7
creating build/lib.linux-armv6l-2.7/luma
creating build/lib.linux-armv6l-2.7/luma/led_matrix
copying luma/led_matrix/virtual.py -> build/lib.linux-armv6l-2.7/luma/led_matrix
copying luma/led_matrix/device.py -> build/lib.linux-armv6l-2.7/luma/led_matrix
copying luma/led_matrix/segment_mapper.py -> build/lib.linux-armv6l-2.7/luma/led_matrix
copying luma/led_matrix/helpers.py -> build/lib.linux-armv6l-2.7/luma/led_matrix
copying luma/led_matrix/legacy.py -> build/lib.linux-armv6l-2.7/luma/led_matrix
copying luma/led_matrix/const.py -> build/lib.linux-armv6l-2.7/luma/led_matrix
copying luma/led_matrix/init.py -> build/lib.linux-armv6l-2.7/luma/led_matrix
creating build/bdist.linux-armv6l
creating build/bdist.linux-armv6l/egg
creating build/bdist.linux-armv6l/egg/luma
creating build/bdist.linux-armv6l/egg/luma/led_matrix
copying build/lib.linux-armv6l-2.7/luma/led_matrix/virtual.py -> build/bdist.linux-armv6l/egg/luma/led_matrix
copying build/lib.linux-armv6l-2.7/luma/led_matrix/device.py -> build/bdist.linux-armv6l/egg/luma/led_matrix
copying build/lib.linux-armv6l-2.7/luma/led_matrix/segment_mapper.py -> build/bdist.linux-armv6l/egg/luma/led_matrix
copying build/lib.linux-armv6l-2.7/luma/led_matrix/helpers.py -> build/bdist.linux-armv6l/egg/luma/led_matrix
copying build/lib.linux-armv6l-2.7/luma/led_matrix/legacy.py -> build/bdist.linux-armv6l/egg/luma/led_matrix
copying build/lib.linux-armv6l-2.7/luma/led_matrix/const.py -> build/bdist.linux-armv6l/egg/luma/led_matrix
copying build/lib.linux-armv6l-2.7/luma/led_matrix/init.py -> build/bdist.linux-armv6l/egg/luma/led_matrix
byte-compiling build/bdist.linux-armv6l/egg/luma/led_matrix/virtual.py to virtual.pyc
byte-compiling build/bdist.linux-armv6l/egg/luma/led_matrix/device.py to device.pyc
byte-compiling build/bdist.linux-armv6l/egg/luma/led_matrix/segment_mapper.py to segment_mapper.pyc
byte-compiling build/bdist.linux-armv6l/egg/luma/led_matrix/helpers.py to helpers.pyc
byte-compiling build/bdist.linux-armv6l/egg/luma/led_matrix/legacy.py to legacy.pyc
byte-compiling build/bdist.linux-armv6l/egg/luma/led_matrix/const.py to const.pyc
byte-compiling build/bdist.linux-armv6l/egg/luma/led_matrix/init.py to init.pyc
creating build/bdist.linux-armv6l/egg/EGG-INFO
copying luma.led_matrix.egg-info/PKG-INFO -> build/bdist.linux-armv6l/egg/EGG-INFO
copying luma.led_matrix.egg-info/SOURCES.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
copying luma.led_matrix.egg-info/dependency_links.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
copying luma.led_matrix.egg-info/namespace_packages.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
copying luma.led_matrix.egg-info/not-zip-safe -> build/bdist.linux-armv6l/egg/EGG-INFO
copying luma.led_matrix.egg-info/requires.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
copying luma.led_matrix.egg-info/top_level.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
creating dist
creating 'dist/luma.led_matrix-0.9.0-py2.7.egg' and adding 'build/bdist.linux-armv6l/egg' to it
removing 'build/bdist.linux-armv6l/egg' (and everything under it)
Processing luma.led_matrix-0.9.0-py2.7.egg
removing '/usr/local/lib/python2.7/dist-packages/luma.led_matrix-0.9.0-py2.7.egg' (and everything under it)
creating /usr/local/lib/python2.7/dist-packages/luma.led_matrix-0.9.0-py2.7.egg
Extracting luma.led_matrix-0.9.0-py2.7.egg to /usr/local/lib/python2.7/dist-packages
luma.led-matrix 0.9.0 is already the active version in easy-install.pth

Installed /usr/local/lib/python2.7/dist-packages/luma.led_matrix-0.9.0-py2.7.egg
Processing dependencies for luma.led-matrix==0.9.0
Searching for ws2812==1.0.0
Best match: ws2812 1.0.0
Processing ws2812-1.0.0-py2.7-linux-armv6l.egg
ws2812 1.0.0 is already the active version in easy-install.pth

Using /usr/local/lib/python2.7/dist-packages/ws2812-1.0.0-py2.7-linux-armv6l.egg
Searching for luma.core==0.7.0
Best match: luma.core 0.7.0
Adding luma.core 0.7.0 to easy-install.pth file

Using /usr/local/lib/python2.7/dist-packages
Searching for spidev==3.2
Best match: spidev 3.2
Adding spidev 3.2 to easy-install.pth file

Using /usr/local/lib/python2.7/dist-packages
Searching for Pillow==4.1.0
Best match: Pillow 4.1.0
Adding Pillow 4.1.0 to easy-install.pth file

Using /usr/local/lib/python2.7/dist-packages
Searching for RPi.GPIO==0.6.3
Best match: RPi.GPIO 0.6.3
Adding RPi.GPIO 0.6.3 to easy-install.pth file

Using /usr/lib/python2.7/dist-packages
Searching for monotonic==1.3
Best match: monotonic 1.3
Adding monotonic 1.3 to easy-install.pth file

Using /usr/local/lib/python2.7/dist-packages
Searching for smbus2==0.1.4
Best match: smbus2 0.1.4
Processing smbus2-0.1.4-py2.7.egg
smbus2 0.1.4 is already the active version in easy-install.pth

Using /usr/local/lib/python2.7/dist-packages/smbus2-0.1.4-py2.7.egg
Searching for olefile==0.44
Best match: olefile 0.44
Adding olefile 0.44 to easy-install.pth file

Using /usr/local/lib/python2.7/dist-packages
Finished processing dependencies for luma.led-matrix==0.9.0
`

and after install i try to execute the example:
`
user@raspberrypi-WEATHER:~/scripts/panel-led/raspberrypi-matrix-led-max7219$ sudo python python/matrix-cpu.py Traceback (most recent call last): File "python/matrix-cpu.py", line 4, in <module> import max7219.led as led ImportError: No module named max7219.led
`
this is the code into the python file:

`#!/usr/bin/env python

import os
import max7219.led as led

def getCPUtemperature():
res = os.popen("vcgencmd measure_temp").readline()
return(res.replace("temp=","").replace("'C\n",""))

device = led.matrix(2)
device.orientation(270)

try:
while True:
device.show_message("CPU temperature: " + getCPUtemperature() + "C");
except KeyboardInterrupt:
pass
finally:
device.clear()
`

any help..?

rm-hull, 2017-04-07:
The library has been completely rewritten and it has a different namespace now. The classes are quite different but there is a nod to some legacy methods.

Have a look at the [documentation](http://luma-led-matrix.readthedocs.io/en/latest/python-usage.html) and [examples](https://github.com/rm-hull/luma.led_matrix/blob/master/examples/matrix_demo.py).

aocana, 2017-04-09:
any example..? I totally lost trying to run this example of code clock in my raspberry pi:

https://rpikitchen.wordpress.com/2016/10/30/clock-using-a-max7219-matrix-display/

:-/

rm-hull, 2017-04-09:
Have you tried running the matrix demo example in the project?

Lost how exactly?

The clock code from that blog is using the older version of the library, but it should be pretty simple to convert it to the new style if you look at the imports and how the device is initialised in the matrix demo code vs. the imports and initialisation in the clock code.

aocana, 2017-04-09:
the case is that when i execute any example in my matrix.led  I cannot see anything. Only I can see the led flapping and moving but I'm not able to read or understood the lines.. :-(
this is my connection between raspberry pi and my matrix.led

Pin  led.matrix to - raspberrypi
1	VCC	+5V	- 2	5V0
2	GND	Ground	- 6	GND
3	DIN	Data In	- 19	GPIO 10 (MOSI)
4	CS	Chip Select - 	24	GPIO 8 (SPI CS0)
5	CLK	Clock	- 23	GPIO 11 (SPI CLK)

.---

any help..?
thanks for your support!

rm-hull, 2017-04-09:
Do any of the LED's light up?
Can you upload a photo? (you can drag & drop images into the comment box)

aocana, 2017-04-09:
sure, the first picture is just with led.matrix connected to the raspberry pi.

[image]

and that other picture is the running sudo python matrix_demo.py, from directory luma.led_matrix/examples:

[image]

it's moving but doesn't show nothing clear to read

rm-hull, 2017-04-09:
Looking at the device, it appears to be one of the 4-in-1 LED matrix devices - have you got a URL link where you bought it from?

I would try a couple of things:
* Ensure you have got a good quality 5V USB power supply that provides at least 1A
* Try connecting the VCC to the 3V3 power pin on the RPi instead of the 5V0 pin

Run the example script with the following params:

    $ python examples/matrix_demo.py --cascaded=4 --block-orientation=90

or

    $ python examples/matrix_demo.py --cascaded=4 --block-orientation=-90

aocana, 2017-04-09:
sure, i bought the led matrix in aliexpress:
https://es.aliexpress.com/item/Free-Shipping-MAX7219-Dot-Matrix-Module-For-Arduino-Microcontroller-4-In-One-Display-with-5P-Line/32622349402.html
(more than 40days waiting for this... :-(

I have changed the usb power for other one, but the issue is the same..
I also changed the VCC connection from 5V0 to 3VC and nothing change... :-/

what is the verdict? defective

rm-hull, 2017-04-09:
Are there two 5-pin headers? (i.e. one at each end?)

Make sure you connect the wires to the end that shows as **`--> IN`** 

Something else to check: if you take off one of the LED matrixes from the PCB, and have a look at the chip, it should show something like:

[image]

Does the chip markings show MAX7219CNG or MAX7219ENG ?

aocana, 2017-04-09:
fixed the issue, seems that it was incorrectly connected, i have swapped the pins of leds and now works!!!

thank you so much  @rm-hull for your support!!!!
by the way, any python file with rss and clock..?

rm-hull, 2017-04-09:
Have a look at https://github.com/rm-hull/luma.examples/blob/master/examples/weather.py

You should be able to run it with:

    $ python examples/weather.py -f conf/max7219.conf --block-orientation=90 --width=32

If 90 doesnt work for block orientation, try -90

aocana, 2017-04-09:
thanks!! I will take a look this file...
can i ask  if I have any doubt with this..?

closed 2017-04-09 by thijstriemstra

viktor6, 2018-01-26:
Please make an example of the clock (timedate).