rm-hull/luma.led_matrix #104: ImportError: No module named led_matrix.device

JohannesRegner, 2017-04-14:
#### Type of Raspberry Pi
I use Raspberry Pi Model 2
#### Linux Kernel version
Linux Jonny 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux

Hi, i've installed the library on my raspberry pi, and i get the following error on the examples:
`Traceback (most recent call last):
  File "examples/matrix_demo.py", line 10, in <module>
    from luma.led_matrix.device import max7219
ImportError: No module named led_matrix.device
`

More informations:
lsmod | grep -i spi
spi_bcm2708             5038  0

pip --version
pip 9.0.1 from /usr/local/lib/python2.7/dist-packages (python 2.7)

python setup.py install
`running install
running bdist_egg
running egg_info
creating luma.led_matrix.egg-info
writing requirements to luma.led_matrix.egg-info/requires.txt
writing luma.led_matrix.egg-info/PKG-INFO
writing namespace_packages to luma.led_matrix.egg-info/namespace_packages.txt
writing top-level names to luma.led_matrix.egg-info/top_level.txt
writing dependency_links to luma.led_matrix.egg-info/dependency_links.txt
writing manifest file 'luma.led_matrix.egg-info/SOURCES.txt'
reading manifest file 'luma.led_matrix.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
no previously-included directories found matching 'doc/_build'
warning: no previously-included files matching '__pycache__' found under directory '*'
warning: no previously-included files matching '*.py[co]' found under directory '*'
warning: no previously-included files matching '*~' found under directory '*'
warning: no previously-included files matching '.coverage' found under directory '*'
writing manifest file 'luma.led_matrix.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
creating build
creating build/lib.linux-armv7l-2.7
creating build/lib.linux-armv7l-2.7/luma
creating build/lib.linux-armv7l-2.7/luma/led_matrix
copying luma/led_matrix/__init__.py -> build/lib.linux-armv7l-2.7/luma/led_matrix
copying luma/led_matrix/virtual.py -> build/lib.linux-armv7l-2.7/luma/led_matrix
copying luma/led_matrix/legacy.py -> build/lib.linux-armv7l-2.7/luma/led_matrix
copying luma/led_matrix/device.py -> build/lib.linux-armv7l-2.7/luma/led_matrix
copying luma/led_matrix/segment_mapper.py -> build/lib.linux-armv7l-2.7/luma/led_matrix
copying luma/led_matrix/const.py -> build/lib.linux-armv7l-2.7/luma/led_matrix
copying luma/led_matrix/helpers.py -> build/lib.linux-armv7l-2.7/luma/led_matrix
creating build/bdist.linux-armv7l
creating build/bdist.linux-armv7l/egg
creating build/bdist.linux-armv7l/egg/luma
creating build/bdist.linux-armv7l/egg/luma/led_matrix
copying build/lib.linux-armv7l-2.7/luma/led_matrix/__init__.py -> build/bdist.linux-armv7l/egg/luma/led_matrix
copying build/lib.linux-armv7l-2.7/luma/led_matrix/virtual.py -> build/bdist.linux-armv7l/egg/luma/led_matrix
copying build/lib.linux-armv7l-2.7/luma/led_matrix/legacy.py -> build/bdist.linux-armv7l/egg/luma/led_matrix
copying build/lib.linux-armv7l-2.7/luma/led_matrix/device.py -> build/bdist.linux-armv7l/egg/luma/led_matrix
copying build/lib.linux-armv7l-2.7/luma/led_matrix/segment_mapper.py -> build/bdist.linux-armv7l/egg/luma/led_matrix
copying build/lib.linux-armv7l-2.7/luma/led_matrix/const.py -> build/bdist.linux-armv7l/egg/luma/led_matrix
copying build/lib.linux-armv7l-2.7/luma/led_matrix/helpers.py -> build/bdist.linux-armv7l/egg/luma/led_matrix
byte-compiling build/bdist.linux-armv7l/egg/luma/led_matrix/__init__.py to __init__.pyc
byte-compiling build/bdist.linux-armv7l/egg/luma/led_matrix/virtual.py to virtual.pyc
byte-compiling build/bdist.linux-armv7l/egg/luma/led_matrix/legacy.py to legacy.pyc
byte-compiling build/bdist.linux-armv7l/egg/luma/led_matrix/device.py to device.pyc
byte-compiling build/bdist.linux-armv7l/egg/luma/led_matrix/segment_mapper.py to segment_mapper.pyc
byte-compiling build/bdist.linux-armv7l/egg/luma/led_matrix/const.py to const.pyc
byte-compiling build/bdist.linux-armv7l/egg/luma/led_matrix/helpers.py to helpers.pyc
creating build/bdist.linux-armv7l/egg/EGG-INFO
copying luma.led_matrix.egg-info/PKG-INFO -> build/bdist.linux-armv7l/egg/EGG-INFO
copying luma.led_matrix.egg-info/SOURCES.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying luma.led_matrix.egg-info/dependency_links.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying luma.led_matrix.egg-info/namespace_packages.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying luma.led_matrix.egg-info/not-zip-safe -> build/bdist.linux-armv7l/egg/EGG-INFO
copying luma.led_matrix.egg-info/requires.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying luma.led_matrix.egg-info/top_level.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
creating dist
creating 'dist/luma.led_matrix-0.9.0-py2.7.egg' and adding 'build/bdist.linux-armv7l/egg' to it
removing 'build/bdist.linux-armv7l/egg' (and everything under it)
Processing luma.led_matrix-0.9.0-py2.7.egg
removing '/usr/local/lib/python2.7/dist-packages/luma.led_matrix-0.9.0-py2.7.egg' (and everything under it)
creating /usr/local/lib/python2.7/dist-packages/luma.led_matrix-0.9.0-py2.7.egg
Extracting luma.led_matrix-0.9.0-py2.7.egg to /usr/local/lib/python2.7/dist-packages
luma.led-matrix 0.9.0 is already the active version in easy-install.pth

Installed /usr/local/lib/python2.7/dist-packages/luma.led_matrix-0.9.0-py2.7.egg
Processing dependencies for luma.led-matrix==0.9.0
Searching for ws2812==1.0.0
Best match: ws2812 1.0.0
Processing ws2812-1.0.0-py2.7-linux-armv7l.egg
ws2812 1.0.0 is already the active version in easy-install.pth

Using /usr/local/lib/python2.7/dist-packages/ws2812-1.0.0-py2.7-linux-armv7l.egg
Searching for luma.core==0.7.5
Best match: luma.core 0.7.5
Adding luma.core 0.7.5 to easy-install.pth file

Using /usr/local/lib/python2.7/dist-packages
Searching for smbus2==0.1.4
Best match: smbus2 0.1.4
Processing smbus2-0.1.4-py2.7.egg
smbus2 0.1.4 is already the active version in easy-install.pth

Using /usr/local/lib/python2.7/dist-packages/smbus2-0.1.4-py2.7.egg
Searching for RPi.GPIO==0.6.3
Best match: RPi.GPIO 0.6.3
Processing RPi.GPIO-0.6.3-py2.7-linux-armv7l.egg
RPi.GPIO 0.6.3 is already the active version in easy-install.pth

Using /usr/local/lib/python2.7/dist-packages/RPi.GPIO-0.6.3-py2.7-linux-armv7l.egg
Searching for monotonic==1.3
Best match: monotonic 1.3
Processing monotonic-1.3-py2.7.egg
monotonic 1.3 is already the active version in easy-install.pth

Using /usr/local/lib/python2.7/dist-packages/monotonic-1.3-py2.7.egg
Searching for Pillow==4.1.0
Best match: Pillow 4.1.0
Adding Pillow 4.1.0 to easy-install.pth file

Using /usr/local/lib/python2.7/dist-packages
Searching for spidev==3.3
Best match: spidev 3.3
Adding spidev 3.3 to easy-install.pth file

Using /usr/local/lib/python2.7/dist-packages
Searching for olefile==0.44
Best match: olefile 0.44
Adding olefile 0.44 to easy-install.pth file

Using /usr/local/lib/python2.7/dist-packages
Finished processing dependencies for luma.led-matrix==0.9.0
`
In the dir /usr/local/lib/python2.7/dist-packages/
Are the following entries:
luma
luma.core-0.7.5-nspkg.pth
luma.core-0.7.5.dist-info
luma.led_matrix-0.9.0-py2.7.egg

What i am doing wrong??

rm-hull, 2017-04-14:
@JohannesRegner - The library has been completely rewritten and it has a different namespace now. The classes are quite different but there is a nod to some legacy methods.

Have a look at the [documentation](http://luma-led-matrix.readthedocs.io/en/latest/python-usage.html) and [examples](https://github.com/rm-hull/luma.led_matrix/blob/master/examples/matrix_demo.py).

rm-hull, 2017-04-14:
Sorry, I misread your description and attributed the same cause as #20 - looks like your problem is actually different.

Did you install with `pip` or `python setup.py install` ? 

I would be inclined to try:

```
$ sudo rm -rf /usr/local/lib/python2.7/dist-packages/luma*
$ sudo -H pip install -U luma.led_matrix
```

And then try the example again

thijstriemstra, 2017-04-14:
what is the output of 
```
$ python --version
```

JohannesRegner, 2017-04-15:
Hi, 
```
 python --version
Python 2.7.3
```

But now it works! 
I have to do your tipp Richard! @rm-hull 

```
$ sudo rm -rf /usr/local/lib/python2.7/dist-packages/luma*
$ sudo -H pip install -U luma.led_matrix
```

Thank you very much! Saved my day 👍

closed 2017-04-15 by JohannesRegner

lovepianist, 2017-10-24:
Hi Richard@rm-hull 

I have the same problem as Johannes did. When I tried to type 
$ sudo rm -rf /usr/local/lib/python2.7/dist-packages/luma*
$ sudo -H pip install -U luma.led_matrix

I get "sudo: pip: command not found". Could you please explain to me what that means and how to fix this? Thank you.

rm-hull, 2017-10-24:
Have a look at the documentation,bespecially this section: http://luma-led-matrix.readthedocs.io/en/latest/install.html#installing-from-pypi

It shows you how to install pip and all the other pre-requisites.

lovepianist, 2017-10-24:
Thank you, Richard!

lmsb, 2018-07-21:
I have same problem but when running 
sudo -H pip install -U luma.led_matrix
I get lots of warnings and almost at the end a problem of not having zlib
"The headers or library files could not be found for zlib,
    a required dependency when compiling Pillow from source.

    Please see the install instructions at:
       https://pillow.readthedocs.io/en/latest/installation.html"

Can you help me?  

Complete output: 

root@linaro-alip ~/max7219# sudo -H pip install -U luma.led_matrix
Collecting luma.led_matrix
  Using cached https://files.pythonhosted.org/packages/af/b5/d51635d462b56be01fa57e5d60a1d0043d4e38500dabdbc6f2c94c9a1cfa/luma.led_matrix-1.0.8-py2.py3-none-any.whl
Collecting luma.core>=1.2.1 (from luma.led_matrix)
  Using cached https://files.pythonhosted.org/packages/1b/54/0b99b02a3f862e23084421628839bac33617f46920a6d1e0f8bb74fc880b/luma.core-1.7.2-py2.py3-none-any.whl
Collecting pillow>=4.0.0 (from luma.core>=1.2.1->luma.led_matrix)
  Using cached https://files.pythonhosted.org/packages/d3/c4/b45b9c0d549f482dd072055e2d3ced88f3b977f7b87c7a990228b20e7da1/Pillow-5.2.0.tar.gz
Requirement already up-to-date: RPI.GPIO; platform_system == "Linux" in /usr/local/lib/python2.7/dist-packages/RPi.GPIO-0.6.3-py2.7-linux-aarch64.egg (from luma.core>=1.2.1->luma.led_matrix)
Requirement already up-to-date: monotonic; python_version < "3.3" in /usr/local/lib/python2.7/dist-packages/monotonic-1.5-py2.7.egg (from luma.core>=1.2.1->luma.led_matrix)
Requirement already up-to-date: smbus2 in /usr/local/lib/python2.7/dist-packages/smbus2-0.2.1-py2.7.egg (from luma.core>=1.2.1->luma.led_matrix)
Requirement already up-to-date: spidev; platform_system == "Linux" in /usr/local/lib/python2.7/dist-packages (from luma.core>=1.2.1->luma.led_matrix)
Building wheels for collected packages: pillow
  Running setup.py bdist_wheel for pillow ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-4GG5DI/pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpep6e5Fpip-wheel- --python-tag cp27:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-aarch64-2.7
  creating build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/ImageColor.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/ImageTk.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/JpegImagePlugin.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/ImageEnhance.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/ContainerIO.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/WebPImagePlugin.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/MpoImagePlugin.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/McIdasImagePlugin.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/PngImagePlugin.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/PalmImagePlugin.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/PsdImagePlugin.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/ImageCms.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/WalImageFile.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/GbrImagePlugin.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/GdImageFile.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/ImtImagePlugin.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/_util.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/PcfFontFile.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/FpxImagePlugin.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/ImageDraw.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/Jpeg2KImagePlugin.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/BufrStubImagePlugin.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/features.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/ExifTags.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/_tkinter_finder.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/SpiderImagePlugin.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/ImageMode.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/XbmImagePlugin.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/ImagePalette.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/ImageSequence.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/ImImagePlugin.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/BlpImagePlugin.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/GifImagePlugin.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/GimpPaletteFile.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/Hdf5StubImagePlugin.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/ImageFont.py -> build/lib.linux-aarch64-2.7/PIL
  copying src/PIL/PaletteFile.py -> build/lib.linux-aarch64-2.7/PIL
[... 225 lines cut ...]
    writing manifest file 'src/Pillow.egg-info/SOURCES.txt'
    running build_ext

    The headers or library files could not be found for zlib,
    a required dependency when compiling Pillow from source.

    Please see the install instructions at:
       https://pillow.readthedocs.io/en/latest/installation.html

    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-4GG5DI/pillow/setup.py", line 807, in <module>
        raise RequiredDependencyException(msg)
    __main__.RequiredDependencyException:

[... cut ...]
    ----------------------------------------
  Can't rollback pillow, nothing uninstalled.
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-4GG5DI/pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-vEmpLm-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-4GG5DI/pillow/
root@linaro-alip ~/max7219#

lmsb, 2018-07-22:
Just fixed my previous issue. I found the instructions on how to install on read the docs "https://luma-led-matrix.readthedocs.io/en/latest/install.html" and applied the next missing libraries
`sudo apt-get install build-essential libfreetype6-dev`
After that I could run the luma install line without error. 
`sudo -H pip install -U luma.led_matrix`