rm-hull/luma.led_matrix #106: Seven segment Neopixel display

msurguy, 2017-06-07:
Hello! 

I am creating a seven segment display based on WS2812 LEDs and was wondering if you could point me in the right direction as far as creating code compatible with your library to drive the LEDs on a Pi Zero. 

So far I got the lib to work with my neopixels and can pass in a mapping similar to Unicorn HAT but now am stuck on creating a font mapping and using it in the library. 

Any help or guidance in what mappings, font definitions I can use would be appreciated.

Thanks!

rm-hull, 2017-06-07:
Sounds interesting!

Are you contemplating putting a single neopixel per segment, like this:

[image]

If you did, then it would follow the same scheme as the MAX7219 seven segment (here's the relevant bit of the data sheet):

[image]

I guess if you laid them out sequentially, A => B => C => D, etc. then you might be able to use the same `sevensegment` wrapper class (see http://luma-core.readthedocs.io/en/latest/api-documentation.html#luma.core.virtual.sevensegment) with a custom mapper. 

You'd then be able to say

```python
sevenseg.text = "1234"
```

I'm sure with a bit of refinement, that class could be made to handle ANSI color codes, which could be translated into the correct Neopixel colors.

msurguy, 2017-06-07:
@rm-hull thanks for your quick reply! 

Yes, there is single LED per segment and they look like this when put all together:
[image]

The mapping is a bit different than the one you have in the datasheet, the order(index of each LED) is a follows:

```
	5555	
4			6
4			6
4			6
	3333
0			2
0			2
0			2
	1111
```

I'm pretty new to your library so I have a hard time understanding the sequence that I have to follow in defining / using the Core classes + Led Matrix classes and functions.

rm-hull, 2017-06-10:
That looks pretty amazing - is that something you've bought or assembled yourself?

So it looks like you have 7 neopixels per digit whereas the `sevensegment` class expects 8 LEDs per digit. 

Something like the below code might work (untested, though)

```python
import time
from luma.led_matrix.device import ws2812
from luma.core.virtual import sevensegment

_DIGITS = {
  # Think this corresponds to the bit positions 
  ' ': 0b00000000,
  '0': 0b11101110,
  '1': 0b00100010,
  '2': 0b11010110,
  # etc
}

def neopixel_sevenseg_mapper(text, not_found="_"):
    try:
        undefined = _DIGITS[notfound]
        iterator = iter(text)
        while True:
            char = next(iterator)
            yield _DIGITS.get(char, undefined)
    except StopIteration:
        pass

device = ws2812(width=6, height=7)
sevenseg = sevensegment(device, segment_mapper=neopixel_sevenseg_mapper)

count = 0
while True:
    sevenseg.text = "{0:06d}".format(count);
    count += 1
    time.sleep(1)
```

If that works, it would neopixls would illuminate in white, we could follow up and try and make it work in color (there would have to be some change to *luma.core* to make this work though - it shouldn't be too difficult though, as we've done something very similar with the terminal class). 

If you're able to source one of the units or provide a link to where you got them then I'd be happy to help implement the color model.

msurguy, 2017-06-12:
@rm-hull Thank you for a detailed write up! I'm going to try doing this very soon and will update you on the progress.

These displays are something I've been working on by myself and are a part of a crowdfunding campaign that is launching soon: https://www.crowdsupply.com/maksmakes/neosegment. I could send you a couple if you can help make this library work with the Neosegment, perhaps adding some animations and transitions :) Send your address to msurguy@gmail.com and I'll send out some units in the next two weeks.

I was researching various libraries for the Pi that could make simulation and animation a little simpler than creating my own from scratch. So far Bibliopixel and your library are my best bets.

thijstriemstra, 2017-07-03:
@msurguy any news?

msurguy, 2017-07-03:
@thijstriemstra The campaign is now live, you can back the project if you want a ready to go product. I will be sending a few units to @rm-hull shortly to be able to collaborate on the luma.led_matrix extension.

rm-hull, 2017-07-14:
Received a package in the post today, so will keep this ticket updated with progress

[image]

msurguy, 2017-07-14:
@rm-hull great! I will post up an Arduino library that I wrote so that you can see my existing code -> 
[DemoCode.zip](https://github.com/rm-hull/luma.led_matrix/files/1149153/DemoCode.zip)

rm-hull, 2017-07-15:
Theres some code in progress on branch [feature/neosegment](https://github.com/rm-hull/luma.led_matrix/tree/feature/neosegment) - see [here](https://github.com/rm-hull/luma.led_matrix/blob/feature/neosegment/luma/led_matrix/device.py#L402) for the main driver class.

It just extends the `sevensegment` class, so you can write some code like:

```python
from luma.led_matrix.device import neosegment

sevenseg = neosegment(width=6)
sevenseg.set_color("red")
sevenseg.text = "HELLO"
sevenseg.device.contrast(65)
```

Still thinking about how to be able to set individual character colors, would actually like it to be:

```python
sevenseg.color = "red"
sevenseg.color[3:5] = "blue"
```

i.e. set every char to red, then use pythons slicing to set only chars 3 and 4 to blue.

Then there's docs & tests still to add, but pretty pleased with how the code has turned out so far.

I originally tried this on an RPi B2, and it didn't work, but it does on a Pi zero. This may be down to the underlying C library it uses to interface to the WS2812s, and is possibly related to issues discussed in #72. Will investigate this further, but I'm certain that this is not specific to NeoSegment, and is actually probably down to the current lib not knowing that different RPi models have different DMA locations.

**They are great devices BTW: excellent build quality & I really like the way they slot together!**

rm-hull, 2017-07-21:
Closing this issue, any questions or remarks please reopen.

I might do some more work on the neopixel implementation, as the library it uses is not python3 compatible, and there is no easy way to hook into the cleanup routine.

closed 2017-07-21 by rm-hull