rm-hull/luma.led_matrix #110: Index out of Range.

jboyoli, 2017-06-28:
Hi
I'm writing a script to parse an rSS feed and show the news on the display but randomly I am getting the following error message and I don't know what can be the cause, I'm using spanish RSS feeds, maybe a character is out of the 256 ascii chars that are in the font file? How can I solve?
Thank you.

 File "rss-feed.py", line 53, in main
    show_message(device, msg, fill="white", font=proportional(SINCLAIR_FONT))
  File "/usr/local/lib/python2.7/dist-packages/luma/core/legacy/__init__.py", line 77, in show_message
    w, h = textsize(msg, font)
  File "/usr/local/lib/python2.7/dist-packages/luma/core/legacy/__init__.py", line 29, in textsize
    src = [c for ascii_code in txt for c in font[ord(ascii_code)]]
  File "/usr/local/lib/python2.7/dist-packages/luma/core/legacy/font.py", line 21, in __getitem__
    bitmap = self.font[asciiCode]
IndexError: list index out of range

rm-hull, 2017-06-28:
Yes, it sounds like you have some UTF8 / unicode in the feed that is outside the range of the ASCII character set.

You could try something like (if your string is in a variable called `text`):

```python
text = re.sub(r'[^\x00-\x7F]+',' ', text)
```

to replace any non-ascii characters with a space, but then this may not be very acceptable if there are accented characters you want to show.

Alternatively, perhaps try:

```
text.encode('latin-1')
```

to try and convert the UTF8 / unicode into the latin character set.

If you let me know what the RSS feed URL is I will have a further look into this.

jboyoli, 2017-06-28:
Thanks for the fast reply.
It appears randomly on other feeds but I found that this one specifically produces the error just after printing fine the phrase: “Siempre es más fácil obtener perdón que permiso” the accented a prints as blank using the font LCD and that is fine as I still have to edit the bytes to add the character to the font.
http://www.bbc.com/mundo/temas/cultura/index.xml
I'm using feed parser wich prints fine on the monitor but when trying to display it on the matrix it fails.

This is another one that produces the same error but not always as this is updated every 20 mins I think.
http://www.eluniversal.com.mx/rss.xml

Thanks for looking in to it, much appreciated.

rm-hull, 2017-06-30:
Your other option would be to use a TTF font that supports accented characters. Have a look at the [tweet_scroll.py](https://github.com/rm-hull/luma.examples/blob/master/examples/tweet_scroll.py) demo to see how that does horizontal scrolling and handles unicode. 

The only thing is you would need to find a font which renders in 8px height nicely...

In the meantime, I'm going to add another wrapper (called `tolerant`) into luma.core so you can do stuff like:

font = proportional(tolerant(LCD_FONT, missing="?"))

and if it encounters a character code it doesn't know about, it will replace it with whatever `missing` (which is optional) is set to.

rm-hull, 2017-07-01:
The [`tolerant`](http://luma-core.readthedocs.io/en/latest/_modules/luma/core/legacy/font.html#tolerant) class is now available in luma.core 0.9.4

thijstriemstra, 2017-07-03:
@jboyoli does the `tolerant` class fix your issue?

jboyoli, 2017-07-06:
Thank you both for having a look at this and provide the tolerant class, I haven't had the chance to test  it but I'm confident that it will solve the issue. However, I'm unsure on how to upgrade luma.core to 0.9.4, sorry :(

rm-hull, 2017-07-06:
There is a new release now (0.9.5) - you should be able to upgrade to the latest with:

    $ sudo -H pip install --upgrade luma.core

jboyoli, 2017-07-11:
I tested the tolerant class today and it worked fine.  My Rss feeds are being displayed complete now.
I will add some bitmap definitions to the fonts I want to use to the accented characters.
Thanks again for your time.

closed 2017-07-11 by jboyoli