rm-hull/luma.led_matrix #111: Question about font.py

jboyoli, 2017-07-12:
Quick question.
I'm trying to modify the bit patterns file  font.py on my raspberry located in "/usr/local/lib/python2.7/dist-packages/luma/core/legacy/font.py"  but it doesn't matter what I do in the file when I run my py program to display an rss feed the characters are not modified on the screen, seems like I;m not editing the file used to display the message on the matrix as even if I remove the fonts.py file from that folders if I run my scrolling message program it displays without errors.
Can you please tell me where the "live" font.py file is located so I can customize the characters or add any other font?

Thanks in advance.

rm-hull, 2017-07-12:
I think the problem is that while that file exists, there will be a [compiled `.pyc` file](https://stackoverflow.com/questions/2998215/if-python-is-interpreted-what-are-pyc-files) which is used instead, so no matter what you edit the `.py` file to, your changes wont be loaded.

I would recommend that you can just create a file called **myfont.py** (or something, in the same directory as your program) with your custom font definition called `MY_FONT`, and then in the program:

```python
from myfont import MY_FONT

...
show_message(device, msg, fill="white", font=MY_FONT)
```

Hence, shouldn't need to alter the existing font.py in luma.core.

jboyoli, 2017-07-13:
You were right, the .pyc file was the one causing this. I followed your recommendation and I created another file with my font definitions and it is working fine.
Thanks for your fast and precise answers!

closed 2017-07-14 by jboyoli