rm-hull/luma.led_matrix #116: matrix.pixel

Nennstiel, 2017-07-31:
My problem is that i'm using an old tutorial from 08.2016.
The code im using looks like this:

```
from luma.core.interface.serial import spi, noop
from luma.led_matrix.device import max7219
from random import randint
import time

serial = spi(port=0, device=0, gpio=noop())
matrix = max7219(serial, cascaded=3, width=24, hight=8)

while True:
     for i in range(8):
       for j in range(8):
          matrix.pixel(i,j,randint(0,1))
          time.sleep(0.0002)
```

The error I receive is:
AttributeError: 'max7219' object has no attribute 'pixel'

rm-hull, 2017-07-31:
So the library have evolved (changed) quite a lot - instead of providing drawing primitives directly, it takes a Pillow image (or exposes a Pillow canvas) onto which you can do the drawing. I would encourage you to look at the docs for that: https://pillow.readthedocs.io/en/4.2.x/reference/ImageDraw.html

To get something similar to your program above, try (I havent tested this so there might be typos)

```python
from luma.core.interface.serial import spi, noop
from luma.led_matrix.device import max7219
from luma.core.render import canvas
from random import randint
import time

serial = spi(port=0, device=0, gpio=noop())
device = max7219(serial, cascaded=1)

while True:
    with canvas(device) as draw:
        for i in range(8):
           for j in range(8):
               if randint(0, 1) == 1:
                   draw.point((i, i), fill="white")
          
    time.sleep(0.0002)
```

Have a look at the docs (https://luma-led-matrix.readthedocs.io/en/latest/python-usage.html) and if anything is confusing or not clear in those docs, please raise another issue.

Nennstiel, 2017-08-07:
After a few test runs and some changes i came to this result
```

from luma.core.interface.serial import spi, noop
from luma.led_matrix.device import max7219
from luma.core.render import canvas
from random import randint
import time

serial = spi(port=0, device=0, gpio=noop())
device = max7219(serial, cascaded=1)

while True:
        with canvas(device) as draw:
                for i in range(8):
                        for j in range(8):
                                if randint(0, 1) == 0:
                                        draw.point((i,  j), fill="white")
        time.sleep(0.2)
```
thank you for your help