rm-hull/luma.led_matrix #118: Python3 version can't be installed on RPi3 and Raspbian

Bartyx, 2017-08-05:
Hey.
I am following installation instructions, but the step: "Proceed to install latest version of the library directly from PyPI: $ sudo -H pip install --upgrade luma.led_matrix" does not work.

Here is a terminal:
pi@raspberrypi:~ $ sudo -H pip3 install --upgrade luma.led_matrix
Collecting luma.led_matrix
  Downloading luma.led_matrix-1.0.0.tar.gz (10.3MB)
    100% |████████████████████████████████| 10.3MB 13kB/s
    Complete output from command python setup.py egg_info:
    error in luma.led_matrix setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers; Invalid requirement, parse error at "'-'"

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-u7akoewq/luma.led-matrix/
pi@raspberrypi:~ $

Can you fix it please? Thanks.

thijstriemstra, 2017-08-05:
@Bartyx you need to upgrade your setuptools version as well. If it's not in the docs yet, it should be.

```
sudo -H pip3 install -U setuptools
```

thijstriemstra, 2017-08-05:
Yes it's in the docs, please follow the instructions before opening a ticket: https://luma-led-matrix.readthedocs.io/en/latest/install.html#installing-from-pypi

thijstriemstra, 2017-08-05:
@rm-hull or should we place that setuptools check in setup.py for this project as well?

Bartyx, 2017-08-05:
@thijstriemstra I did it all! Of course i did. I followed instructions step-by-step, but it's not working. :(

Bartyx, 2017-08-05:
Look at this:
`pi@raspberrypi:~ $ sudo -i pip install --upgrade pip setuptools

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

Requirement already up-to-date: pip in /usr/local/lib/python3.4/dist-packages
Requirement already up-to-date: setuptools in /usr/local/lib/python3.4/dist-packages
pi@raspberrypi:~ $ sudo -i pip3 install --upgrade pip setuptools

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

Requirement already up-to-date: pip in /usr/local/lib/python3.4/dist-packages
Requirement already up-to-date: setuptools in /usr/local/lib/python3.4/dist-packages
pi@raspberrypi:~ $ sudo -H pip3 install --upgrade luma.led_matrix
Collecting luma.led_matrix
  Using cached luma.led_matrix-1.0.0.tar.gz
    Complete output from command python setup.py egg_info:
    error in luma.led_matrix setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers; Invalid requirement, parse error at "'-'"

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-k92hdf3i/luma.led-matrix/
pi@raspberrypi:~ $`

rm-hull, 2017-08-05:
I think there is a typo in the setup here: https://github.com/rm-hull/luma.led_matrix/blob/master/setup.py#L30

the package name should be `rpi_ws281x` rather than `rpi-ws281x`

I will rebuild and retest

closed 2017-08-05 by rm-hull

rm-hull, 2017-08-05:
Just released **luma.led_matrix 1.0.2** with a fix for this (don't ask, ... too embarrassing!) and installed it on a RPi B2 on both py2.7 and py3.4 successfully - let me know if you hit any problems. 

Thanks for taking the time to submit a bug report.

Bartyx, 2017-08-06:
It works fine now. Thanks for the fix.

thijstriemstra, 2017-08-06:
I wonder why this wasn't caught by Travis.

commit 4839aff7 (2017-08-05): Change install_deps: python3 objects to hyphens in package names

fixes #118