rm-hull/luma.led_matrix #121: 'proportional' object has no attribute 'getmask'

tarasius, 2017-08-17:
**pip**
```
Pillow (4.2.1)
luma.core (1.0.0)
luma.emulator (0.2.4)
luma.led-matrix (1.0.0)
```

**uname -a**
`Linux raspberrypi 4.9.39-v7+ #1021 SMP Mon Jul 24 11:32:18 BST 2017 armv7l GNU/Linux`

**Code**
```
# txt is an array of lines to show 
with canvas(virtual) as draw:
	for idx, item in enumerate(txt):
		item=item.strip()
		draw.text((0, (idx*8)-1), item, fill="white", font=proportional(LCD_FONT))
		
for idx, item in enumerate(txt):
	print(str(idx)+' = ' + item)
	maxx=len(item)*8
	for x in range(maxx):
		virtual.set_position((x, idx*8))
		time.sleep(DELAY)
```

**Traceback (most recent call last):**
```
  File "testweather.py", line 119, in <module>
    draw.text((0, (idx*8)-1), item, fill="white", font=proportional(LCD_FONT))
  File "/usr/local/lib/python2.7/dist-packages/PIL/ImageDraw.py", line 221, in text
    mask = font.getmask(text, self.fontmode, *args, **kwargs)
AttributeError: 'proportional' object has no attribute 'getmask'
```

tarasius, 2017-08-17:
If remove 'proportional' then
```
Traceback (most recent call last):
  File "testweather.py", line 119, in <module>
    draw.text((0, (idx*8)-1), item, fill="white", font=LCD_FONT)
  File "/usr/local/lib/python2.7/dist-packages/PIL/ImageDraw.py", line 221, in text
    mask = font.getmask(text, self.fontmode, *args, **kwargs)
AttributeError: 'list' object has no attribute 'getmask'

```

rm-hull, 2017-08-18:
If you want to use the legacy fonts then you must also use the legacy text draw methods, not the pillow methods. See https://github.com/rm-hull/luma.led_matrix/blob/master/examples/matrix_demo.py

```python
from Luna.core.legacy import text

...

text(draw, (0, (idx*8)-1), item, fill="white", font=proportional(LCD_FONT))
```

closed 2017-09-05 by thijstriemstra

Father001, 2022-05-12:
> If you want to use the legacy fonts then you must also use the legacy text draw methods, not the pillow methods. See https://github.com/rm-hull/luma.led_matrix/blob/master/examples/matrix_demo.py
> 
> ```python
> from luma.core.legacy import text
> 
> ...
> 
> text(draw, (0, (idx*8)-1), item, fill="white", font=proportional(LCD_FONT))
> ```

thanks