rm-hull/luma.led_matrix #123: error when installing library

3zra, 2017-09-17:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

rm-hull, 2017-09-17:
So the actual error is `unable to execute 'arm-linux-gnueabihf-gcc': No such file or directory` -
 basically there is no C compiler installed on your RPi. This sounds like **python-dev** was not installed correctly (it should ensure that all the compiler dependencies were installed). What is the output of:

    $ apt show python-dev

And then can you retry this:

    $ sudo apt-get install python-dev libfreetype6-dev libjpeg-dev
    $ sudo -H pip install --upgrade luma.led_matrix

3zra, 2017-09-17:
Thank you rm-hull for the quick reply. From what I can see, python-dev seems to be installed correctly:
```
osmc@osmc:~$ apt show python-dev
Package: python-dev
Source: python-defaults
Version: 2.7.9-1
Installed-Size: 46,1 kB
Maintainer: Matthias Klose <doko@debian.org>
Depends: python (= 2.7.9-1), libpython-dev (= 2.7.9-1), python2.7-dev (>= 2.7.9-1~)
Homepage: http://www.python.org/
Tag: devel::lang:python, devel::library, implemented-in::python,
 role::devel-lib
Section: python
Priority: optional
Download-Size: 1.180 B
APT-Manual-Installed: yes
APT-Sources: http://ftp.debian.org/debian/ jessie/main armhf Packages
Description: Header-Dateien und eine statische Bibliothek für Python (Standard)
 Header-Dateien, eine statische Bibliothek und Entwicklungswerkzeuge zum
 Erstellen von Python-Modulen, Erweitern des Python-Interpreters oder
 Einbetten von Python in Anwendungen.
 .
 Dieses Paket enthält lediglich eine Abhängigkeit zu Debians Standard-Python-
 Version (derzeit 2.7).
```
Sorry for using a german OS. Let me know if something is unclear. After that, when retrying:
```
$ sudo apt-get install python-dev libfreetype6-dev libjpeg-dev
$ sudo -H pip install --upgrade luma.led_matrix
```
the error `unable to execute 'arm-linux-gnueabihf-gcc': No such file or directory` remains. 

Any further ideas?

thijstriemstra, 2017-09-17:
@3zra what about `sudo apt-get install build-essential`?

3zra, 2017-09-18:
@thijstriemstra installing `build-essential` did the job. Now, the installation of the `luma.led_matrix` libraries works. Thanks a lot.

closed 2017-09-18 by 3zra

thijstriemstra, 2017-09-18:
Awesome. Now we'll have to fix the documentation across projects I think @rm-hull?

rm-hull, 2017-09-18:
Yep... although I wonder if we should actually expend some effort in unifying the docs into https://github.com/rm-hull/luma.docs