rm-hull/luma.led_matrix #125: setup.py dependency issue

thijstriemstra, 2017-10-10:
This is probably interpreted too late:
https://github.com/rm-hull/luma.led_matrix/blob/c93ee585b9131503cdd709ef83140e90882cd905/setup.py#L29

rm-hull, 2017-10-10:
Ah, yes - the same issue as before. Going to be harder to fix though

rm-hull, 2017-10-10:
Wonder if better to remove from setup altogether and rely on manual install, but add a try/catch around the import statement with a user-friendly message  saying the PKG should be manually installed?

thijstriemstra, 2017-10-22:
Only reference to ARM I can find on a Python is using `platform.machine()`:

```
$ python2.7
Python 2.7.13 (default, Jan 19 2017, 14:48:08) 
[GCC 6.3.0 20170124] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import platform
>>> platform.processor()
''
>>> platform.release()
'4.9.41-v7+'
>>> platform.uname()
('Linux', 'raspberrypi', '4.9.41-v7+', '#1023 SMP Tue Aug 8 16:00:15 BST 2017', 'armv7l', '')
>>> platform.machine()
'armv7l'
>>> 
```

rm-hull, 2017-10-22:
Awesome - that's probably the best overall solution :+1:

thijstriemstra, 2017-10-22:
So it appears it's a non-issue, closing.

rm-hull, 2017-10-22:
How are you installing though? it is the wheel version that is not working... I suspect we _do_ need to change the `setup.py` file to be:

```python

install_deps = [
    "luma.core>=1.0.3",
    'rpi_ws281x;platform_machine=="armv7l"',
    'ws2812;platform_machine=="armv7l"'
]
```

thijstriemstra, 2017-10-22:
I was using `pip install luma.led_matrix`. It should automatically have given me the wheel version (for luma.core and luma.led_matrix) no? Here's rest of log:

```
pip install luma.led_matrix(photobooth-router) photobooth@photoboothrouter1:~$ pip install luma.led_matrix
Collecting luma.led_matrix
  Downloading luma.led_matrix-1.0.3.tar.gz (10.3MB)
    100% |████████████████████████████████| 10.3MB 17kB/s 
Collecting luma.core>=1.0.3 (from luma.led_matrix)
  Downloading luma.core-1.0.3-py2.py3-none-any.whl (49kB)
    100% |████████████████████████████████| 51kB 309kB/s 
Collecting rpi_ws281x (from luma.led_matrix)
  Downloading rpi_ws281x-3.0.3.tar.gz (58kB)
    100% |████████████████████████████████| 61kB 463kB/s 
Collecting ws2812 (from luma.led_matrix)
  Downloading ws2812-1.0.0.tar.gz (58kB)
    100% |████████████████████████████████| 61kB 362kB/s 
Collecting smbus2 (from luma.core>=1.0.3->luma.led_matrix)
  Downloading smbus2-0.2.0.tar.gz
Collecting RPI.GPIO; platform_system == "Linux" (from luma.core>=1.0.3->luma.led_matrix)
  Downloading RPi.GPIO-0.6.3.tar.gz
Collecting pillow>=4.0.0 (from luma.core>=1.0.3->luma.led_matrix)
  Downloading Pillow-4.3.0.tar.gz (13.9MB)
    100% |████████████████████████████████| 13.9MB 12kB/s 
Collecting spidev; platform_system == "Linux" (from luma.core>=1.0.3->luma.led_matrix)
  Downloading spidev-3.2.tar.gz
Collecting olefile (from pillow>=4.0.0->luma.core>=1.0.3->luma.led_matrix)
  Downloading olefile-0.44.zip (74kB)
    100% |████████████████████████████████| 81kB 676kB/s 
Building wheels for collected packages: luma.led-matrix, rpi-ws281x, ws2812, smbus2, RPI.GPIO, pillow, spidev, olefile
  Running setup.py bdist_wheel for luma.led-matrix ... done
  Stored in directory: /home/foo/.cache/pip/wheels/ca/fa/c4/ed1bd2aef9965c8433669087e62a1479c56d42ed705c1ebcf2
  Running setup.py bdist_wheel for rpi-ws281x ... done
  Stored in directory: /home/foo/.cache/pip/wheels/9d/77/da/367c58885f83e0ee79278d944d5e184983f2b94a5901ca364c
  Running setup.py bdist_wheel for ws2812 ... done
  Stored in directory: /home/foo/.cache/pip/wheels/d0/11/09/7e3801ed8b638d1397caa9eb2d8da3ff02db3346a0d26bcf1d
  Running setup.py bdist_wheel for smbus2 ... done
  Stored in directory: /home/foo/.cache/pip/wheels/90/71/b4/9f90d8e2d0349ab55fef07169a81bd8f925965f16174e2f809
  Running setup.py bdist_wheel for RPI.GPIO ... done
  Stored in directory: /home/foo/.cache/pip/wheels/ae/4d/3b/e924997dbf06810adf3b2e37f1d9627b2327eb9cbb285949c9
  Running setup.py bdist_wheel for pillow ... done
  Stored in directory: /home/foo/.cache/pip/wheels/49/97/22/638a2ad41d85bc151da7c987ae644da429a23e95733fabfd76
  Running setup.py bdist_wheel for spidev ... done
  Stored in directory: /home/foo/.cache/pip/wheels/e4/9b/5f/cf0ec030fc958b72315a15412130e4e1dc6040cdb490aa21fb
  Running setup.py bdist_wheel for olefile ... done
  Stored in directory: /home/foo/.cache/pip/wheels/20/58/49/cc7bd00345397059149a10b0259ef38b867935ea2ecff99a9b
Successfully built luma.led-matrix rpi-ws281x ws2812 smbus2 RPI.GPIO pillow spidev olefile
```

thijstriemstra, 2017-10-22:
Ah the wheel isn't universal, e.g. Python 2.x only: https://pypi.python.org/packages/64/c2/6a2eb197de0e5bb6a57f1bdfbc63893d00e69ca5d3ad1526b1b5a26bda8f/luma.led_matrix-1.0.3-py2-none-any.whl#md5=18e0127810aac1ffeafeb951b67e8a52 compared to https://pypi.python.org/packages/05/c4/e31bf66de37be96912a597989788130bb84152f9a10e405c2fbf2fd62c7d/luma.core-1.0.3-py2.py3-none-any.whl#md5=1e7d5124de3f8ae2b427fd19a8f9a960

thijstriemstra, 2017-10-22:
So yea, I'll have to test again using Python 2.7 because I was using Python 3 (which was not installing the wheel file) once #127 is released.

rm-hull, 2017-10-22:
> Also, is armv7l fine on all ARM platforms? There's not something like armv6l or armv8l (yet)?

ummm, not sure ... possibly the original distros were armv6l images, but lets just support armv7l for now.

rm-hull, 2017-10-22:
Let me cut a new release now that #127 is merged

thijstriemstra, 2017-10-22:
To make triple sure, include the platform name as well, e.g.:

```
'rpi_ws281x;platform_machine=="armv7l" and platform_system=="Linux"',
```

rm-hull, 2017-10-22:
will release 1.0.4 with your two fixes, and then 1.0.5 with the platform_machine env marker

thijstriemstra, 2017-10-22:
Ok sounds good, if you can give me some time to test 1.0.4.

rm-hull, 2017-10-22:
Sure thing. 
1.0.4 is published

thijstriemstra, 2017-10-22:
[image]

Great to see whl size so small compared to the tarball, but 9MB for tarball is extreme, hmm

thijstriemstra, 2017-10-22:
Inspecting the tarball shows some unnecessary files:

```
luma.led_matrix-1.0.4/doc/.ropeproject/
luma.led_matrix-1.0.4/doc/.ropeproject/config.py
luma.led_matrix-1.0.4/doc/.ropeproject/objectdb
luma.led_matrix-1.0.4/doc/.ropeproject/globalnames
luma.led_matrix-1.0.4/doc/.ropeproject/history
```

(wtf is rope)

```
luma.led_matrix-1.0.4/doc/.DS_Store
```

thijstriemstra, 2017-10-22:
Tested again with 1.0.4 on python 3.6 on 64-bit and the packages are excluded:

```
$ pip install luma.led_matrix
Collecting luma.led_matrix
  Downloading luma.led_matrix-1.0.4-py2.py3-none-any.whl
Collecting luma.core>=1.0.3 (from luma.led_matrix)
  Using cached luma.core-1.0.3-py2.py3-none-any.whl
Collecting RPI.GPIO; platform_system == "Linux" (from luma.core>=1.0.3->luma.led_matrix)
Collecting smbus2 (from luma.core>=1.0.3->luma.led_matrix)
Collecting pillow>=4.0.0 (from luma.core>=1.0.3->luma.led_matrix)
  Using cached Pillow-4.3.0-cp36-cp36m-manylinux1_x86_64.whl
Collecting spidev; platform_system == "Linux" (from luma.core>=1.0.3->luma.led_matrix)
Collecting olefile (from pillow>=4.0.0->luma.core>=1.0.3->luma.led_matrix)
Installing collected packages: RPI.GPIO, smbus2, olefile, pillow, spidev, luma.core, luma.led-matrix
Successfully installed RPI.GPIO-0.6.3 luma.core-1.0.3 luma.led-matrix-1.0.4 olefile-0.44 pillow-4.3.0 smbus2-0.2.0 spidev-3.2
```

And yep, you're right, on a raspberrypi with python 3.5.3 it omits the extra packages:

```
$ pip install luma.led_matrix
Collecting luma.led_matrix
  Downloading luma.led_matrix-1.0.4-py2.py3-none-any.whl
Collecting luma.core>=1.0.3 (from luma.led_matrix)
  Using cached luma.core-1.0.3-py2.py3-none-any.whl
Collecting spidev; platform_system == "Linux" (from luma.core>=1.0.3->luma.led_matrix)
Collecting pillow>=4.0.0 (from luma.core>=1.0.3->luma.led_matrix)
Collecting smbus2 (from luma.core>=1.0.3->luma.led_matrix)
Collecting RPI.GPIO; platform_system == "Linux" (from luma.core>=1.0.3->luma.led_matrix)
Collecting olefile (from pillow>=4.0.0->luma.core>=1.0.3->luma.led_matrix)
Installing collected packages: spidev, olefile, pillow, smbus2, RPI.GPIO, luma.core, luma.led-matrix
Successfully installed RPI.GPIO-0.6.3 luma.core-1.0.3 luma.led-matrix-1.0.4 olefile-0.44 pillow-4.3.0 smbus2-0.2.0 spidev-3.2
```

closed 2017-10-22 by thijstriemstra

rm-hull, 2017-10-22:
1.0.5 published - good effort on all the PRs tonight !  :cake:

thijstriemstra, 2017-10-23:
1.0.5 on raspberrypi:

```
$ pip install luma.led_matrix
Collecting luma.led_matrix
  Downloading luma.led_matrix-1.0.5-py2.py3-none-any.whl
Collecting rpi-ws281x; platform_machine == "armv7l" and platform_system == "Linux" (from luma.led_matrix)
Collecting ws2812; platform_machine == "armv7l" and platform_system == "Linux" (from luma.led_matrix)
Collecting luma.core>=1.0.3 (from luma.led_matrix)
  Using cached luma.core-1.0.3-py2.py3-none-any.whl
Collecting pillow>=4.0.0 (from luma.core>=1.0.3->luma.led_matrix)
Collecting spidev; platform_system == "Linux" (from luma.core>=1.0.3->luma.led_matrix)
Collecting smbus2 (from luma.core>=1.0.3->luma.led_matrix)
Collecting RPI.GPIO; platform_system == "Linux" (from luma.core>=1.0.3->luma.led_matrix)
Collecting olefile (from pillow>=4.0.0->luma.core>=1.0.3->luma.led_matrix)
Installing collected packages: rpi-ws281x, ws2812, olefile, pillow, spidev, smbus2, RPI.GPIO, luma.core, luma.led-matrix
Successfully installed RPI.GPIO-0.6.3 luma.core-1.0.3 luma.led-matrix-1.0.5 olefile-0.44 pillow-4.3.0 rpi-ws281x-3.0.3 smbus2-0.2.0 spidev-3.2 ws2812-1.0.0
```

Sweet.

PR #129 (2017-10-22): only install these dependencies on ARM linux
fixes #125