rm-hull/luma.led_matrix #132: retrieve number of pixels of string to be displayed

3zra, 2017-10-28:
Hello,

To foresee whether a string to be displayed fits into a cascaded 8x8 matrix display of given length and height it would be very helpfull to have means of retrieving the number of pixels needed to display a given string. Is there a way to do this? 

Supplement:

RPi Model:
RPi2B

Kernel version:

```
osmc@osmc:~$ uname -a
Linux osmc 4.9.29-9-osmc #1 SMP PREEMPT Sat Jul 29 01:44:42 UTC 2017 armv7l GNU/Linux
```

rm-hull, 2017-10-28:
If you use the non-legacy text drawing methods then you can use the `textsize` method on pillow's [ImageDraw](https://pillow.readthedocs.io/en/4.3.x/reference/ImageDraw.html#PIL.ImageDraw.PIL.ImageDraw.ImageDraw.textsize) object. Here's an example of where we used it in the demos: https://github.com/rm-hull/luma.examples/blob/master/examples/game_of_life.py#L64

If you are using the legecy text drawing mechanisms, then there is a shadow function (also called `textsize`) - see http://luma-core.readthedocs.io/en/latest/api-documentation.html#luma.core.legacy.textsize

3zra, 2017-10-28:
@rm-hull Thanks a lot. Your suggestion worked out nicely.

closed 2017-10-28 by 3zra