rm-hull/luma.led_matrix #133: Human-readable list of supported devices

thijstriemstra, 2017-11-03:
device.py contains:

```python
__all__ = ["max7219", "ws2812", "neopixel", "neosegment", "apa102"]
```

but this cannot be used in the examples headline for example. Having a method that returns proper names of these devices instead would be useful.

rm-hull, 2018-01-02:
I seem to think we (mis)used Python's `__all__` variable so that the demos could introspect which devices were available. [https://docs.python.org/3/tutorial/modules.html#importing-from-a-package]

What do mean by proper names?

thijstriemstra, 2018-01-03:
@rm-hull the names in the project description is what I'm trying to get into the documentation

```
 (MAX7219) and RGB NeoPixels (WS2812 / APA102) 
```

thijstriemstra, 2019-03-18:
Specifically, I want to get rid of the hard-coded devices here: https://github.com/rm-hull/luma.led_matrix/blob/master/doc/conf.py#L61

thijstriemstra, 2019-03-18:
Similar to the approach in luma.oled: https://github.com/rm-hull/luma.oled/blob/3.1.0/doc/conf.py#L69

thijstriemstra, 2019-03-18:
I suggest something like `__devices__` or `__deviceNames`..

```python
__all__ = ["max7219", "ws2812", "neopixel", "neosegment", "apa102"]

__deviceNames__ = {
    "max7219": "MAX7219",
    "ws2812": "WS2812",
    "neopixel": "Neopixel",
    "neosegment": "Neosegment",
    "apa102": "APA102"
}
```

And then some method that retrieves the device name by classname etc.

Or we could add a `label` class var to the devices with the label name.

Note that whatever approach has to work for all other libraries as well.

Thoughts @rm-hull?

rm-hull, 2019-03-18:
How about 

    __deviceNames__ = { ... }

    __all__ = __deviceNames__.keys()

?

thijstriemstra, 2019-03-18:
That's fine with me as well. I'll make a PR.

thijstriemstra, 2019-03-18:
Actually, I'd rather not mess around with `__all__` and make it depend on other code. I've seen this result in strange bugs (GC) that we don't want to get into. I'd rather keep `__all__` hard-coded + new `__deviceNames` with duplication, or use the `device.label` thing. Thoughts @rm-hull ?

rm-hull, 2019-03-18:
Ok, no problem. Don’t mind which approach tbh ... maybe go with

> keep `__all__` hard-coded + new `__deviceNames__` with duplication

to keep them located together