rm-hull/luma.led_matrix #135: 7-seg - How to align number to the right?

Bartyx, 2017-11-21:
Hey.
Sorry for my bad english. 
I want to send 4-digits number to the 7-seg and i need to align it to the right side. 
Now i am using code like this: 
`seg.text = ("    ")+number`
for skipping first 4 digits.
Now my number raising to 5-digits. I need to erase one space manualy from the code. 
Is there any way to automatize align to right?
Thanks for help.

rm-hull, 2017-11-21:
Have a look at https://pyformat.info/#string_pad_align.

You should be able to do something like:

```python
seg.text = '{:8d}'.format(number)
```

or if you want zero's:

```python
seg.text = '{:08d}'.format(number)
```

Bartyx, 2017-11-21:
Thank you very much. That works perfect!

closed 2017-11-21 by rm-hull