rm-hull/luma.led_matrix #139: Depreciated version no installing correctly via 'sudo pip install max7219'

DavidMS51, 2018-01-02:
Hi,
I am still using the depreciated version of the s/w installing via 'sudo pip install max7219'. I have just done a clean install for the first time in a while and it did not work - that is it reported it had installed correctly but nothing was displayed when I ran my existing python app or one of your test examples.

When I tried re-installing via the longer route from your github site it now seems to work ? I guess my question is has anything changed that you know of that could stop the pip install method working ?

I am going to try it with an older version of Rasbian to check that it not the issue

Thanks

Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

thijstriemstra, 2018-01-02:
Use pip install luma.led_matrix instead.

DavidMS51, 2018-01-02:
Thanks but I cannot see that working - unfortunately my code is dependent on features only available in the old max7219 variant of the code

thijstriemstra, 2018-01-02:
Porting it over should be trivial, can you show some of your code?

DavidMS51, 2018-01-02:
The problem is that I use a function to control individual LEDs in the matrix - for instance self.device.pixel(col, row, id, redraw)
Which are not available in updated versions

thijstriemstra, 2018-01-02:
@DavidMS51 not sure if it's not available in the newer version, maybe @rm-hull has an idea.

DavidMS51, 2018-01-02:
It's not unfortunately - I checked in with him when he made the initial change

thijstriemstra, 2018-01-02:
@DavidMS51 can you paste the console output of what you did or trying to do? errors and log etc

rm-hull, 2018-01-03:
@DavidMS51 - the old `pixel` method could be simulated by creating (and holding onto) a blank 8x8 Pillow image, and then calling any number of `image.putpixel((x, y), 1)` to set pixels followed by `device.display(image)`. 

To blank out a pixel: `image.putpixel((x, y), 0)` followed by `device.display(image)`.

See http://pillow.readthedocs.io/en/4.3.x/reference/Image.html#PIL.Image.Image.putpixel

DavidMS51, 2018-01-03:
Thanks I guess this is the push I need to update the code !

I have got it installed and managed to get an individual pixel to flash on an off. I'm not quite sure what you mean by 'holding onto' the Pillow image but I guess this will be come clear at some stage !

DavidMS51, 2018-01-03:
One Thought you might want to update your instructions in terms of accessing the legacy version to suggest users will have problems with recent versions of Raspbian

thijstriemstra, 2018-01-04:
Please create a pull request with these changes so we can incorporate them @davidms51

thijstriemstra, 2018-01-22:
Only note I can find describing this is here: https://github.com/rm-hull/luma.led_matrix/blob/master/README.rst#breaking-changes

DavidMS51, 2018-01-22:
>     This email has been scanned by Netintelligence

Yes that this the correct bit, need some words along the lines of ;

Existing text

"There is no direct migration path, but the olddocs https://max7219.readthedocs.io/ andPyPi packages https://pypi.python.org/pypi/max7219 will remain available indefinitely, but that deprecated codebase will no longer recieve updates or fixes"

Proposed text

'There is no direct migration path, but the olddocs https://max7219.readthedocs.io/ andPyPi packages https://pypi.python.org/pypi/max7219 will remain available indefinitely, but that deprecated codebase will no longer receive updates or fixes. It is also likely that future updates to the Raspbian will create incompatibilities which stop or cause the deprecated max7219 library be unstable.'

Sorry but my knowledge of GitHub is limited, so not sure of process of suggest this change as a pull request.

David

thijstriemstra, 2018-01-22:
@DavidMS51 to make a pull request: use github to find the file you want to edit and click the edit button to make changes in the browser and press save/create pull request, and et voila.

DavidMS51, 2018-01-22:
ok will have a go thanks

thonwontwork, 2018-03-29:
Any chance someone can give me a quick explanation on how this works:

"The old pixel method could be simulated by creating (and holding onto) a blank 8x8 Pillow image, and then calling any number of image.putpixel((x, y), 1) to set pixels followed by device.display(image).

To blank out a pixel: image.putpixel((x, y), 0) followed by device.display(image)."

New to programming and trying to make a the leds light up one by one in quick succession.

DavidMS51, 2018-04-02:
Try having a look at my updated TFwordclock code - it will give you a working example to start from
[https://github.com/DavidMS51/TFWordclock/blob/master/timewrd5ca_eng.py](url)