rm-hull/luma.led_matrix #140: show_message, text draw not show text properly

macpit, 2018-01-02:
Show_message and text draw show text not properly:

[image]

#### Type of Raspberry Pi
Pi 3B

#### Linux Kernel version / Python
`Linux raspberrypi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux.`

`root@raspberrypi:~/# pip3 list|grep luma
luma.core (1.2.1)
luma.led-matrix (1.0.7)
root@raspberrypi:~/# python3 -V
Python 3.5.3`

#### Expected behaviour
Test Code:

`import time`

`from luma.led_matrix.device import neopixel`
`from luma.core.render import canvas`
`from luma.core.legacy import text, show_message`
`from luma.core.legacy.font import TINY_FONT, CP437_FONT`

`device = neopixel(width=60, height=10)`
`device.contrast(50)`

`def main():`
 `   with canvas(device, dither=True) as draw:`
  `      text(draw, (0, -1), txt="TEST", fill="red", font=CP437_FONT)`

   ` time.sleep(10)`

`if __name__ == "__main__":`
   ` main()`

#### Actual behaviour
ws281x Library installed from https://github.com/richardghirst/rpi_ws281x,
The provided Demo works:

[image]

rm-hull, 2018-01-02:
Can I ask how the neopixels are arranged? They are serially addressed, so there needs to be a mapping of some sort to turn them into an array of 60x10 pixels.

The unicorn HAT has a snake layout but you can invent your own, see the relevant docs -
 https://luma-led-matrix.readthedocs.io/en/latest/python-usage.html#pimoroni-unicorn-hat and https://github.com/rm-hull/luma.led_matrix/blob/master/luma/led_matrix/device.py#L335-L346

In order to work out the layout, alter the value for `cascaded` here: https://github.com/rm-hull/luma.led_matrix/blob/master/examples/neopixel_crawl.py#L11 and then run that program. Observe how the green dot moves over the array, and this should give you enough of an idea how to create your own mapping. If you can't make a video of it, upload to YouTube and post a link here and we'll help.

rm-hull, 2018-01-02:
Also, your code:

```python
    with canvas(device, dither=True) as draw:
        text(draw, (0, -1), txt="TEST", fill="red", font=CP437_FONT)
```

Neopixels are full-color RGB, so no need for the `dither=True`, ... and try drawing from (0, 0) as well.

What would be useful to debug this is to run the following from [luma.examples](https://github.com/rm-hull/luma.examples):

```
$ python examples/demo.py --display neopixel --width=60 --height=10
```

And if that works, post a pic of the neopixels with that running.

macpit, 2018-01-03:
I have create my own mapping, now all works fine.
[image]

[image]

Thank you!

closed 2018-01-03 by rm-hull

thijstriemstra, 2018-01-04:
Can you share this mapping here so others can use it?

thijstriemstra, 2018-01-04:
P.s. that thing looks huge 😁

macpit, 2018-01-04:
The screen is a Dell 43 Zoll 4K Screen :-) The mapping you can find bellow:

```python
a = [540, 541, 542, 543, 544, 545, 546, 547, 548, 549, 550, 551, 552, 553, 554, 555, 556, 557, 558, 559, 560, 561, 562, 563, 564, 565, 566, 567, 568, 569, 570, 571, 572, 573, 574, 575, 576, 577, 578, 579, 580, 581, 582, 583, 584, 585, 586, 587, 588, 589, 590, 591, 592, 593, 594, 595, 596, 597, 598, 599]
b = [539, 538, 537, 536, 535, 534, 533, 532, 531, 530, 529, 528, 527, 526, 525, 524, 523, 522, 521, 520, 519, 518, 517, 516, 515, 514, 513, 512, 511, 510, 509, 508, 507, 506, 505, 504, 503, 502, 501, 500, 499, 498, 497, 496, 495, 494, 493, 492, 491, 490, 489, 488, 487, 486, 485, 484, 483, 482, 481, 480]
c = [420, 421, 422, 423, 424, 425, 426, 427, 428, 429, 430, 431, 432, 433, 434, 435, 436, 437, 438, 439, 440, 441, 442, 443, 444, 445, 446, 447, 448, 449, 450, 451, 452, 453, 454, 455, 456, 457, 458, 459, 460, 461, 462, 463, 464, 465, 466, 467, 468, 469, 470, 471, 472, 473, 474, 475, 476, 477, 478, 479]
d = [419, 418, 417, 416, 415, 414, 413, 412, 411, 410, 409, 408, 407, 406, 405, 404, 403, 402, 401, 400, 399, 398, 397, 396, 395, 394, 393, 392, 391, 390, 389, 388, 387, 386, 385, 384, 383, 382, 381, 380, 379, 378, 377, 376, 375, 374, 373, 372, 371, 370, 369, 368, 367, 366, 365, 364, 363, 362, 361, 360]
e = [300, 301, 302, 303, 304, 305, 306, 307, 308, 309, 310, 311, 312, 313, 314, 315, 316, 317, 318, 319, 320, 321, 322, 323, 324, 325, 326, 327, 328, 329, 330, 331, 332, 333, 334, 335, 336, 337, 338, 339, 340, 341, 342, 343, 344, 345, 346, 347, 348, 349, 350, 351, 352, 353, 354, 355, 356, 357, 358, 359]
f = [299, 298, 297, 296, 295, 294, 293, 292, 291, 290, 289, 288, 287, 286, 285, 284, 283, 282, 281, 280, 279, 278, 277, 276, 275, 274, 273, 272, 271, 270, 269, 268, 267, 266, 265, 264, 263, 262, 261, 260, 259, 258, 257, 256, 255, 254, 253, 252, 251, 250, 249, 248, 247, 246, 245, 244, 243, 242, 241, 240]
g = [180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, 200, 201, 202, 203, 204, 205, 206, 207, 208, 209, 210, 211, 212, 213, 214, 215, 216, 217, 218, 219, 220, 221, 222, 223, 224, 225, 226, 227, 228, 229, 230, 231, 232, 233, 234, 235, 236, 237, 238, 239]
h = [179, 178, 177, 176, 175, 174, 173, 172, 171, 170, 169, 168, 167, 166, 165, 164, 163, 162, 161, 160, 159, 158, 157, 156, 155, 154, 153, 152, 151, 150, 149, 148, 147, 146, 145, 144, 143, 142, 141, 140, 139, 138, 137, 136, 135, 134, 133, 132, 131, 130, 129, 128, 127, 126, 125, 124, 123, 122, 121, 120]
i = [60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76, 77, 78, 79, 80, 81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98, 99, 100, 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119]
j = [59, 58, 57, 56, 55, 54, 53, 52, 51, 50, 49, 48, 47, 46, 45, 44, 43, 42, 41, 40, 39, 38, 37, 36, 35, 34, 33, 32, 31, 30, 29, 28, 27, 26, 25, 24, 23, 22, 21, 20, 19, 18, 17, 16, 15, 14, 13, 12, 11, 10, 9, 8, 7, 6, 5, 4, 3, 2, 1, 0]

p600 = a + b + c + d + e + f + g + h + i + j
````