rm-hull/luma.led_matrix #144: Russian text

viktor6, 2018-01-28:
Hello. Help with the Russian coding in the running line
Raspberry PI3
Max7219 8x8x4
`pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.9.78-v7+ #1084 SMP Thu Jan 25 18:05:49 GMT 2018 armv7l GNU/Linux
pi@raspberrypi:~ $`

`#!/usr/bin/env python 
# -*- coding: utf-8 -*- 
# Copyright (c) 2017-18 Richard Hull and contributors 
# See LICENSE.rst for details. 

import re
import sys
import datetime
import time
import argparse

from luma.led_matrix.device import max7219
from luma.core.interface.serial import spi, noop
from luma.core.render import canvas
from luma.core.virtual import viewport
from luma.core.legacy import text, show_message
from luma.core.legacy.font import proportional, CP437_FONT, TINY_FONT, SINCLAIR_FONT, LCD_FONT

serial = spi(port=0, device=0, gpio=noop())
device = max7219(serial, cascaded=4, block_orientation=-90)

while(True):
	msg = "Всем привет!!!"
        show_message(device, msg, fill="white", font=proportional(SINCLAIR_FONT))
        time.sleep(1)

for offset in range(8):
    virtual.set_position((offset, offset))`

rm-hull, 2018-01-28:
Hello @viktor6, I am not sure what your question is, or more specifically, what problem you are having. If it is how to display Cyrillic characters, possibly we can call upon @tarasius and @vortigont to help (both contributed to the Russian-language fixed-width fonts).

viktor6, 2018-01-29:
Yes display Cyrillic characters

tarasius, 2018-01-29:
@viktor6 This is my solution. To avoid concurrent messages, I made a daemon, which receives commands what to show with a pipe message. I use it with online radio, showing the station, artist and song.
Here is daemon code:

```
#!/usr/bin/env python
# -*- coding: utf-8 -*-
import logging
logging.basicConfig(format='%(asctime)s %(message)s') #,filename='LedSay')
log = logging.getLogger()
log.setLevel(logging.DEBUG)

import os, time, sys
import multiprocessing
from tendo import singleton
from luma.led_matrix.device import max7219
from luma.core.interface.serial import spi, noop
from luma.core.render import canvas
from luma.core.legacy import text, show_message
from luma.core.legacy.font import LCD_FONT, proportional
#CP437_FONT, TINY_FONT, SINCLAIR_FONT,

def say(msg, delay, intensity, orientation):
    # create matrix device
    serial = spi(port=0, device=0, gpio=noop())
    device = max7219(serial, cascaded=1, block_orientation=0, rotate=orientation)    
    device.contrast(intensity * 16)

    # font=proportional(SINCLAIR_FONT)
    show_message(device, msg, fill="white", font=proportional(LCD_FONT), scroll_delay=delay)

if __name__ == "__main__":
	me = singleton.SingleInstance() # will sys.exit(-1) if other instance is running
	
	PipeName = '/LedsButtons/LedSayPipe'
	if not os.path.exists(PipeName):
		log.debug('Pipe not present. Creating.')
		os.mkfifo(PipeName, 0777)
		log.debug('Pipe is made')
	else:
		log.debug('Pipe already present')
		
	p = multiprocessing.Process(target=say, args=('Hello',0.1,1,3))
	p.start()
	
	log.debug('Open pipe for reading')
	PipeIn = open(PipeName, 'r')
	log.debug('Pipe is open for reading')

	while True:
		try:
			PipeString = PipeIn.readline() [:-1]
			if (len(PipeString)>0):
				log.info('LedSay '+PipeString)				
				if (p.is_alive()):
					log.debug('Terminating')
					p.terminate()
					p.join()
					del p
				d,i,o,m = PipeString.split('#')	
				log.debug('d='+d+' i='+i+' o='+o+' m='+m)				
				p = multiprocessing.Process(target=say, args=(m,float(d),int(i),int(o)))
				p.start()
				log.debug('Started')
		except:
			log.debug('Cant read Pipe')
		
```

And here is the script which communicates with daemon:

```
#!/usr/bin/env python
import logging
logging.basicConfig(format='%(asctime)s %(message)s') 
log = logging.getLogger()
log.setLevel(logging.DEBUG)

import os, time, sys
import argparse
import threading

def run(message, orientation, delay, intensity):
	PipeName = '/LedsButtons/LedSayPipe'
	if not os.path.exists(PipeName):
		log.debug('Pipe not present. Creating.')
		os.mkfifo(PipeName, 0777)
	
	PipeOut = open(PipeName, 'w')	
	DataPipeString = str(delay)+'#'+str(intensity)+'#'+str(orientation)+'#'+message+'\n'
	log.info(DataPipeString)
	PipeOut.write(DataPipeString)
	PipeOut.close	
	
def background(message, orientation, delay, intensity):
  led_thread = threading.Thread(target=run, args=[message, orientation, delay, intensity])
  led_thread.start()

	
if __name__ == "__main__":
	parser = argparse.ArgumentParser(description='ledsay arguments',
		formatter_class=argparse.ArgumentDefaultsHelpFormatter)

	parser.add_argument('message',metavar='m', help='Phrase to scroll on MAX7219 LED matrix')
	parser.add_argument('orientation', metavar='o',type=int, default=1, help='Block orientation')
	parser.add_argument('delay', metavar='d',type=float, default=1.0, help='Scroll delay')
	parser.add_argument('intensity', metavar='i', type=int, default=1, choices=range(1,16), help='Screen intensity')

	args = parser.parse_args()

	run(args.message, args.orientation, args.delay, args.intensity)

	
```

vortigont, 2018-01-29:
@viktor6  you should use LCD_FONT if you need to display Cyrillic text. Only this bitmap font contains sprites with cyr letters.

thijstriemstra, 2018-01-29:
@tarasius thanks for the example. I noticed there's an import that you didn't define, is it an external package?

```python
from tendo import singleton
```

viktor6, 2018-01-29:
Thank you.
Guys very very much you I ask help or assist to en sign a script hours. Already how much I try is not what does not work

```
#!/usr/bin/env python

import os, time, sys
import multiprocessing
from luma.led_matrix.device import max7219
from luma.core.interface.serial import spi, noop
from luma.core.render import canvas
from luma.core.legacy import text, show_message
from luma.core.legacy.font import LCD_FONT, proportional

serial = spi(port=0, device=0, gpio=noop())
device = max7219(serial, cascaded=4, block_orientation=-90)

while(True):
	clock = time.strftime("%H%M")
	seconds = int(time.strftime("%S"))
	month = time.strftime(" %m")
	day = time.strftime(" %d")
	year = time.strftime("%Y")
	for n, c in enumerate(clock): 
		device.letter(n, ord(c))
	if seconds == 59 :
		for row in range(8):
			device.scroll_up()
			time.sleep(0.1)
		for n, c in enumerate(month): 
			device.letter(n, ord(c))
		time.sleep(1)
		for row in range(8):
			device.scroll_down()
			time.sleep(0.1)
		for n, c in enumerate(day): 
			device.letter(n, ord(c))
		time.sleep(1)
		for row in range(8):
			device.scroll_down()
			time.sleep(0.1)
		for n, c in enumerate(year): 
			device.letter(n, ord(c))
		time.sleep(1)
		for row in range(8):
			device.scroll_down()
			time.sleep(0.1)
		

if __name__ == "__main__":
    try:
        main()
    except KeyboardInterrupt:
        pass		
```

tarasius, 2018-01-30:
@thijstriemstra This is a way to protect the script from running more than once in the same time.
https://pythonhosted.org/tendo/ 
I had some problem when daemon script was started more than once. Or for the case daemon started the script, and user starts it once again manually.

closed 2018-03-28 by thijstriemstra