rm-hull/luma.led_matrix #145: Display Number with no Scrolling - Cascaded LED Matrix MAX7219

jonathanrjpereira, 2018-02-01:
Raspbian Jessie on Raspberry Pi 3. Python 2.7

I have a cascade of 4 max7219 matrix (each is 8x8 LEDs).
I simply want to display a number without any scrolling. Is this possible?

rm-hull, 2018-02-01:
Yes, have a look at http://luma-led-matrix.readthedocs.io/en/latest/python-usage.html#x8-led-matrices and https://luma-core.readthedocs.io/en/latest/api-documentation.html#luma.core.legacy.text

jonathanrjpereira, 2018-02-01:
I tried it matrix example out, but I am still facing some issues.
The text is still scrolling individually across each matrix instead of scrolling horizontally across the entire cascaded matrix.
I can't find an example to make the text static

thijstriemstra, 2018-02-01:
@jonathanrjpereira can you show some of the code you're using? a youtube video?

jonathanrjpereira, 2018-02-02:
I solved the horizontal display across the entire cascaded matrix by simply changing the `block_orientation` to `-90`.
and for static text, I used the `text()` to draw out the string.

closed 2018-02-02 by jonathanrjpereira

sudipshil, 2019-07-02:
HELP ...anyone please

i tired a matrix example :--google assistance controlling display in led matrix MAX7219

The text is still scrolling individually across each matrix instead of scrolling horizontally across the entire cascaded matrix.
i have attached the code below in notepad :-- plz help 
please reply as soon as possible 

[google assistance controlling display.txt](https://github.com/rm-hull/luma.led_matrix/files/3351613/google.assistance.controlling.display.txt)