rm-hull/luma.led_matrix #146: How to Write Custom Fonts - 7 Segment Font on Dot Matrix

jonathanrjpereira, 2018-02-02:
I am currently running static text on a matrix consisting of 4 cascaded 8x8 dot matrix using a Raspberry Pi 3 (Raspbian Jessie) & Python 2,7

In my project, I want to display some scrolling text followed by some 8-digit numbers. Although I want these numbers to be static, 

I want a single digit to have a width = 3 dots & height = 7 dots. This will make it possible to display two numbers on a single 8x8 matrix with adequate space between the two digits & also maximizing the pixel usage. The font will be very similar to the 7 Segment Display font.

Hence on a 4 cascaded 8x8 matrix, one will be able to statically display an 8 digit number

Obviously, the `7SEG_FONT` will not be able to display all characters but it will allow all numbers & a few signs to be displayed.

How do I go about creating a custom font such as a 7 Segment Font?

rm-hull, 2018-02-02:
Have a look at TINY_FONT (https://github.com/rm-hull/luma.core/blob/master/luma/core/legacy/font.py#L1116) - this may already be fit for your purpose. If not, and you still want to design your own font, then each number in the individual letter array represents the bit pattern for a column in the character.

jonathanrjpereira, 2018-02-02:
Unfortunately `TINY_FONT` does not optimize the pixel area being used as compared to a `7SEG_FONT `for digits only.

What format is the number in the individual letter array representing the bit pattern for a column in the character? It's a Hexadecimal number, but how does it represent the bit pattern? How do I convert the bit pattern into its corresponding Hex number?

rm-hull, 2018-02-02:
Convert each hex number into binary, and then in a column, draw a pixel for where there is a 1 and leave a blank for where there is a zero. Take an example, draw it out on paper and you will see how the characters are formed.

closed 2018-02-22 by jonathanrjpereira