rm-hull/luma.led_matrix #149: text movement

viktor6, 2018-02-13:
Hello! how to make the text move up and down, as well as left and right and stop.
Check the correct I wrote the script? I just started learning python
#### Type of Raspberry Pi
RPi3.
#### Linux Kernel version
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.9.78-v7+ #1084 SMP Thu Jan 25 18:05:49 GMT 2018 armv7l GNU/Linux
pi@raspberrypi:~ $

```
#!/usr/bin/env python

from luma.led_matrix.device import max7219
from luma.core.interface.serial import spi, noop
import time
from luma.core.legacy import text, show_message
from luma.core.legacy.font import proportional, CP437_FONT, TINY_FONT, SINCLAIR_FONT, LCD_FONT, UKR_FONT

serial = spi(port=0, device=0, gpio=noop())
device = max7219(serial, cascaded=4, block_orientation=-90)

while(True):
	t=time.strftime("%H:%M")

	msg = (t)
        show_message(device, msg, fill="white", font=proportional(LCD_FONT), scroll_delay=0.04)
        time.sleep(0)		
		
for offset in range(8):
    virtual.set_position((offset, offset))
```

rm-hull, 2018-02-13:
I would suggest you look at [crawl.py](https://github.com/rm-hull/luma.examples/blob/master/examples/crawl.py) or [scrolling_pixelart.py](https://github.com/rm-hull/luma.examples/blob/master/examples/scrolling_pixelart.py) demos in the **luma.examples** repo for examples of how to scroll up or down

viktor6, 2018-02-13:
For me, these are still complex examples, can something else be simpler?

thijstriemstra, 2018-05-30:
@viktor6 please reopen this ticket if you see any issues.

closed 2018-05-30 by thijstriemstra