rm-hull/luma.led_matrix #150: Recommended approach to reset MAX7219 devices?

drspangle, 2018-02-23:
I'm in the process of trying to figure out why my large (>100 matrices) MAX7219 display board is encountering glitches. I've spent a lot of time experimenting with clock speeds, bus transfer rates, and I've got something very close to a stable state. Despite this, the most common glitch is that single matrices seem to shut off and on, or change contrast for some reason, and it's not clear why. 

I want to do some debugging but aside from rerunning `get_device()`, which is the only high level abstraction that I can find which runs the `device`'s `show()` method (it's a `max7219` class instance) I don't have any means of resetting the devices. There's also the `clear()` method... I'm really not sure which one of the two, or both, are responsible for resetting the device to a stable state again.

My hypothesis is that if I'm rendering frames on the display at a constant rate, I could inject the command that successfully cleans up the glitch every second (framerate+1 frames) and it would prevent the glitched matrices from appearing glitched to the human eye, with minimal impact on performance.

I'm also analyzing the signals using a protocol analyzer and oscilloscope to see if there's some other reason why the MAX7219 modules are doing this strange behavior. From this perspective, the hypothesis is that the devices being affected by the glitch are missing a clock cycle or something similar. Not sure about this yet. Power reads clean on the oscilloscope, signals are 5V0 using a boost converter from the pi's 3V3 outputs, everything is wired in parallel except the DOUT to DIN pins on each module. I'm using PCBs with 4x MAX7219 chips each, the typical ones found on Amazon and Ebay. There's no significant ringing on any of the signal pulses as far as I can see. At this point there's nothing I can really change with respect to wiring, the clock speed, or the SPI MTU chunk size that makes it glitch any less, so I'm hoping I might be able to get closer to a solution by resetting the devices as quickly as possible on a regular interval.

drspangle, 2018-02-23:
So, after a bunch of experimentation, it seems that the `show()`, `hide()` and `clear()` methods don't solve my problem. Something else getting called (perhaps during the destruction of the previous class instantiated by `get_device()`?) is resetting the devices and cleaning up the glitch. I'm hoping it's something that can be done rapidly because it seems that the `show()`, `hide()` and `clear()` methods introduce a non-trivial delay in rendering, but if they're not the necessary part of the `get_device()` call then perhaps my hypothesis that the display could be reset on a regular interval to fix the glitches is still correct.

thijstriemstra, 2018-03-28:
ping @rm-hull

drspangle, 2018-03-28:
@thijstriemstra @rm-hull  This is still ongoing. I don't think this is a software issue, I think there's actually a signal degradation problem with the MAX7219 devices arranged in cascades. 

I'm still in the process of developing a PCB that can be used to strap together 4x4 modules so that they'll work reliably. I'm learning that PCB design takes a lot of time and effort, though, so it might not be reasonable to expect a resolution in the very short term.

drspangle, 2018-03-28:
Oh, also, my hypothesis about resetting the displays rapidly turned out to be a bust. It takes too long to maintain a steady framerate, and the glitches I'm encountering can happen often enough that the display will go noticeably wonky before the refresh can fix the issue (even if you're doing it as often as every second, or even every few frames).

drspangle, 2018-05-30:
I'm still waiting for the PCBs I developed to fix this problem to come back from fabrication. I'll update at that time - it'll be a couple weeks probably.

Overall, there's no software bug that I can see. If you're less concerned about the hardware solution to this signal degradation problem (as previously discussed in #108), you can close this issue now.

thijstriemstra, 2018-05-31:
>  you can close this issue now.

no rush, it's your ticket now ;)

Jerry981028, 2018-06-02:
I also have such problem with my 32x24 max7219 device. I managed to solve the issue by using a AMS1117 to provide 3.3v vcc.

drspangle, 2018-06-02:
Surely you must have meant 5v?

Jerry981028, 2018-06-02:
@drspangle 3.3v same voltage as gpio

closed 2018-12-08 by rm-hull

drspangle, 2018-12-08:
I had forgotten about this but the closed notification reminded me about it.

Turns out that the reason why the cascades were screwing up is parasitic capacitance from using dupont connectors. Once I used a PCB (with no special hardware on it) to connect all the cascaded modules, I was able to build a 32x160 matrix which works a treat. I'll be open sourcing the design for the display pretty soon.

rm-hull, 2018-12-08:
32x160 matrix ...  so that 4x20 units - nice!  
Would love to see a video of that in action!

Yeah, i closed based on what you said in https://github.com/rm-hull/luma.led_matrix/issues/150#issuecomment-393332793.

Good to know what the cause was. I'm not an electrical engineer, but of course it had to be something like that.

drspangle, 2018-12-09:
Check it out...
[image]

I'm pretty confident that you could daisy chain these pretty much ad infinitum. This huge 5x 4x4 matrix array was exactly what I was trying to do.