rm-hull/luma.led_matrix #151: No module named luma.led.matrix.device

boringtrousers, 2018-04-07:
I'll start by apologising, I'm new to python and raspberry pi. I have very limited programming experience from when I used to mess around with some things a few years ago but mostly forgotten.

I'm trying to program an array of 4 8x8 LED matrices to display crypto currency prices. Right now Im stuck on the basics

I've cloned the repository, and when I try to run 
`$ python examples/matrix_demo.py`
i get hit with
`Traceback (most recent call last):
  File "examples/matrix_demo.py", line 10, in <module>
    from luma.led_matrix.device import max7219
ImportError: No module named luma.led_matrix.device`

I've been trying to figure out why its happening but I really dont know
Please help :(

rm-hull, 2018-04-07:
Did you follow https://luma-led-matrix.readthedocs.io/en/latest/install.html ?

boringtrousers, 2018-04-08:
I did, and I just read through it again to double check but everything seems fine up until I try to test the example

thijstriemstra, 2018-04-08:
@boringtrousers if you used `sudo` to install the packages you also need to sudo to run the demo, e.g. `sudo python examples/matrix_demo.py`.

boringtrousers, 2018-04-08:
@thijstriemstra I've tried that too but it still doesnt work :(

boringtrousers, 2018-04-08:
@rm-hull 
I seem to have both those entries showing up, here
```
$ pip list | grep luma
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
luma.core (1.7.2)
luma.led-matrix (1.0.8)

```

Thanks for the help, by the way

rm-hull, 2018-04-08:
* What version of python are you using (use `python -V`) ?

* What is the output of `pip show luma.led_matrix`? 

I think pip will treat hyphen and underscore interchangeably, BTW.

boringtrousers, 2018-04-08:
Python 2.7.13

```
$ pip show luma.led_matrix
Name: luma.led-matrix
Version: 1.0.8
Summary: A library to drive a MAX7219 LED serializer (using SPI) and WS2812 NeoPixels (using DMA)
Home-page: https://github.com/rm-hull/luma.led_matrix
Author: Richard Hull
Author-email: richard.hull@destructuring-bind.org
License: MIT
Location: /usr/local/lib/python2.7/dist-packages/luma.led_matrix-1.0.8-py2.7.egg
Requires: luma.core, rpi-ws281x, ws2812

```

rm-hull, 2018-04-09:
@boringtrousers - see comment https://github.com/rm-hull/luma.examples/issues/24#issuecomment-379677387 - maybe a force reinstall will work for you?

boringtrousers, 2018-04-19:
Thank you, I saw your comment but I haven't been able to try this for the past couple of weeks, as I have been away.
Just tried force reinstalling and it fixed the problem.
Thanks a lot

closed 2018-04-19 by boringtrousers