rm-hull/luma.led_matrix #153: Display Text Without scrolling Max7219 Casc=4 (Rpi3)

bima-joe, 2018-05-10:
Hi, is it possible to display text without scrolling? (eg:A100) 
I'm using Rpi3
I have followed this doc http://luma-led-matrix.readthedocs.io/en/latest/python-usage.html#daisy-chaining & https://luma-core.readthedocs.io/en/latest/api-documentation.html#luma.core.legacy.text
 but it just blinks once and doesn't show the text

thanks

rm-hull, 2018-05-10:
Yes it is possible.
Please post the program you have tried, and/or look at the Luma.examples repo

bima-joe, 2018-05-10:
from luma.led_matrix.device import max7219
from luma.core.interface.serial import spi, noop
from luma.core.render import canvas
from luma.core import legacy
from luma.core.legacy import text, show_message
from luma.core.legacy.font import proportional, CP437_FONT, TINY_FONT, SINCLAIR_FONT, LCD_FONT

serial = spi(port=0, device=1, gpio=noop())
device = max7219(serial, cascaded=4, block_orientation=90, rotate=1)
    
with canvas(device) as draw:
	text(draw, (0, 0), "A100", fill="white", font=proportional(CP437_FONT))

i'm just modify the code from matrix_demo.py

gitbs, 2018-05-17:
actually I hit the same problem when tried to display the time for a few seconds - I have an RPI 1 B with an 8x64 matrix
I have a time.sleep(3) at the end but I see a blink and that's it, the next sequence is executed. 

I have tried `"show_message(blah...blah, scroll_delay=0.0"` but it seems to check for zero value and is set back to default; 
Maybe if you would remove the fallback option we could use (well, exploit) that method.

rm-hull, 2018-05-17:
@bima-joe - once the program terminates, it cleans up the display driver and turns it off. This probably goes someway to explain your "blinks once" description.

Try doing a `time.sleep(...)` or have a look at the code snippet in https://github.com/rm-hull/luma.led_matrix/issues/76#issuecomment-287530682 where it overrides the cleanup method to do nothing.

gitbs, 2018-05-20:
@rm-hull - now works, cannot explain what was the glitch but now it works. I still get the "blink" effect but after that it text(draw, (0,0), "message here", ...) works as expected

@bima-joe - if you get matrix_demo.py working then just steal the example code - the brightness test shows one letter at the first position so it must work. 

Ticket can be closed as far as I am concerned.

closed 2018-05-21 by thijstriemstra