rm-hull/luma.led_matrix #154: Python3 ImportError (libtiff.so.5)

Blanchot, 2018-06-17:
First off let me thank you for the fine luma.core libraries!

I'm getting the following error when I try to import the Luma libraries under Python 3.5.3:

Traceback (most recent call last):
  File "plutus02.py", line 5, in <module>
    from luma.core.render import canvas
  File "/usr/local/lib/python3.5/dist-packages/luma/core/render.py", line 5, in <module>
    from PIL import Image, ImageDraw
  File "/usr/local/lib/python3.5/dist-packages/PIL/Image.py", line 60, in <module>
    from . import _imaging as core
ImportError: libtiff.so.5: cannot open shared object file: No such file or directory

My code runs fine under Python 2.7.13.
The odd thing is last week I installed luma.led_matrix for both Python 2 and 3 on another pi and got the same code working... albeit BOTH on that machine and now this one, I  ran into another ImportError:

File "/usr/local/lib/python3.5/dist-packages/PIL/Image.py", line 60, in <module>
    from . import _imaging as core
ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory

Searching a bit online I found I could solve that problem and get my code to work by installing libopenjp2-7 :
sudo apt-get install libopenjp2-7

But I can't seem to find a fix for the libtiff.so.5 error.

BTW: I have tried everything I could think of including:
sudo -H pip3 install --upgrade --force-reinstall --ignore-installed luma.led_matrix
to no avail.

I would appreciate any and all help here!

-- B.

Blanchot, 2018-06-17:
Ah... a bit more poking around online and I found a solution:

sudo apt install libtiff5

seems to have fixed the problem.

Hopefully this might help someone else. But why aren't libopenjp2-7 and libtiff5 installed when one follows the normal installation instructions?

FWIW: I'm running the latest Raspbian Stretch Lite (Kernel v. 4.14) --updated and upgraded on Raspberry Pi Zero W's.
 
Thanks again for a great a library!
--B.

mb720, 2019-07-20:
Somehow related to this, I ran into

      File "/usr/local/lib/python3.5/dist-packages/PIL/Image.py", line 95, in <module>
        from . import _imaging as core
    ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory

when running `examples/sevensegment_demo.py`.

The fix is:

    sudo apt install libopenjp2-7

KarolinaPondel, 2024-08-05:
[Blanchot](https://github.com/Blanchot)
I have similar problem, unfortunately after _sudo apt install libtiff5_ I get _E: Unable to locate package libtiff5_

thijstriemstra, 2024-08-06:
@KarolinaPondel when using debian bookworm, use libtiff6.

KarolinaPondel, 2024-08-06:
@thijstriemstra 1. I have Ubuntu on the server I use. 2. I have libtiff6 installed, unfortunately the code from the repo I want to run requires version 5.

thijstriemstra, 2024-08-06:
any error messages? it should work with both; you can also try libtiff5-dev i think.

KarolinaPondel, 2024-08-06:
Same error like above: 
```
ImportError: libtiff.so.5: cannot open shared object file: No such file or directory
```
I tried libtiff5-dev but it doesn't work.

thijstriemstra, 2024-08-06:
probably related to Pillow version you're using, but I can't tell, you didn't paste the full traceback/error message.

I would try upgrading Pillow with `pip3 install -U Pillow`(probably needs sudo).

ABC00012345, 2025-04-22:
do symlink from libtiff.so.6 (after installing it) if libtiff5 or libtiff5-dev not installable